clough42/electronic-leadscrew #4: Cheap Power Alternative

gusosborne, 2019-04-25:
Thinking this could be useful to anyone that doesn’t have a proper bench power supply (and doesn’t already know about these). It’s a cheap way to get your 3.3v / 5v / 12v / 24v rails all in one shot. Just need an old ATX PSU and one of these breakout boards. 

http://dangerousprototypes.com/blog/2014/09/19/atx-breakout-board-bench-top-power-supply-build/

Most PSUs are probably not stout enough to drive a stepper long-term, but it should be good enough for testing. These are available for a few bucks everywhere.

clough42, 2019-04-26:
If all you need is 12V max, this is a pretty cool little board.  For this kind of project, though, you'll likely want to be running much higher voltages.

closed 2019-04-26 by clough42

gusosborne, 2019-04-27:
It does 24v as well   (+12 & -12), just good enough to test a stepper / driver.