clough42/electronic-leadscrew #6: Collaboration

gusosborne, 2019-04-29:
I wasn’t sure of the best way to contact you, or where to post. I did get all the hardware and your most recent firmware up and running last night - thank you for this project! 

I also have a G0602, so I’ll probably start experimenting with some physical mounting options next. I’m happy to bugtest, maybe create some docs, or whatever else I can do to help.

closed 2019-04-30 by gusosborne

clough42, 2019-05-01:
Very cool.

For the physical mounting, I'm thinking about a couple of things.  Other
ideas are always welcome:

   - I was going to gear-drive the encoder, but I'm now thinking about
   pairing a GT2 pulley with the first nylon gear and driving the encoder from
   that
   - To drive the leadscrew, I think there's room for a NEMA23 motor or
   servo above the motor, below the electrical box on the back of the lathe,
   and a path for a timing belt to the input of the existing gearbox.

James

gusosborne, 2019-05-01:
Sounds good, though I was thinking of direct coupling the stepper to the end of the lead screw, like some others have done. That or a very small timing belt at the tail end (stepper under it, parallel). I’m thinking about the backlash being eliminated by bypassing the gearbox. Is there a reason you want to run the ELS through the gearbox? 

I’d like to be able to retain the stock gearbox use, just for quick turning, if possible. I am not sure if this 425oz stepper with a 48v drive is up to every task, especially at 1:1 gearing, however. 

I really like the idea of the encoder geared 1:1 to the spindle, either direct to the 40t, or using the adjacent 60t as an idler. I’m currently printing a couple standoffs to try out with that. 

Oh- One more quick question - Should this line:
https://github.com/clough42/electronic-leadscrew/blob/892a21fc1712b56361fc15996be963822e62e7f3/els-f280049c/Encoder.cpp#L87

..have a variable at the end, more like:
rpm = count * 60 * RPM_CALC_RATE_HZ / **_ENCODER_RESOLUTION_**;

To keep the encoder option configurable, regarding RPM? Or am I just misunderstanding this calculation..?

I’m currently using a 600p/r encoder while waiting on the suggested Omron to arrive from China. 

Thanks again,
Gus

clough42, 2019-05-01:
I'm planning to run through the existing gearbox just to minimize the
modifications to the lathe.  I want to make this as accessible as
possible.  There's always going to be backlash in the halfnut anyway, so
I'm not worried about the backlash in the gearbox.  If I can put a keyed
pulley right on the input shaft, retained with the original screw, it's a
piece of cake to take off the pulley and put on change gears if I want to
use them again for some reason.

For prototyping, I 3D-printed a bracket to hold the encoder in the right
place with a Noga Popeye magnetic base.

Yes, I think that line of code should be updated to use the define from
Configuration.h.  I submitted an issue:
https://github.com/clough42/electronic-leadscrew/issues/7

James

gusosborne, 2019-05-02:
Ok, that makes sense to me. Any chance you could post a couple quick pics?