clough42/electronic-leadscrew #12: Add max speed protection

clough42, 2019-08-25:
There should be a limit to how fast the ELS will dry to drive the physical leadscrew.  Without protection, it's possible to set the ratio for 8TPI (4mm?) and then spin the lathe up to 3000RPM.  This is completely unrealistic and may cause loss of synchronization or even physical damage.

Suggest an RPM limit in Configuration.h.

The ELS should hard-limit the speed and display a warning on the control panel.

pauly7300, 2020-01-01:
Agree this would be a good enhancement.  Now that it's so easy with just a press of a button, one can quickly get into a precarious situation.  

I assume you're envisioning a lead screw RPM limit which would be derived from the calculations already being made to turn the lead screw?

Thinking through this a bit more, here's my take on what the requirements for the feature might look like.

Assumptions:  Each lathe owner will have a finite LS RPM limit in mind for their particular setup.  This limit is not conditional on any particular operation being performed (threading or turning) and is merely a means by which a user can safeguard against unintended configurations of spindle speed to operation values (TPI/Turning feed rates).

1. ELS code should allow the user to specify two ELS RPM limits (Warning and Shutdown).
    A.  Warning limit - when the LS RPM reaches this value, the panel lights flash a warning message indicating the user is close to exceeding the RPM.  "LMT-WARN"
    B.  Shutdown limit - when the LS RPM reaches this value, the LS rotation is disabled/halted with a flashing message indicating "LMT-STOP".
2. In order to 'reset' the ELS, the spindle RPM should be taken to 0.  IE, the user should stop the spindle.
3. The 'buffer' between the warning and shutdown limit is up to the user to determine.  A recommended approach could be stated in the comments of the code but ultimately it's up to the owner to determine.  If the shutdown limit is <= the warning limit, the ELS should immediately halt.

The above seems like a practical approach to at least provide a warning before shutting down in the middle of an operation.  With only a single hard limit it's not hard to imagine being very close to the RPM limit specified and then nudging the RPM just a bit higher (during the operation) and it stops abruptly, ruining the part etc.

Thanks as always and I'll be interested to see what others think.  I would offer to assist with coding but unfortunately coding (at this level of complexity at least) isn't in my skill set.  ;-)

clough42, 2020-01-22:
I agree with this.  And this is exactly how I've been thinking about it.
There has to be a warning before the leadscrew stops.  I'm imagining a
scenario where the operator is running just below the limit without knowing
it and having the leadscrew stop suddenly on a speed fluctuation during a
threading operation, ruining the part and potentially breaking a tool or
damaging the machine.

It also makes me wish that I had added a piezo buzzer to the PCB.  :)

Is there anything else reasonable to do instead of just stopping the
leadscrew?  I have been thinking about this for a while and haven't come up
with anything reasonable.  All the choices are bad, including just letting
it overspeed and loose sync.

James

pauly7300, 2020-01-23:
With features like this, it's easy to fall into the trap of thinking the feature is leading someone into a bad situation but as long as the feature is optional and has a warning buffer, all the power is in the users hands to use the feature or not.  Certainly no one should be frustrated with the results if they test the behavior first.  

And I'm fairly sure a good bit of warning and description in the code and on the wiki will sufficiently educate everyone.  And anyone that already experienced out of sync or stoppage by the hybrid stepper itself will appreciate (and want) the feature anyway.

alexphredorg, 2020-02-19:
I just submitted a pull request which implements this:
https://github.com/clough42/electronic-leadscrew/pull/54

It's a little weird to figure out leadscrew RPM from the feedRatio, but it works on my machine.  

I also hooked up the power button because this errors out the spindle drive by disabling enable on it.  You need the power button to turn it back on.

Since the set button was unused I made it toggle to show spindle vs leadscrew RPM.  That is a debug hack that can be removed.

This setup is a lot nicer than erroring out the stepper driver and having to power cycle the ELS.

Thanks for making this, it's working great on my Emco Compact 10.