clough42/electronic-leadscrew #15: Configurable ratios for stepper / encoder pulleys / gears

gusosborne, 2019-09-01:
Hello,

Maybe I’m missing something simple again- but where would we configure the ratio for the pulleys (1:2 stepper driving lead screw, for example)? I would actually prefer to use gears in my setup, if at all possible. Would something more like a teeth setting ratio be possible (120:127 gears) to drive the leads screw? 

I expect my encoder to be a 1:1, but curious if that configuration parameter exists (or will exist) as well?

Thank you again for this project! I’m pretty close to actually testing mine (especially if I can use gears;).

clough42, 2019-09-01:
I don't have configurable parameters for those ratios.  I just include them in the step and count values.

I have my servo driver set to 1000 steps per revolution, and the belt drive is 3:1.  For this setup, I set the steps value to:. 1000 * 3

I could also set it to 1000 and set the microsteps to 3.

I could add an additional numerator and denominator for each, but it's already pretty complicated, and I was trying to keep it simple.

gusosborne, 2019-09-01:
That all makes sense. Sounds like I can get what I need done, with some simple math as well then. Something like setting my steps to 1058.3 (assuming a decimal is allowed?), for my 120 driving 127 gears? 

Is it important to set your steps to 1000 * 3 rather than 3000 for some reason ?

Thanks again.

clough42, 2019-09-01:
The math is done with whole numbers only to avoid representation issues inherent to floating point arithmetic.

I use 1000 x 3 just for clarity, so I'll remember what I did.

Why a 127-tooth gear?  That's usually used for metric/imperial conversion.

gusosborne, 2019-09-01:
It’s just the perfect size to be able to bury the stepper inside of the lathe, keep everything flush and hidden. I don’t have to do it that way, but I’d like to. 

[image]

[image]

gusosborne, 2019-09-02:
I’ll just flip the bar around and try to use an idler then. I’m thinking this 2:1 (120 / 60 tooth) is going to be my best bet, using the stock gears. Then I can just configure my steps to 1000 * 2 (and you can close this thread;).  

[image]

I was hoping to jam the encoder in between the ways (using a modified nylon 60 tooth) -  it would probably need some extra protection from oil and debris though.

gusosborne, 2019-09-13:
Freakin beautiful man!

LotusElan, 2019-09-13:
Thanks gusosborne.  This is how I handled the encoder.

 
[image]

J642, 2019-10-31:
Hi LotusElan The gears is it 40 tooth on both side. Jorgen

LotusElan, 2019-10-31:
The sprockets on the stepper and leadscrew are both 38 tooth and the ones on the encoder and spindle are 45 tooth.  So both are a 1:1 ratio. I based the encoder/spindle ones on what size would fit over the spindle shaft.  I think I based the stepper/leadscrew ones on what size I could make fit the leadscrew shaft.  I did have to machine 3 of the 4 pulleys to fit. Yes, the stepper is tucked inside the bed.  It didn't look like James had that option on his lathe.  Mine is in working order and I have used it but I can't wait to get a board for it so I can have a more polished install.

John

gusosborne, 2019-11-05:
Finally got back out to the shop today - happy with it so far. Now on to the encoder..

[image]
[image]
[image]
[image]

clough42, 2019-11-09:
Looking good!

Greecher, 2019-11-30:
As this seemed where the install showing where, here is mine.  First is my laser cut faceplate, second is showing how 3.3v straight through connection to TI board for display (no level shifter in place) works,  I think good brightness.  Next is power/servo controller  on the side of the cart that holds the lathe, last are internal mount of servo and encoder.  I think I'm pretty happy with the robustness of the 3d print that holds the servo in place, was able to re-use the banjo bolt holder and side cover bolt to secure it in place.  

For the encoder I chose to use a double spaced bracket with skateboard bearings to support the pulley, as I didn't want to overstress the encoder bearings.  This was more necessary in my case, as when I pulled the old gear out and replaced it with a trimmed pulley (it is the same dimension pulley as attached to encoder, but trimmed down to just be the teeth, and a 3M 6mm width to match the dimension of the gear width I pulled out).  I may replace those 3d printed plates with aluminum, but for now it seems fairly steady.  Note I also used an alternate encoder that has a place to insert the end of the turned spindle that fits in the bearings and pulley center, so that it is more low profile.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

gusosborne, 2019-11-30:
Awesome! I love seeing all the different ideas and implementations on these mounts.

J642, 2019-11-30:
It looks like my lathe and the set up I am trying to make. Awesome. I am
missing the interface board from James It came to Denmark but 150 mile from
my home. Ebay returned my money an I wait for a new batch of boards.

Den lør. 30. nov. 2019 kl. 14.45 skrev gusosborne <notifications@github.com

Greecher, 2019-11-30:
> It looks like my lathe and the set up I am trying to make. Awesome. I am missing the interface board from James It came to Denmark but 150 mile from my home. Ebay returned my money an I wait for a [...]

As I chose to just power the TI board via the usb (which allows me to have programming access for updates), and just used 3.3V logic direct to the display board, I only had to build a small board for the 3 regular servo feeds (Resistor 1K, Capacitor 47pF, Mosfet 2N7000 for each) + alarm (2 resistor  10K, Capacitor 47pF).  James has a nice diagram for them in Circuit Maker, which I breadboarded first, then just made a small I think 12/12 hole perfboard.  Only thing I don't have is the support for the memory module, which depending upon software upgrades of ELS, I may either breadboard or purchase one of Jame's boards.

clough42, 2019-11-30:
Thanks for sharing your build.

Just for others that might read this, do not try to power the LaunchPad
board via USB (and the on-board 3.3V regulator) if you're using one of my
interface boards, or the 3.3V regulators will fight against each other.  I
honestly don't know what would happen, but I wouldn't recommend it.

James

Greecher, 2019-11-30:
> Thanks for sharing your build. Just for others that might read this, do not try to power the LaunchPad board via USB (and the on-board 3.3V regulator) if you're using one of my interface boards, or [...]

I would agree with this, it's fine for my one off, probably wonky results will ensue under double power.  Do note, I would still remove the jumpers and leave a cable in place, as if you are like me, I buttoned up my board in the back space, and didn't want to leave it open with the HV caps on the driver board for my lathe, so if you want upgrade access, you can still do so without cracking it open.

Dave

clough42, 2019-11-30:
One other thought...

If you're running the board off a 5V supply through USB, the 3.3V supply is
provided by the TPS79601DRBR regulator on the LaunchXL board.  You'll want
to keep an eye on the current draw and temperature to make sure you don't
run into trouble down the road--especially if your installation is in a
space that gets hot.

James

carneeki, 2019-12-14:
I just had a thought for the mounting of the stepper. I have a CQ6125x550 which is a metric variant of the G0602. If you're fine with direct drive, then the gearbox could be removed, and a new bearing block + stepper bracket could be fastened to the casting where the gearbox used to go. Finish up with a new sheet metal cover and the control box and VFD knob can be put on something nicely etched and look factory fitted :)

(I realise it's a lot more work but I see it as one of those "how long is a piece of string?" type questions as to how far one wants to take this)

closed 2020-10-11 by clough42