clough42/electronic-leadscrew #18: PCB Files

wkellyoconnor, 2019-09-07:
James, what an awesome project and they way you explain stuff, is way above most anyone else I've seen. I'm not sure how else to get in touch with you, I did leave  message in Youtube but you may not have seen it, under Video 8. Please let me know of a better way to ask a question.  I have all my parts, I have started building mounts for the encoder and servo, If you would be willing to share the  gerber files for the small board you designed I could get it ordered and on the way and build it. I do have a reflow oven as well. 

Sincerely, Kelly

clough42, 2019-09-26:
The PCB design is available on CircuitMaker, but if you wait just a little longer, I have a batch of PC boards ordered and I'll be making them available very soon.

closed 2019-09-26 by clough42

wkellyoconnor, 2019-09-28:
> The PCB design is available on CircuitMaker, but if you wait just a little longer, I have a batch of PC boards ordered and I'll be making them available very soon.

Thanks James