clough42/electronic-leadscrew #20: New feature: encoder driver handwheel

siwyroot, 2019-09-24:
Hello James,

Is it possible to add following feature:
Switching between encodes input base on lead screw engagement leaver. Something like in this clip: [https://www.youtube.com/watch?v=lMS2DAUD_DA](https://www.youtube.com/watch?v=lMS2DAUD_DA) .I have a small metric lathe during retrofitting to your ELS and wanted to use ball-screw with closed loop stepper around 8.5N.m  remove split nut. It would be nice to know what you thing of that idea.

Thanks!

clough42, 2019-09-26:
So the idea is to be able to jog manually so you could use it on a lathe without a halfnut?

What would be really awesome would be to have a virtual halfnut switch and put an encoder directly on the normal handwheel of the lathe.

I think this is pretty advanced, but something that might make sense someday.

siwyroot, 2019-09-27:
Hello,

Exactly this is the idea, I wonder how controller would behave in moment
when we would switch by some chip input of the encoder to another one.

BR

clough42, 2019-10-01:
The microcontroller board has a second encoder input, so something like
this would be possible.  Right now I'm focused on getting the basic
functionality wrapped up and available for people, but this is definitely
something to think about.

James

dentikhval, 2019-10-20:
Hi, this is exactly what I am thinking about and what I am building.
I will have a ballscrew nut directly connected to the apron, no halfnuts or hand feed options on the lathe. (It was a weird special purpose lathe so it originally had no leadscrew or auto feed)
Hence, I am thinking of using the second encoder input QEP2 to read a 100PPR MPG wheel that would act as manual apron feed would on a normal lathe. That would call for three operation modes: Auto feed as it exists now, Manual feed with a settable MPG/Apron ratio, and Threading as it exists now.

Also, the code has to be modified in our case to use buttons for auto feed to be able to stop at the right moment and not run into something (which wouldn't be a problem if we could unlock the halfnut). 
The buttons on the LED&KEY module won't suit this purpose so bigger, more responsive, bigger non-locking buttons (or better a lever switch for Auto feed FWD/STOP/REV) would have to be installed in a comfortable place.

While we're at it, the mode selection buttons should also be bigger physical switches because the flimsy buttons of the LED&KEY will surely suffer from frequent presses in a no-halfnut environment. 

I'll keep you posted if I have any progress with that.