clough42/electronic-leadscrew #22: Question about basic functions

Hansveer, 2019-10-16:
Hi James,

My ELS is functional! Thanks for your initiative in realizing this project!
However, having watched your vlogs several times I was unable to find an answer to three basic questions:

Are the following functions already implemented? 
- Power on/off. If so, it doesn’t work on my system.
- Set function. Pressing it returns a "No Setting” message. Is it supposed to do more at this stage or is it Work in Progress?
- Eprom function: What is (or will be) the purpose and how does (will) it work?

I am not sure if this is the right place to ask questions like these. If not please advise.
Kind regards,

Hans

clough42, 2019-10-18:
These are all in there for the future.  The power switch will eventually
shut down the display and disable the servo/stepper driver.  That may or
may not be useful if you have everything on a power switch or contactor.

The settings button is there so the leadscrew, encoder, stepper, etc. can
be configured through the display instead of requiring it to be set up in
the Configuration.h file.

The EEPROM is there to store the settings, and perhaps store the last
activated mode so it will return to the same threading/feed/units mode on
power-up.

James

Hansveer, 2019-10-21:
Thank you for your prompt response and clarification!
I am really looking forward to the Eprom function being implemented!

Regards,

Hans

georgehilliard, 2019-11-23:
Thank you Hans for opening this issue and thank you James for answering. I got one of the first boards (great service by the way) ,replaced the breadboard and everything lit up and worked, but have spent several hours trying to figure out why the power button didn't do anything.

clough42, 2019-11-30:
Sorry about that.  Someday...

AndrrewG, 2020-02-27:
I'd like to add my vote for control panel power switch functionality. The control panel seems like the right place to shutdown the motor driver and display rather than the cabinet which is less accessible. Everything else is super low power so happy for it to be on when ever the workshop is powered up (I have the entire workshop on an isolator switch).

Johnboy251, 2020-02-27:
James and all.

I would like to second what Andrew G has emailed. My C++ programming skill are a lot to be desired but I’m still learning. The TI ccs takes a bit of head scratching from time to time.😳

AndrrewG, 2020-02-27:
I would give it a go but I'm not a FW guy. I managed some small tweaks (default to metric mode, added a slower feed rate to the table) but not sure where to start with power down. I would imagine that its fairly straightforward to have it  just disable the motor driver and power down the screen but nothing is ever as simple as it seems. I would have thought that the micro has lower power modes but I'm not fussed about using that to be honest, just powering down the driver. I didn't go for a closer loop stepper for this so its always drawing ~20w even when its not in motion.

Johnboy251, 2020-02-27:
Thinking about this - if the stepper/servo Drive was ‘just’ inhibited so that the drive was effectively in stand by power mode, I think that’s all I would need. Something for me to ponder.🤔👍

AndrrewG, 2020-02-27:
> Thinking about this - if the stepper/servo Drive was ‘just’ inhibited so that the drive was effectively in stand by power mode, I think that’s all I would need. Something for me to ponder.🤔👍

Yes that is what I want to do, and also turn off the screen. I have a rice cooker (instant pot) where the screen displays "OFF" when it's supposedly powered down which is infuriating, why not just actually turn the screen off!

Johnboy251, 2020-02-27:
Andrew - good idea. I’ll give it a go. My C++ mentor will be back holiday soon, I get him to steer me hopefully in the right direction.
I say one thing - James project has inspired me to build  the ELS and start back learning C again. I’ve been playing with Arduino projects a while back.

clough42, 2020-10-11:
I just added the power button functionality on the dev branch.  It will be released soon (version 1.3.01)

closed 2020-10-11 by clough42