clough42/electronic-leadscrew #23: New feature idea: Electronic thread indicator for hybrid thread cutting.

bondarchook, 2019-10-29:
Hello James,

After watching your video about complications of cutting hybrid threads, I got an idea. 

So during hybrid thread cutting it is not possible to use thread indicator because conversion math between inch and millimeters does not produce whole number on thread indicator. 

But since ELS already know position of the lead screw at any time you can calculate ideal position to reengage half-nut. 
And then present this in the UI in form of "progress bar" of some kind so user can anticipate and lock half-nut in correct position. 

The only problem I can see with this idea. Is that fact that when half-nut closing it will snap to closest thread on the lead screw and this may shift it from ideal position. 
But I am not machinist my self so I can not asses if this will be a problem or not.

clough42, 2019-11-05:
This would require more data--either an encoder on a threading dial gear or an encoder on position of the carriage.  Since the correct position will be a multiple of 127 revolutions of the leadscrew, this doesn't seem very practical.

RJSakowski, 2019-11-07:
There is a method for using the half nuts and thread dial for cutting metric threads with a Imperial pitch lead screw.  The half nuts are engaged at a particular mark on the thread dial and the threading pass is made.  At the end of the pass, the half nuts are disengaged and the lathe is shut down.  The tool is backed out and the lathe reversed. when the mark is aligned on the threading dial, the half nuts are re-engaged.  The carriage is taken back to the start of the pass, the lathe turned off, and the threading tool positioned for the next pass.  The lathe is started in the forward direction.  

This doesn't save any time as the carriage still has to be run back but it does allow one to thread to a shoulder.

michalsok, 2021-01-29:
Hello, I also would like to see this feature and I believe that the half nuts need to stay engaged to stay in sync.
We have the index signal from the encoder so we would know when to start moving again but a hard stop switch would be required and then the return feed to starting position would have to be done using the stepper so precise position can be preserved.
Just a theory, I did not think it through but I think as long as the half-nut stays engaged and we know the spindle/chuck position using encoder and index signal, we should be able to do it.
Mike.

Fastdruid, 2021-03-19:
You need more info to do it. The best way would be to do it off an encoder geared off the leadscrew and attached to the carriage.  

While you don't actually need an encoder on the lead screw itself because you "know" where the lead screw is and can calculate it depending on the relative number of turns that only works if you have the carriage in the same start position *and* to "set" the half-nuts somehow so the ELS "knows" where it is. That would be a royal pain. 

Unfortunately without feedback of the location of the carriage and some way to "set" the half nut position it's going to be very awkward. I think you could do it standalone though, I was watching this video https://www.youtube.com/watch?v=zG9XYDCDwWY where he made a geared dial indicator for multiple threads and I was thinking that you could make a standalone which would do the same, it wouldn't need to be a high resolution encoder either, you could probably manage it with an arduino.