clough42/electronic-leadscrew #25: Mini-Lathe

fla-940, 2019-11-08:
Has anyone converted the Harbor-Freight 7x14, SIEG style lathes and have documentation on it as to the steps you took?

gadjet, 2019-11-15:
Would that be similar to this one?
[Warco mini hobby lathe](https://www.warco.co.uk/metal-lathes-metalworking-lathe-machine/302922-mini-hobby-lathe.html)

TheRealGreatOldOne, 2019-11-16:
I am in the process of converting my Sieg SC4. I have all of the electronics squared away, and now need to fit the stepper, spindle encoder and drive pulleys to the lathe.

fla-940, 2019-11-16:
> 
> 
> Would that be similar to this one?
> [Warco mini hobby lathe](https://www.warco.co.uk/metal-lathes-metalworking-lathe-machine/302922-mini-hobby-lathe.html)

Yes sir very close.....

fla-940, 2019-11-16:
> 
> 
> I am in the process of converting my Sieg SC4. I have all of the electronics squared away, and now need to fit the stepper, spindle encoder and drive pulleys to the lathe.

Looking forward to your project out come! Keep us in the loop!

gadjet, 2019-11-16:
I'm going to put the ELS on my Warco mini lathe, I've already installed a stepper motor onto the lathe and control it via an Arduino but just for an auto feed, I'm currently working to add the ELS project to give it threading ability, I will put up a post on my blog when completed.
[image]
[image]
[image]

TheRealGreatOldOne, 2019-11-16:
Here’s my work so far:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

gadjet, 2019-11-16:
Looks good, where did you get your large enclosure from?

TheRealGreatOldOne, 2019-11-16:
Amazon:
[https://www.amazon.co.uk/380mm-Junction-Weatherproof-Waterproof-Enclosure/dp/B07B66ZH1G/ref=pd_nav_hcs_rp_2/260-5082970-1356367?_encoding=UTF8&pd_rd_i=B07B66ZH1G&pd_rd_r=3a7b5a9d-d51d-4815-a229-3c198e51c3a8&pd_rd_w=qLMlq&pd_rd_wg=qYRmx&pf_rd_p=f6051d3e-949c-405f-913b-20ea296ef01b&pf_rd_r=VT6EJ383BKPH1Z4R72CM&psc=1&refRID=VT6EJ383BKPH1Z4R72CM](url)

clough42, 2019-11-16:
That looks really nice.

It looks like you have a common ground between the motor power supply and
the logic 5V supply.  Some people have had trouble caused by motor noise
interfering with the logic.

James

TheRealGreatOldOne, 2019-11-16:
> That looks really nice. It looks like you have a common ground between the motor power supply and the logic 5V supply. Some people have had trouble caused by motor noise interfering with the logic. James
> […](#)

Thanks James. I’ll see how I get on once I have the motor and encoder mounted on the lathe, and re route if I get into issues.

TheRealGreatOldOne, 2019-11-20:
[image]
[image]
[https://www.thingiverse.com/thing:3989108](url)
[https://www.youtube.com/watch?v=lNFqjpXHLy4](url)

The RPM agrees with the lathe... Ish. It jumps around a bit, but not a lot. Has anyone else encountered this?

clough42, 2019-11-22:
It's updating 10 times per second by default.  You can slow it down by
modifying  RPM_CALC_RATE_HZ in Configuration.h.  Setting it to 2 would
update twice per second, for example.

Slowing it down will make it less responsive, but also clean up some of the
jitter.

James

TheRealGreatOldOne, 2019-11-24:
USB socket added to the control box for easy firmware upgrades. Servo / Hybrid stepper mounted, Control box hung on the back of the lathe, change gear cover modified to fit over the encoder and servo. Oh, and wired in so it all turns on when the lathe is switched on. :)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

https://youtu.be/Yq_rMzeSS1s

What I have encountered whilst testing is I can quite easily get the driver alarm to trigger... depending on spindle speed and thread pitch / feed selected. My lathe has a top speed of 2000rpm, and with the pulley ratio of 3:1 and then whatever the multiplier is for feed - you can quite easily top out the max RPM of the servo. 

I've just made a chart of max possible spindle speed for a given feed. For the most part, the feeds are fine. it's the thread pitches that can make it go down... but I very much doubt I'd ever need to do any single point threading above a couple of hundred RPM anyway. And the convenience of not having to bugger about with change gears far outweighs the possible need to do a 5 TPI thread at over 1000RPM

TheRealGreatOldOne, 2019-11-26:
Final piece of the puzzle - mounted the display.

[image]
[image]

clough42, 2019-11-30:
Thanks for sharing your build.  This is awesome!

James

buzzkc, 2020-03-10:
> I'm going to put the ELS on my Warco mini lathe, I've already installed a stepper motor onto the lathe and control it via an Arduino but just for an auto feed, I'm currently working to add the ELS [...]

Would you be willing to share your bracket design? I've got a 7x14 I'd like to adapt this to.

gadjet, 2020-03-10:
No problem, I'll put it on thingiverse and post the link here.

buzzkc, 2020-03-20:
> No problem, I'll put it on thingiverse and post the link here.

No rush on this, I realize with all that's going on with the world that this isn't a priority. I was wondering what gear tooth count you're using on the encoder. It looks like my encoder and display should come by this weekend, so may mess around with programming the launchpad...Thanks!

gadjet, 2020-03-20:
Hi,
I've added the files to Thingiverse.
[https://www.thingiverse.com/thing:4232927](url)

pikolo335, 2020-04-09:
Hi !what king of hybrid motor controller is this that you have? Can i have the wiring please?

Vrossi050, 2020-04-09:
Same as what James recommended from Automation Technologies.

https://www.automationtechnologiesinc.com/products-page/hybrid-servo-system/hybrid-servo-drive-kl-5080h/

davidgleach, 2020-09-23:
For what its worth, here is my conversion on my Seig mini lathe.  It works very well. 
I 3D printed the belts, pulleys, several of the mountings, control panel and the extension to the end cover.
I have removed the cross slide to allow the provision of a very stiff tool post.  The  feed screw can be rotated manually.  consequently, I have included a switch to isolate the stepper motor for manual feeding.
[image]
[image]

JonWoellhaf, 2020-09-23:
Very nice, David!

clough42, 2020-10-11:
Wow.  That looks really nice!

davidgleach, 2020-10-12:
Thanks James.  In order to improve stiffness of the little Seig lathe, I 
have removed the top slide and use the lead screw in manual mode (A 
simple DRO ensures accuracy).  Hence I hard wired a switch into the 
motor circuit so that I could disable it's supply power.  This works 
pretty well for me as I only thread occasionally.
Very grateful for your design and videos - they have taught me a lot.

Cheers
David
David Leach

Home: 03 5310 6313

Mob:    0416 119 381

Cc: "davidgleach" <davidgleach@iinet.net.au>; "Comment" 
<comment@noreply.github.com>
Sent: 12/10/2020 10:11:04 AM

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

clough42, 2020-10-12:
I just released version 1.3.01 so you can now use the power switch to turn off the motor and still retain the RPM readout.

closed 2020-10-12 by clough42