clough42/electronic-leadscrew #26: 3.3V display board

Greecher, 2019-11-22:
When I was doing my implementation, and had issue with my display board, I went to test it out, and found most all references for the use of it just used 3.3V supply, foregoing the  level shifter all together.  When I got a few replacement ones, they worked happily that way out of the box, so a simpler circuit where the display connects directly to the TI board was all that was needed.

clough42, 2019-11-30:
That's good to hear.  I checked the specifications for the TM1638 driver on
the display board, and the datasheet only shows a nominal voltage of 5V.
It doesn't show a minimum.  I would think it would make a difference what
color of LEDs you were trying to drive with it, given their different
forward voltage drops.  I don't know.

James

Greecher, 2019-11-30:
I finished my install, had to hand build a small interface board for servo to controller, but left the 3.3v straight through for the display board.  I doesn't seem dim at all, and can see no ill effect.  I think I may fly this way without the board you make, but I want to give you something without getting the board, ie save the board for the next purchaser, but give you money for your great effort!  You got a patreon page or something?

closed 2019-12-01 by Greecher