clough42/electronic-leadscrew #27: Error on Stepper Driver

AZ-Ron, 2019-11-24:
Greetings all. I've got my rig set up to test, I have the Launchpad XL, James' interface board, and the Hybrid Stepper and controller he recommends KL-23N-1000 and KL-5080H, OMRON Encoder E6B2-CWZ6C, and the LED/Key interface board. I flashed the Launchpad after changing the TPI of the leadscrew to 8 (Logan 14x40). The Interface displays CLOUGH42 upon bootup and ELS 1.0.0.3. 
I've got a Toroidal Power supply putting out 46V going into the Hybrid Controller. 
I get a green LED on the 5080H, but after spinning the encoder a few times, the red error LED comes on, flashing 7 times, pause, then repeats, indicating a positioning error. The motor will give a little jerk maybe once or twice before the error, otherwise, the motor doesn't move. I've double checked my wiring, and honestly am at a loss as to the issue.
I've tried it in TPI, Metric and SAE, and none of that seems to matter. If I press the 'Set' button, the display shows "NO SETTINGS"... Did I miss something? Probably something silly, but I can't seem to see the forest through the trees. Any thoughts or direction to go would be much appreciated.

Thanks!

AZ-Ron, 2019-11-24:
OK... It was me. I'm going to leave this up here though just in case someone else Bone-heads this. 
The Encoder cable from the motor to the controller has a connector on the end of it that is a bit too wide to fully seat into the 5080H... You need to use the additional cable that comes with it. It will allow the connector to fully seat. Even though I thought the cable directly from the motor was seated far enough into the 5080H, it wasn't. 
Everything is working fine now.

clough42, 2019-11-30:
Thanks for the update.  I have seen situations like this.  I've even seen a
situation where you had to attach the connectors in the right order for
them to fit properly.

James

Greecher, 2019-11-30:
I think it is more than that AZ-Ron, I have same servo, and I note that the plug on the extension is an add on.  I think the plug ends are remapped, as I tried ti trim the bit of excess of the plug for a direct connection, and got a good seat, ie able to complete screw it in and not interfere with the side connectors, and 100% same issue.  Put other cable back in, and all works again. I'll take the time to do the pinout of the cable pins to see where they map to.

Greecher, 2019-11-30:
Attached is the pinout analysis, and it is as I thought, the extra cable is not an extension but an adapter cable, apparently the servo has a fairly significant difference to what the controller needs.
[image]

clough42, 2019-11-30:
Oh, thanks for this.  I'll be making up a cable set for permanent
installation on my lathe soon.

Greecher, 2019-11-30:
> Oh, thanks for this. I'll be making up a cable set for permanent installation on my lathe soon.

Do note, that the female end when looking at it runs 5-1 instead of like male pin, probably obvious, but better safe than sorry.  I think I'll just shorten the cable I got to be a 3-6" dongle, as my controller location is very close to servo.

AZ-Ron, 2019-12-01:
Awesome work Greecher!! I'd thought the cable was seated far enough, but after dealing with serial cables for several decades, I know that they can appear fully seated and just be a hair out, and problems pop up like getting gremlins wet!!

itstallion, 2020-04-17:
> Greetings all. I've got my rig set up to test, I have the Launchpad XL, James' interface board, and the Hybrid Stepper and controller he recommends KL-23N-1000 and KL-5080H, OMRON Encoder [...]

Hey AZ-Ron,

I would contact you directly but I don't have your info.  I too have a Logan 14x40 (6561H) that I'm putting the ELS on.  Just curious what size stepper did you go with and have you mounted it to your lathe yet?  If so do you have some pics to show your install.

Thanks
Mark

Vrossi050, 2020-04-17:
Can you share some pictures and settings?

AZ-Ron, 2020-04-17:
HI Mark. I wound up doing a transfer within my company up to Idaho Falls.
My entire shop is in storage right now. I've had my unit assembled on the
bench, and it seems to work fine. It'll be at least a few months before I
get settled into a new shop. I need to sell my house in Arizona, and
wouldn't you know it, during a pandemic is a.lousy time to try and sell a
house. I'm in an apartment for the time being. I'd be more than happy to
chat about the Logan stuff, but as of yet, I haven't made a mount or
anything to actually install it yet.

As far as getting the errors after a few spins... I had the exact same
thing happen. There's is a "loose" cable that comes with the stepper
driver... this is NOT an extension. I left it off at first, thinking I
didn't need the extra length. The pins on it are wired slightly different.
Install that cable, and 10 to 1 your issue is gone. 😁

Cheers!!

AZ_Ron... guess I should change it to ID_Ron!

clough42, 2020-10-12:
I think everything noted here has been added to the troubleshooting document, so I'm closing this issue.

closed 2020-10-12 by clough42