clough42/electronic-leadscrew #28: Drawing controlpanel fusion 360

J642, 2019-11-26:
Hi have anyone made drawings in Fusion 360 of the controlpanel

TheRealGreatOldOne, 2019-11-26:
I modified James's files as they didn’t fit my LED&Key module. 
https://www.thingiverse.com/thing:3904034

James’s originals:
https://www.thingiverse.com/thing:3740612/files

https://myhub.autodesk360.com/ue290368a/data/permalink/DT1640fQT03b86d98cd47547742998bd6ec2
https://myhub.autodesk360.com/ue290368a/data/permalink/DT1640fQT03b86d98cd4888f386417a4072b

gusosborne, 2019-11-26:
Awesome- thanks you!

tsbrandt, 2020-01-08:
Hi,
I have my ELS functioning and it works great! Thank you James! 
I now need to enclose everything. Couple of questions:
- What enclosure did you use for the control panel?
-Can you recommend a vendor that would print out the 3D components (i.e., spacer and button covers)?
-What material should I spec for the button covers?

Thanks for such an excellent project!

clough42, 2020-01-22:
I used a BUD Industries CU-124 box:  https://amzn.to/2XSsORX

I printed the spacer in ABS and the buttons in NinjaFlex, but any rigid
material should work for the spacer, and any flexible material for the
buttons.

James

AndrrewG, 2020-02-27:
My "LED and KEY" board matches neither the original model or the modified model posted here. In my case, the mounting holes line up but button pitch is different and the 7 segment modules are a slightly different size. I will create yet another control panel lid model and post it here in case it is useful for anyone.

TheRealGreatOldOne, 2020-02-27:
I have a feeling that the QA on these boards is minimal, and the variation in the spacing of features is going to be the norm. We could end up with a library of lids! 🤓

AndrrewG, 2020-02-27:
I don't think its a QA issue, the PCB design is clearly (slightly) different. I guess each factory makes their own version of what is probably the chip manufacturers original dev board based on the components they already have available or the cheapest available locally. At this qty, the boards you receive will be left overs from a large customer order. I paid $2 a board delivered so its hard to complain. The only reason you can buy small numbers of these at close to or quite possibly below cost price is because the people selling these are scratching a living by obtaining surplus stock in the electronics markets (physical market places) and relying on the fact that the government in China fully subsidises international postage for "samples", a strategy originally intended to boost international commerce, now widely abused but retained anyway because it brings a lot of USD into the country. They just have to take the package to the state run post office. It's impossible for anyone from another country to compete against the worlds largest electronics manufacturing base and zero shipping costs for the seller.

Johnboy251, 2020-02-27:
Regards
John

Johnboy251, 2020-02-27:
Has any else made the mistake I made with led & key module? I measured the pitch of the led’s & tactile switches a couple of times and convinced myself they were on an 8 mm pitch! - Wrong! The switch are on the ‘industry’ standard grid of 0.1” (2.54mm) so the pitch for the switches & led’s came out to be 0.3” I only noticed this after I had milled the box lid in what I thought was quite an expensive little box! Should have know better - still ‘can’t win ‘em all’!

Regards
John

AndrrewG, 2020-02-27:
I had every intention of making the mistake of assuming that my board was the same as James's one. I did all the CAM for the lid and posted the g code. Then I 3D printed the spacer and realised the holes didn't line up with the buttons.

Johnboy251, 2020-02-27:
Andrew... you did better than me! I didn’t either the LED’s to line or the button actuator holes! 🤭

clough42, 2020-10-12:
This issue spooked me a bit, and that's why I decided to just include the LED&Key board with the control panel kits.  I've purchased more than 700 of these boards from ebay seller "alice1101983" and while there are a couple of different silkscreens on the boards, with different manufacturer logos, the hole, LED and button spacing has all been the same.

I'm going to go ahead and close this issue, unless there's something else that needs to be done about it.

closed 2020-10-12 by clough42