clough42/electronic-leadscrew #30: Software Direction toggle

Greecher, 2019-11-30:
When I got my install done, I noted that because of not going through any gears at all, I think my default direction is backward, and after power on, i have to reverse the direction to get it to feed in.  Possible to add this to the configuration.h file as a flag?

clough42, 2019-11-30:
If you are using the version of code on the head of the master branch,
there is a setting to invert the direction pin in Configuration.h.  I
haven't had time to do the testing I would like before marking it as an
official release, but I'll likely get to that soon.

James

Greecher, 2019-11-30:
umm, i'm an idiot, just set the INVERT_DIRECTION_PIN true free in configuration.h file, and all is good, case closed.

closed 2019-11-30 by Greecher

clough42, 2019-11-30:
Awesome!  Have fun!