clough42/electronic-leadscrew #31: Feed Rate/Threading calibration issue

Greecher, 2019-11-30:
Finished up an install, and noted that the feed rate was way too slow, and when doing some initial testing of threads, double checking my setup several times, ie 1:3 gear ratio, on a 1000 pulse servo, 8 tpi screw (Precision Mathews 1022V), I was for a tpi of 8 (had to use the coarsest pitch to get a match), I was actually getting a 26 tpi.  Simple math fix of 3.25 fixed it for me, and every thread pitch I did in either standard or metric is now spot on with a thread pitch tool check.  My final settings where 1000*9.75 (3 for the gear ratio, 3.75 for the calibration fix).  

Also, during setup, and before I got the servo programming cable, there seemed to be a very odd interplay with what I set for the encoder settings, as it was 4x off (my encoder was a 1K, but i had to set it to 250 to get a reading the same as the rpm already on machine.  This seemed to be because of the feedback of the 1K i was using for my servo (which had the default 4K setting from factory).  When I set it to 1K, then my encoder setting had to be set back to normal (as it was now 4x too high), and all seemed to be good.  

Is there a setting you used for your built in servo encoder, I left it at the 4K factory default, only changing the pulse count from 4K to 1K as you showed in your video.

HarrysWorkshop, 2019-11-30:
Hi there, similar comments.  I have mine bench testing but find that the values in the TI program have little to no effect.  I am running a 2000 pulse/rev incremental encoder (used for my CNC system) to a Automation servo 1000 pulse/rev encoder, 3:1 ratio to an 8 TPI lead screw.  I had to program my the servo drive to 1562,5 pulsed/rev (1600 x 1000/1024) to get a 1:1 ratio of the head stock to lead screw at 8 TPI.  If I use 8000 for my encoder, the speed is 2X, seems OK at 4000.  Bottom line, it sure would be nice to be able to either develop my own reference tables or to have the ability of setting a ratio in the TI program.  Changing the lead screw TPI has no effect.  I have not tried changing the TI servo setting of 1000 but will see what effect that has.

clough42, 2019-11-30:
Are you using the 1.0.03 release code, or did you download the code from
the head of the master branch?

The settings in Configuration.h (on the head of master) should allow you to
configure for your servo and encoder without any kind of special
compensation.

James

Greecher, 2019-11-30:
I am using the electornic-leadscrew-1.0.03, or I guess the experimental release.

Greecher, 2019-11-30:
James,
  I swapped over to the Master branch, but had an initial issue of setting my 1K*3 to the STEPPER_RESOLUTION as code errors due to 2K being max cap.  No big deal, split it 3 to Stepper_Microsteps, 1K to SR.  

  After setting it up for a quick threading pass, I had same initial bad result, 8TPI results in 26TPI, meaning same 3.25 correction factor was needed.  

  In trying to apply the correction to master branch code, ran into more issues where I tried to apply the fractional correction, and it says both Resolution & Microsteps needed to be whole numbers, doh!  In the end, I just fell back to the Experimental leg, and let well enough be.

Dave (aka Greecher).

Greecher, 2019-11-30:
James,
  went back to Master Branch code set, but just removed the sanity check for microsteps.  I think this will cause no issues if the float option is used, so may suggest a change to code set that nests both resolution and microsteps in another if for the float option, or change code that will allow it to validate a float value

clough42, 2019-11-30:
I just merged over all of my development from other branches to master, so
it's up-to-date with the latest, and I just made a new release (1.1.00).

You should be able to configure it properly so you don't have to do any
compensation anywhere.  If you still can't make it come out right with
straightforward configuration, let me know and I'll take a deeper look.

James

Greecher, 2019-12-01:
James,
  Reran new 1.1.0.0 Master branch, set TPI to 8, Microsteps to 3, Stepper Resolution to 1000 (I am using a belt drive 20 tooth on servo, 60 tooth on leadscrew), Encoder Resolution of 4000 (1:1 on 72 tooth belt from spindle, 1000 pulse encoder), left all other items the same.  

  Result, same bad, for 8TPI thread, I'm getting spot on 26, or 3.25 greater.  

Added back in Microstep of 9.75 (3*3.25), dropped microstep sanity check.

  Result, all good, checked several thread pitches including 8, 12, 20, all are spot on.

Side note, I would add a better note on the Encoder resolution, as 4X nominal isn't exactly obvious why.

Thanks for being greatly responsive, my CPP is a bit rusty, but I may try to pitch in to find out where the error is emanating from, simple math tricks of 12/8 for leadscrew, or anything to do with gearing, as I have same 1:3 ratio.

Best regards,
Dave

clough42, 2019-12-01:
What is your leadscrew pitch?

James

clough42, 2019-12-01:
Also, is your RPM readout correct?

James

clough42, 2019-12-01:
I just ran a test on the bench.  Here's my hardware setup:

- 1024 P/R Encoder
- S/D hybrid servo, configured for 1000 steps/revolution

I'm imagining this theoretical lathe (I think the same as yours?):

- 8 TPI leadscrew
- Encoder belt drive 1:1 to spindle
- Servo bent drive 3:1 to leadscrew

Configuration.h settings (file attached): [Configuration.zip](https://github.com/clough42/electronic-leadscrew/files/3907792/Configuration.zip)

- #define LEADSCREW_TPI 8
- #define STEPPER_MICROSTEPS 3
- #define STEPPER_RESOLUTION 1000
- #define ENCODER_RESOLUTION 4096
- #define USE_FLOATING_POINT

Selected mode:

- Imperial threading, 8TPI

Expected outcome:  The leadscrew should turn exactly in sync with the spindle, since I'm cutting an 8TPI thread and the lathe has an 8TPI leadscrew.  Since I have the servo configured for a 3:1 belt drive (STEPPER_MICROSTEPS=3) I would expect one revolution of the encoder to produce three revolutions of the servo.

Actual result:  I see exactly this.  I just marked the pulleys with a Sharpie, and I get what looks like exactly three revolutions of the servo for one revolution of the encoder.

I tried switching off USE_FLOATING_POINT, and that made no difference I could see.  I also tried setting the encoder to 4000 to simulate your 1000 P/R encoder, just to see if that causes some kind of weird interaction somewhere., but it seems to work fine.  There's a little error, since my encoder is actually 1024 P/R, but only the expected tiny fraction.

So I'm still unable to explain the behavior you're seeing.

clough42, 2019-12-01:
A couple of notes on debugging:

If you're getting the wrong ratio at the leadscrew, it could be a problem with the configuration of the encoder, or with the configuration of the stepper/servo output.

You should be able to tell the difference by checking the accuracy of the RPM display.  If the RPM is correct, the encoder is probably configured correctly and the problem is with the output steps.

Related: if you compensate for errors by changing the encoder parameters, it will affect both step ratio and RPM.  If you compensate by changing stepper parameters, it will not affect RPM.

I'm very curious to see where this leads.  It's probably something very simple.  Or more likely two very simple somethings that are interacting, confusing the issue.

Greecher, 2019-12-01:
James,
  I see a match with the rpm guage, i do have an 8 tpi leadscrew, and rest as you state, 1:1 encoder (1000 pulses/rev), exact same servo/driver you have, configured to 1000 pulses/rev (I did ask question about encoder resolution setting for the driver, I left it @ 4000, didn't think it made much difference, ill check that tomorrow setting it to 1000 also), and also do have the 3:1 pulley ratio.

  I also note that the feel of the inch feeds feels right with the correction factor.  I just re-ran my course pitch threads in metric and inch as they seem to tell the story the best, ie easy to read and see the difference, so 5 for metric & 8 tpi, and noted that it was not exactly right, it was in fact 3.3 multiplier instead of the 3.25 i was using, so close to 10, did a quick change to that for a check, but alas no go there, easier to explain 10x diff.   

  I reviewed the specs for the difference in the encoders, and noted that the only difference I saw was in response frequency, the one you have is 100khz, mine is 300khz (maximums) aside from 1024 vs straight 100, but don't think it is that.  That is backed up by me breaking out the tach and checking it manually, turns out to match better than cruddy 4 magnet sensor they put in, so good there.

  I do have the same encoder you use, and could mock up a replacement, but I think that is a dead end with the rpm testing out as expected.  I just reviewed your vid on the gear ratios, and it all looks good, and as the inch feed rates, metric thread & inch thread are all good with the correction, I think it has to be in the data train that calls the stepper, but I could not find an odd thing in that data flow, ie no unexpected multipliers etc.  Guess I'll hope it is the encoder settings on the controller for the stepper.  Could you give me your settings screen snapshot for programming the controller, wanted to see what might be different there.

Thanks Much,
Dave

clough42, 2019-12-01:
Here are the parameters currently on my drive:

[image]

And the motor settings:

[image]

I/O settings:

[image]

I think the only thing I changed was to set the Pulses/Revolution to 1000.  The encoder resolution of 4000 appears right, given the data on the Automation Technologies web site.  They indicate it's a 1000-pulse (4000-count) encoder in the motor.

Greecher, 2019-12-01:
James,
 I'll start off by saying thank you so much for your patience!

Now, the suprise reveal, I'm an idiot.  Wanna know the ratio of using the feed lever over the threading lever for doing threads?   Bet you can guess....  Yeah, sat there and stared at the machine this morning, and thinking there was nothing I could see wrong with your code, my setup etc, and was stuck on idea that it had to be my drive train, checked the pitch of my leadscrew, then had the brilliant DOH! moment.

  So, set it back up as it should be, 3 microsteps, 1000 revs, put sanity code back as it was with microstep check, installed, and re-ran tests again.  100% spot on, checking both inch/metric threads at several pitches.

  So in the end, sorry for wild goose chase, the code is solid, but not against idiots.

Best Regards,
Greecher aka Dave

closed 2019-12-02 by Greecher