clough42/electronic-leadscrew #33: Known good configs in the Wiki?

TheRealGreatOldOne, 2019-12-02:
As title - how about a known good config section, where we can share lathe, leadscrew, encoder and motor configurations that have been implemented and shown to work? Pics of hardware placement as well. 

Might be of use to those starting out, or thinking of moving their current setup to a new lathe.

ElsGear, 2020-08-08:
In one of James video's he asked us to give him lathe metric leadscrew pitch values.
If everyone could give specs on their lathe we could start building a config-table that we could implement.
Something like this:

User | Lathe Model | Spindle Gear Tooth | Encoder  | Leadscrew Gear Tooth | Leadscrew Pitch | Recommended RotEnc Gear Tooth | Recommended Motor Gear Tooth | Optional Compensation value.

* "Leadscrew Gear Tooth" and "Recommended Motor Gear Tooth" is a ratio value, when using belts instead.

I just finished building my ELS and findings video and have yet to figure out ratio's on my lathe.
This could be helpful.
[image]

patl1, 2023-06-12:
Great idea for the wiki page of installations to help others