clough42/electronic-leadscrew #34: Comment Code or Add Default Feed Table variables

Greecher, 2019-12-03:
Noted, i had to hunt around on how default values where set for the feed table (factory), ended up adding this message to the Tables.cpp code above 'FeedTableFactory::FeedTableFactory(void):' line:

// Last section of each factory definition is the default value used when initializing that table,
// zero based count of course.  Values from that point are sticky, ie whatever user set last.
FeedTableFactory::FeedTableFactory(void):

I did modify mine to start with an initial inch feed rate of .01 instead of .005, and was really easy to modify tables to my liking.  Note, my thread rates are actually set around UNC/UNF for  for 1" down to 6 size screw, other entries where insanely small.  For metric, used M3 on up, as the ones below are sooo tiny.  Note I did kill the ones above 3.0 (which are > 1" equivalent size anyhoo), as those where insanely large, and fairly hard on drive mechanism (ie swap over to those rates would cause servo fail mode if you wheren't at zero speed, and even at start up, you would need to be going very slow, ie 50-100 rpm, not to cause a fault) .  

My own preferences.

clough42, 2019-12-06:
I started initially with the feeds and threads that are supported natively
on the Grizzly G0602 lathe.  I limited it to about 8TPI on the big end,
just to try to limit the craziness.  You are correct, though, that you
should not attempt to cut 8TPI threads at 2400RPM.  When things slow down a
bit for me, I plan to add some speed limits to the code to handle
situations like this.

James

patl1, 2023-06-12:
Thanks for posting this information Greecher. I spent an hour trying to sort out where the defaults re set in the code. Would be nice to have this info in the "configurations" page on the wiki

johnsonm, 2023-06-12:
See PR #255 for an implementation that adds additional thread specs, allows an additional digit of precision in thread specs, and lets you choose the default rates from Configuration.h

7severn7, 2024-08-16:
I agree those really coarse pitches should be speed limited. For machining those coarse pitches before the days of ELS I would use a spindle handwheel and use a tool-post milling accessory. Being code illiterate I always thought that was the original reason for the EEPROM on the boost board as well as producing worms which usually need a factor of pi.