clough42/electronic-leadscrew #35: Touch screen controller

broke-again, 2019-12-06:
Is there room (on board and in code) to incorporate an lcd touch-screen control panel ?

clough42, 2019-12-06:
A touchscreen UI would be possible.  The existing interface PCB provides a
level-shifted open-drain SPI bus, so it might be possible to connect a
touchscreen that uses that protocol and write new code in the firmware to
drive it.  If the LCD uses a different protocol, a different connection
would be required.  I'm sure the microcontroller has enough pins, but it
isn't currently accommodated on the PC board.

James

broke-again, 2019-12-08:
Is your installed code editable or is it locked?  Please excuse my ignorance as I've not yet received your product.  Thanks, in advance, for your cooperation

gadjet, 2019-12-08:
How about a Nextion display which uses the serial port and comes in different sizes.

clough42, 2019-12-08:
The code is offered under the MIT license, so you can modify it as you wish.

James

clough42, 2019-12-08:
The nextion display uses separate tx and rx lines.  The microcontroller can
handle it easily, but I haven't provided connections for it on the
interface PCB.

Though I think it uses 3.3v logic, IIRC, so it might not require any
interface circuitry.

gadjet, 2019-12-08:
Yes, from what I can find out the Nextion display has a 5V feed but uses an on-board 3.3V regulator for the logic.

alexphredorg, 2020-02-19:
I'm going to move over to this display from Matrix Orbital:
https://www.matrixorbital.com/glk19264a-7t-1u

There are some cheap ones on eBay (I paid about $25).  It looks easy to drive but I'll need to move over to the UART.  It handles 5V so it's still an easy to do 4 wire interface.  It isn't touchscreen, but in a shop I prefer real buttons to touchscreen.

The full text display will allow me to add some improved threading modes to work on a lathe with no compound.  I'm going to remove the compound on my lathe to get better rigidity and a little more height clearance for my toolpost.