clough42/electronic-leadscrew #37: formula calculation

ana4732, 2019-12-19:
I have see all your videos , I have a point which is not clear to me.
The rotation of the spindle is transferred to the encoder via 2 set of gears :
42 tooth gear (spindle)  >>>  62 tooth (1st shaft) , ratio is  42/62 = 0.6774…. ) and
40 tooth (1st shaft)   >>>  60 tooth  (encoder shaft) , ratio is 40/60 =  0.6666….. 
So the ratio for the gears is not exactly 1:1. Also , stepper motor to leadscrew is  calculated  have  ratio 3:1. I know you have enter all these ratios to your final calculating formula.
In may case is difficult to use 42/62 tooth gear (ratio 0.6774....)  to read the spindle rotation, this means other ratio , Let me know please how can compensate this calculation error ? Need to change any multiplier at configuration file ? 
For example if I use: 60/80 tooth (ratio 0.75)  ( instead of 42/62 ratio=0.6774....... ) for the 1st set gear and 40/60 for the 2nd set the calculations will be not correct. I need a "compensation. What will be the compensated value and where I shall declare it ? 
Thanks for  helping

Greecher, 2019-12-25:
I think the gears he used from spindle down where 40-60-40-60, don’t recall the 42/62 pairing, and when I inspected mine, main was 40 On spindle, the second gear is configurable.

ana4732, 2019-12-26:
thanks for answer.
Shown the videos, the used gear is 42/62. Using 40/60  will produce an error in calculations.  2000 rpm on spindle (40/60), will gives 2032 rpm/min9 (42/62). Which means accordingly will produce an error in theads.  So please confirm what is the right ratio for calculations and if I will not use this ratio , how I can compensate the difference.
thanks

Zymurgy42, 2019-12-27:
In Part 5, Lathe Electronic Leadscrew Part 5: Mounting the Encoder and Servo on the Lathe,  James clearly shows 40 to 60 and 60 back to 40, it's a 1:1 ratio.

Greecher, 2019-12-27:
regardless of what you choose for your encoder gear train, a 1:1 to final is what is spec'ed out, but you can modify it easily by modifying the encoder resolution to fit, ie add in a multiplier, so if it is 2:1 or 1:2, double or half the encoder resolution to suit.

clough42, 2020-01-22:
On my lathe, it's 40:60 * 60:40, resulting in 1:1.

If yours isn't 1:1, you can adjust the encoder resolution to compensate, as
Greecher points out.  The exception would be if you're using prime numbers
or some other ratio that doesn't result in a whole number of counts per
revolution.  That will introduce an accumulating error.

James

closed 2020-10-11 by clough42