clough42/electronic-leadscrew #39: Stuff you will need issues

loeschr, 2019-12-25:
In the project wiki, the list of stuff you need has bad links.  The servo/stepper hybrid motor and the motor driver are dead links.  Could you either post updated links or enough specs on both so I can source them?

Greecher, 2019-12-27:
Here is a link to Automation Technology product page for the exact same one that is in Jame's vids:
https://www.automationtechnologiesinc.com/products-page/hybrid-servo-system/hybrid-servo-drive-kl-5080h/

Greecher

Greecher, 2019-12-27:
Also, here is the copypasta of specs for future reference:
NEMA 23 Hybrid Servo kit (closed-loop stepper system):
one Hybrid Servo Driver KL-5080H, one NEMA23 KL-23-2N-1000 Hybrid Servo Motor with one Encoder 1000 line

A: KL-5080H, 3-phase Hybrid Servo Driver for NEMA23 Hybrid Servo Motor
20-50V, 8.0A Peak, No Tuning, Nulls loss of Synchronization
Closed-loop, eliminates loss of synchronization
Broader operating range – higher torque and higher speed
Reduced motor heating and more efficient
Smooth motion and super-low motor noise
Do not need a high torque margin
No Tuning and always stable
Fast response, no delay and almost no settle time
High torque at starting and low speed, high stiffness at standstill
Lower cost
B: NEMA23 Hybrid Servo Motor, KL23-2N-1000, 1/4" shaft
Rated Torque 282 oz-in, 2 N-m, with one Encoder 1000 line

1000 pulses per revolution

Quadrature digital signals: 4000 steps per revolution

4000 / .05 pitch

80,000 counts per unit

pauly7300, 2020-01-01:
I haven't been able to make a serial connection to the driver.  Some older forums posts I read suggest they have prevented it from working for some reason.  They've told me in an email that I shouldn't have to make a serial connection and that it was set correctly already.  It is working for me however I have to set the Stepper_Microsteps to 8 in the code.  I'm using 2:1 pulley reduction (72-tooth on lead screw and 36-tooth on the motor) and would have expected to set the microsteps to 2 based on your notes in the code.   I've successfully cut 1/4-20 thread and it's accurate so it's working.  Just not sure if that is expected.

Does this seem correct James?

clough42, 2020-01-22:
Sorry for the broken links.  The link I provided is to the Automation
Technology eBay listing, which saves a couple of dollars on shipping.  Not
sure why they changed it, but I put the new link in the wiki page.

I had no trouble connecting to the serial port on the driver using the
64-bit software linked on the data cable page on the Automation Technology
web site here:
https://www.automationtechnologiesinc.com/products-page/nema23-closed-loop-stepper-motor-system-hybrid-servo-kit/calibration-cabel/

If I remember correctly, the driver comes factory-set to 1000 steps per
revolution.  I have a 3:1 belt drive to the leadscrew, so I have my steps
set to 1000 and my microsteps set to 3.

If your ratio is weird, you need to determine if it's your encoder settings
or your stepper settings.  If the RPM display is correct, your encoder is
set up correctly.  If not, start there.

On the stepper side, if the numbers look right, but you're getting the
wrong ratio, check any other gearing that might be in the loop.  On the
Grizzly G0602, there is a stock gearbox with a couple of knobs that can
also change the ratio.  I think setting them to A and 1 results in 1:1
gearing.  Set the knobs, mark the leadscrew with a sharpie and turn by hand
to be sure.

James

RayFriesen, 2020-01-26:
James:  Can you give me a good source for the gears and belts??  I have accumulated all/most of the parts and am about ready to start 'breadboard' assembly and testing.

DavidThijs, 2020-06-19:
I'm also looking for specs of that enclosure. Couldn't find it on the wiki at the first glance.

loeschr, 2020-06-19:
I found a great (and inexpensive) source for just about any kind of pulley you need.  Check out https://us.misumi-ec.com/  You pick the style and then can configure the size, number of teeth, shaft dia, key or set screw, etc.  It takes about a week or so because they are made to order.  I got all my pulleys for this project from them.

pauly7300, 2020-06-21:
> I found a great (and inexpensive) source

This is great to hear.  Can you share a rough pricing with us for comparison sake?

J642, 2020-08-08:
They only trade with company not private

Den søn. 21. jun. 2020 kl. 19.25 skrev Paul Morley <notifications@github.com