clough42/electronic-leadscrew #40: Problem with Circuit Board

RayFriesen, 2019-12-28:
I got the circuit board on the first batch and put the connectors on it and powered it up.  Thought I would, just for the record, measure the voltages and make sure all was OK.  Found the 5v a bit high, 5.37v and measured about 35mv of ripple on both the 5v and the 3.3v.  Somewhat high, so I put the scope on it to take a look; found some strange kind of oscillation that looked like 'ringing' with a pulse of it about every 15 or 20 milli-seconds.  Not large but running about .5-volts at the maximum.  It looked like the regulator was oscillating, but weird even for that.  For this I was using the brand new transformer 5v supply that was listed by the design. 

 Just to be sure, thinking maybe it was somehow a defective board, I went ahead and ordered another board, but in the meantime thought I would do some checking.  I didn't like the 5v being at 5.3 volts, so, I connected up another 'wall-wart' 5v supply I had just to see what it would do.  Lo and behold, no oscillations.  Five-volts was 5.1v, which is exactly what it should be and 3.3 was 3.27v and ripple measured zero -- scope showed 'straight-line' with sensitivity at 5mv/div.

Connected back to original supply and the oscillations returned.  Somehow it seems that the wall-wart was causing the oscillations?? I have never seen that, but II will not use that supply and now I have a spare PC-board that I hope I don't need.  I don't know that the oscillations would have affected the operation of the system, but with low-voltage digital, you never know; half a volt of 'garbage' is pretty significant to a 3.3v signal.

clough42, 2019-12-28:
I'm not sure.  The '1117 voltage regulator can oscillate if the output
capacitance has too low of an impedance, which is why there's a 2.2 ohm
resistor in series with the output bypass cap.  That should put it deep
within the stable region, and I haven't had any trouble.

I have seen 5V wall adapters do all sorts of odd things--especially cell
phone chargers.  I have a bunch of (probably counterfeit) "Samsung"
adapters running esp8266 sensors on my refrigerators, freezers, etc.  They
have 3.3v regulators, but they monitor and report the 5V input, and they
vary all over the place, from about 4.3 to 5.2v.

I would guess you're seeing odd behavior from the 5v adapter.  IIRC, the
dropout on the 3.3v reg is about a volt, so if it's dipping below 4.4 or so
you might see artifacts of that in the 3.3.  I'm just speculating now.

The regulation of the 5V output might improve with load.  I've seen that,
too, but 5.37v seems pretty high.

James

RayFriesen, 2019-12-29:
Actually, the one that is 'behaving' is a cell phone charger.  I cut the cell phone plug off and put on a coaxial plug.. I'm convinced that the problem is the ebay 5v supply; I have never thought much about what is inside one of those 'wall-wart' units.  I guess a transformer and some form of regulator?  Anyway, it seems to be working great now using the cell charger, so I am going to stick with this one.. I think I am going order another 5v supply unit from ebay and see if it has the same problem.  

I have most of the parts, so I am going to proceed on the project. By the way, do you have a source for the pulleys?

You have done a most impressive job on this ELS project and thanks for the speedy reply on my 'quirky' problem..

closed 2019-12-29 by RayFriesen