clough42/electronic-leadscrew #41: Hybrid Stepper not moving properly - SOLUTION

t3kboi, 2019-12-28:
OK, 

I spent way too much time down electrical rabbit holes on this one....

KL-5080H Hybrid Servo Drive / KL23-2N-1000 Nema 23 hybrid stepper

Problem: everything works, but the hybrid stepper only moves a few steps in either direction at very slow speed, and then stops moving.  Holding the stepper frame, you can feel the step pulses continuing, but no more rotation, until re-powering the driver.

Solution: Unplug the stepper encoder from the driver, and plug it into the 6' cable that came with it.  Apparently the pinout from one end of the cable to the other is not straight-through.  Plugging the motor directly into the drive causes the problem.

clough42, 2019-12-28:
Yeah... This catches a lot of people.  I need to put it in a
troubleshooting page

James

t3kboi, 2019-12-29:
Not really a problem with your ELS project - but definitely something that should be included in the text/documentation at Automation Technology....

:-)

pauly7300, 2020-01-01:
I experienced the same prob and wrote them a "passionate" email asking them to mention that the extra cable was NOT an extension cable and was required in order to use.  They have since put up another diagram on the listing but it doesn't directly mention the cable.

themadreefer, 2020-01-04:
> OK,
>
> I spent way too much time down electrical rabbit holes on this one....
>
> KL-5080H Hybrid Servo Drive / KL23-2N-1000 Nema 23 hybrid stepper
>
> Problem: everything works, but the hybrid stepper only [...]

i had the exact same issue with mine...

clough42, 2020-10-12:
I added this to the troubleshooting documentation.  Closing the issue.

closed 2020-10-12 by clough42