clough42/electronic-leadscrew #44: Encoder wiring

oliveat, 2020-01-08:
James, Hi. I have the software and hardware to the test status. There are no wiring details on the wiring sequence for the encoder to the TI board. I am using identical components to your set up. There are 18 possible ways to connect the encoders 3 wires to the  three input pins on the board.  Please enlighten  me to the correct sequence. Thank you, Richard. PS It works as described bar the forward/ reverse does not work.

pauly7300, 2020-01-10:
I created a quick diagram.  I didn't connect the shield to anything as of yet.  I'm not sure if it needs to go to the chassis ground or not.  I've not had any issues so far.

[image]

oliveat, 2020-01-10:
Thank you pauly 7300. This is exactly as I connected the encoder wiring. I had a careful look at James's set up and am fairly sure he also sets the wires the same. I will look at the code to try and determine why the reverse button does not trigger the reverse motion at the servo. I could have a corruption in that area and the DIR pin does not invert. I swapped out the driver board to be sure this was not the issue preventing the reverse problem. Once again, Thank you.

pauly7300, 2020-01-10:
Hi oliveat.  Your welcome.  Is the reverse button a separate issue you're having as that wouldn't/shouldn't have anything to do with the wiring of the encoder.  When you press the button to reverse the direction, does the light change on the panel?  Also I would be careful not to reverse it while the spindle is turning as that will have a very abrupt change in direction and could (at minimum) lead to missing steps.

oliveat, 2020-01-10:
Hi again, yes, the reverse light comes on when I press the reverse button. However the servo does not reverse direction when I spin the encoder cw or ccw. It always spins ccw. That is why I questioned the encoder wiring.

oliveat, 2020-01-11:
Hi, well I have solved the problem. The issue is the encoder on the hybrid stepper. The pins on the cable have been incorrectly placed. I have fixed this and all is well. Thanks again for your help, Richard.

pauly7300, 2020-01-12:
That's good to hear Oliveat.  Do you mind providing just a bit more detail that might help others with that issue?  It sounds like you may be describing [this](https://github.com/clough42/electronic-leadscrew/issues/41) problem that quite a few of us have experienced.  James even mentions it [here](https://youtu.be/W-Lm8HxKwi0?t=1640) in his latest video on making the enclosure.  I had the same problem as described in the the referenced issue above but your description sounded a bit different.

Thanks and glad you got it working.  

One other basic question.  What gear ratio are you using between motor and lead screw and what have you set microsteps to in your project code?  I'm using a 2:1 reduction (72 to 36 tooth - leadscrew to motor) but I ended up having to set microsteps to 8 in the code in order to get the correct travel.  

Here are my exact settings (code comments removed).  And for reference, I have purchased exactly the same hybrid stepper that James has in the vid.  And my lathe is the same as his as well.

// For Imperial leadscrews: pitch in Threads Per Inch (TPI)
#define LEADSCREW_TPI 12

// Steps and microsteps
#define STEPPER_MICROSTEPS 8
#define STEPPER_RESOLUTION 1000

#define INVERT_ENABLE_PIN true
#define INVERT_ALARM_PIN true

// Enable servo alarm feedback
#define USE_ALARM_PIN

#define ENCODER_RESOLUTION 4096

// Use floating-point math for gear ratios
#define USE_FLOATING_POINT

oliveat, 2020-02-03:
Well, the saga is over and I have cut a beautiful 1.5 pitch 10mm diameter thread. The final problem was solved when I read the driver board spec sheet. I had set nothing on the driver bar the steps per rev, however, there was a setting from the factory that turned off Step/Direction and the driver was enabled for cont rotation ccw!  Once I figured that, all was well. The problem with the wiring was I had mixed a Stepperone driver with another brand hybrid stepper. The A channel and B channel were back to front wrt the driver board. I swapped them over and bingo, it all worked. To answer your question, the ratio is 2: 1 as is yours. The driver is set to 16 and the config:h set to 8 as I recall.
As to hardware the spindle has a 40 M1.5 tooth pinion so I made a 40 tooth 1.5 gear out of polypropylene from a domestic breadboard and drive it via a metal idler on the banjo of the lathe with a suitable bushing and spacer pined to the fabricated gear to align the gear set. I bent up a s.steel bracket from !.5 mm material and attached the Encoder to the banjo using one of the banjo T nuts. 
The hybrid stepper is nested in the space under the spindle with a s.steel bracket bent to the correct offset to align the pulleys and secured with M4 screws The belt is T5/16 220mm. The bracket for the stepper has slotted attachments so it can be moved laterally  to adjust belt tension. The installation feels vey secure and is quiet running.  Regards, Richard.

oliveat, 2020-02-03:
[image]

oliveat, 2020-02-03:
[image]

oliveat, 2020-02-03:
[image]

oliveat, 2020-02-03:
[image]

pikolo335, 2020-04-16:
> Hi, well I have solved the problem. The issue is the encoder on the hybrid stepper. The pins on the cable have been incorrectly placed. I have fixed this and all is well. Thanks again for your help, Richard.

hi there i have the excact same problem ...what did you mean you fixed it? the motor only moves cw  ...when i move the encoder anti clockwise  the stepper doesnt move and the servo moves freely.

JonWoellhaf, 2020-04-16:
Pikolo335,

I suggest you close the “Stepper does not move” issue #62 and open a new issue.

Do I understand correctly that the servo turns when you run the encoder in one direction, but the servo does not turn when you run the encoder in the opposite direction?

Jan

From: pikolo335 
Sent: April 16, 2020 09:39
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Encoder wiring (#44)

  Hi, well I have solved the problem. The issue is the encoder on the hybrid stepper. The pins on the cable have been incorrectly placed. I have fixed this and all is well. Thanks again for your help, Richard.

hi there i have the excact same problem ...what did you mean you fixed it? the motor only moves cw ...when i move the encoder anti clockwise the stepper doesnt move and the servo moves freely.

pikolo335, 2020-04-16:
Should i make a new issue or continue writing here? Yes this is the case it moves on one direction only.When i had connected the motr to my cnc with the diagram that my cnc had it was rotating both ways  if this helps.

closed 2020-10-11 by clough42