clough42/electronic-leadscrew #45: Connector on front of display board interferes with mounting in box

MrGadget49, 2020-01-11:
The pin headers on the front of the display board will not allow for a connector to be attached when the display is mounted inside the enclosure.

From the video where James is engraving and mounting the box, it looks to me like he has moved the connector to the back of the display board.  

I noticed in one scene, the display board he is using on his breadboard has the connector on the top.  So it seems, that he has the same type of board that I do.

Jody

Zymurgy42, 2020-01-11:
I'm pretty sure you are correct in that James just moved it from front to rear.
[image]

Gordon.

clough42, 2020-01-13:
You are correct.  I just moved it from the front to the rear.  I thought
perhaps I had mentioned that at one point, but perhaps I didn't.

James

MrGadget49, 2020-01-15:
Thanks for the replies.   It may have been mentioned or shown in one of the videos, but it's been a long time so I forget.

I don't know what they were thinking when they put this connector on the front of the board like that.  I can't really think of any application where that would work.  (Except for unenclosed on a breadboard)

I guess I'll try moving it.

Thanks again,

Jody

gusosborne, 2020-01-30:
Probably too late but just wanted to mention: Be very careful if you don’t have hot air option. I trashed one of the LED & Key boards this way, just being lazy. The solder pads on mine seemed to be much thinner / weaker than usual. Luckily, I had two of them.

MrGadget49, 2020-02-01:
I managed to get mine moved to the back.

I used the method James describes in the "Trouble Shooting" section.  I heated each pin individually and pulled them straight out.  Then I soldered the right angle connector on from the back.

I didn't know at the time, that James had included an extra right angle connector with his boost kit, so I bought some.  It's just as well, because when I bought my display boards, I didn't know that they came in lots of two, so I ordered 3 of them and ended up with 6 boards!  I guess I'll have plenty to play with.

Jody

closed 2020-02-01 by MrGadget49

JonWoellhaf, 2020-02-02:
Yeah, their two each ploy tricked me, too! But I only ended up with four. 

From: MrGadget49 
Sent: February 1, 2020 16:27
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Connector on front of display board interferes with mounting in box (#45)

I managed to get mine moved to the back.

I used the method James describes in the "Trouble Shooting" section. I heated each pin individually and pulled them straight out. Then I soldered the right angle connector on from the back.

I didn't know at the time, that James had included an extra right angle connector with his boost kit, so I bought some. It's just as well, because when I bought my display boards, I didn't know that they came in lots of two, so I ordered 3 of them and ended up with 6 boards! I guess I'll have plenty to play with.

Jody