clough42/electronic-leadscrew #47: Control panel update frequency -- question

JonWoellhaf, 2020-01-29:
With version 1.1.02 running in Release mode, When I measure the time pin 75 of J8 stays high, I get about 1.5 ms. That should be the time required to update the panel.

The pulse's repetition period is about 17 milliseconds, but I expected it to be about 11 ms.

It appears line 149 {DELAY_US(1000000 / UI_REFRESH_RATE_HZ);} in Main is not working as expected.

UI_REFRESH_RATE_HZ is 100, so the delay should be 10,000 us, or 10 ms.

What am I missing?

JonWoellhaf, 2020-01-29:
I find that changing 1000000 to 550000 in the code above puts the loop close to 10 ms.

But that's an ugly kludge.

Pedantic _is_ my middle name, but I'm just trying to understand the firmware, and this has me stumped.

clough42, 2020-01-29:
I'm not looking at the code at the moment, but I believe that is a delay,
not a period.  So other stuff done in the loop takes time and slows it down.

The timing of the UI loop isn't critical.

James

JonWoellhaf, 2020-01-29:
Hi, James.

Yes, it's a delay. I'm just curious why the delay isn't 10 ms, as it appears it should be.

This is a TREMENDOUS project! Very, very interesting. I can't imagine how you have made time to accomplish it!

Johnboy251, 2020-01-29:
James & Jon
I’d like to second that comment about this project, I’ve been following this from the start. 
One thing that I found is that my TI LaunchPad was supplied with the S9 set as a default to CAN Route. Which I overlooked on initial installation. Otherwise it would have run first time after download the code.

Great job James! 

Jon thanks for your input - I’ll add this function to my setup. 
John 🇬🇧

JonWoellhaf, 2020-01-29:
My bad.

DELAY_US() uses a simple assembly loop, and I was not accounting for the time required to service 2000 interrupts every 10 ms.

With interrupts disabled, DELAY_US(10000) is 10.0 ms as measured on my scope.

Sorry to have bothered you all. :)

closed 2020-01-29 by JonWoellhaf