clough42/electronic-leadscrew #50: SUCCESS!!  Well, with a problem -- help

RayFriesen, 2020-02-06:
OK.  I have the ELS built and working, well essentially working, on the workbench.  Basically using James' configuration with the servo; and it all works -- almost like it should.  My servo is turning only about 1/8th proper speed.  I am using the motor and controller package from Automation Technologies and here is what is happening:
With the ELS set for 12 threads/inch, if you rotate the encoder one revolution, by my thinking the servo should rotate 3 revolutions (assuming a 3:1 gearing ration, servo to Lead Screw); it only rotates about 1/4 of a revolution.  The servo speed tracks as you vary the settings mostly as it should but just too slow
Am I missing some adaptation parameter?  Or do I need to go into the servo driver and change something?  I do have the programming adapter and I can see the motor on my computer; I have not changed any parameters on the motor/driver.

Anyone???

clough42, 2020-02-07:
Did you look through the troubleshooting steps here?

https://github.com/clough42/electronic-leadscrew/wiki/Troubleshooting#my-leadscrew-isnt-turning-at-the-right-ratio-or-my-threads-are-the-wrong-pitch

Make sure your motor has a 1000-step encoder and the servo driver is set to
1000 steps/rev.  I think that is the default, but you never know.

James

RayFriesen, 2020-02-07:
OK... I think I am gaining ground.. Here is what I have:
I have set the servo controller (using the driver S/W and RS-232 Link) to: 
Pulses/Rev = 1000, Encoder Res = 4000 --this is the servo-encoder.
In the Configuration.h:
I set STEPPER_MICROSTEPS 3
and: STEPPER_RESOLUTION 1000
ENCODER_RESOLUTION 4096 (The encoder is the 1024/quadrature-this is the spindle encoder.)
Then I loaded the code again to the LaunchPad with the following results:
With the ELS set to Thread/12 tpi and driving the spindle encoder with a drill, I measured 250rpm on the drill with a laser tach,  and the rpm display on the ELS is 250, so they agree.
With the drill running at about 300 rpm, the servo measures about 900 rpm which would be correct speed for the setting assuming a 3:1 gearing ration to the lead screw.
At this point, all seems OK.  But, if I try to set the servo encoder resolution to 1000, which I think is what you recommended, I get an immediate error and the servo stops dead.  Also, if I run the ELS using the little Arduino simulator, the servo runs at about twice what it should.  Not sure about that, but since the 'drill-method' seems to produce the correct result, I am going to put that down to some 'quirk' of the simulator.

RayFriesen, 2020-02-07:
didn't really mean to close this...

clough42, 2020-02-07:
If the servo stops, it should flash a code on the red LED to say why.

RayFriesen, 2020-02-07:
Flashes seven times.  Pretty sure that's what it is -- it flashes pretty fast.. I haven't had a chance to look in the documentation to see what that is.  When it gets that error, you have to unplug it and plug it back in.  And if you don't change it back to 4000, first time you move the encoder it does it again.

TheRealGreatOldOne, 2020-02-09:
https://www.hobby-machinist.com/threads/electronic-lead-screw.76101/page-4#post-696946

Also check the status of the dip switches on the controller.

closed 2020-10-11 by clough42