clough42/electronic-leadscrew #51: Feature Request: Simple dividing head readout

carneeki, 2020-02-08:
Hi James et al,

I am gauging interest for a feature idea. Would we also want the control board to have a 'dividing head' mode? Something where, suppose a brake is fitted to the spindle, one can rotate the spindle around a certain number of encoder counts for simple divisions and engage the brake. Useful for simple divisions of a circle for cross drilling holes, or slotting. If one were desperate, they could also use it for involute cutting a gear using the carriage like a shaper, though I question how accurate this would really be.

I am willing to work on writing this. I figure, given there is already an encoder on the spindle, and a spare button or two on the control board, perhaps this is doable.

If the interest is not all that high, I'll just build put some optocouplers in as buffers and fanout the encoder to a second board for my own DRO project.

gusosborne, 2020-02-11:
I think that’s a great feature request - I would use it.

danek241, 2020-07-08:
Hi James 
I'm a happy owner of myford ml7 lathe.
I'm currently trying to implement a electronic lead screw, thanks to You. 
I'm also interested in this option. 
I was looking to get myford dividing attachment, however it is expensive and rare item for sale.
If it is possible to have dividing function, it would be just great! It would make my hobby project even easier.
At the moment I use one of the change gears attached to the other end of the lathe spindle and locking pin engage to it.

Best Regards 
Daniel

johnsonm, 2023-04-05:
See #231 which includes this feature.