clough42/electronic-leadscrew #52: Software installed and working 👍

Johnboy251, 2020-02-13:
James 
Software installed and encoder & hybrid stepper working ok. The next phase is to get the hardware mounted on the lathe using suitably designed brackets/mountings. I will be using this on a metric lead screw with a 3mm pitch.

I’ve incorporated the Jon Woellhaf fast scroll/ auto increments mods (#48) to the code which works well. 

Great project...

Best regards 
John 🇬🇧

JonWoellhaf, 2020-02-15:
John,

I'm delighted you had the courage to try my button hack. :) Glad it's working for you!

Jan (Yan)

Johnboy251, 2020-02-15:
Jan & James...

Yes you fast scroll is a really useful addition 👍
I’m looking at putting the setup that has been selected into a non volatile (flash?) memory location so when power is turned off the configuration is not lost. This would be helpful if Power was lost through a ‘brown out’ or power cut.
My coding skills are a bit to be desired to say the least but I added extra trailing zeros in the tables and got them to work ok.

Thanks for the email.

Regards
John

closed 2020-02-27 by Johnboy251