clough42/electronic-leadscrew #53: Hi Mr Clough42

J642, 2020-02-15:
I am mounting the display with bottoms for operations. Then I see the arrows keys are turned 180 degrade so the arrow up (plus) is mounted in the position off the arrow down(minus. I  did not put names on.
I have looked in the configuration code but I am not a programmer . I can not find the place where s6 and s8 is mentioned. Is there a easy way to change  this. Thank you for your fantastic work. I have seen all your videos  many times. Jorgen Denmark

JonWoellhaf, 2020-02-15:
Hi, Jorgen.

It is easy to exchange the Up and Down buttons.

In ControlPanel.h, make this change

struct KEY_BITS
{
    Uint16 DOWN:1;  // was UP:1;
    Uint16 reserved1:1;
    Uint16 UP:1;    // was DOWN:1;
    Uint16 IN_MM:1;
    Uint16 FEED_THREAD:1;
    Uint16 FWD_REV:1;
    Uint16 SET:1;
    Uint16 POWER:1;
};

But if I understand you correctly, with the LED&KEY board mounted upside 
down, the seven-segment displays will be wrong.

Held og lykke!

Jan (Yan)

From: J642
Sent: February 15, 2020 12:31
To: clough42/electronic-leadscrew
Cc: Subscribed
Subject: [clough42/electronic-leadscrew] Hi Mr Clough42 (#53)

I am mounting the display with bottoms for operations. Then I see the arrows 
keys are turned 180 degrade so the arrow up (plus) is mounted in the 
position off the arrow down(minus. I did not put names on.
I have looked in the configuration code but I am not a programmer . I can 
not find the place where s6 and s8 is mentioned. Is there a easy way to 
change this. Thank you for your fantastic work. I have seen all your videos 
many times. Jorgen Denmark

Johnboy251, 2020-02-15:
James....

I think Jorgen  Is trying to say he’s “mounting the buttons for operation“ 
You just beat me to it to reply to his request but it was got to find where the code for the buttons is located.
I’m learn a lot looking at your code - thanks 👍

Regards
John🇬🇧

Johnboy251, 2020-02-15:
Sorry. Jan 😳

Regards
John

J642, 2020-02-15:
[image: IMG_20200213_192448.jpg]
As you can see. I have to buy so  a new box, dril and saws it.  AS you can
see it is label with Dymo  so I can call them as I want

Den lør. 15. feb. 2020 kl. 22.33 skrev Johnboy251 <notifications@github.com

Johnboy251, 2020-02-15:
Jorgen

I also need a new box as well. I convinced myself that the dimensions for the switches and LED were an 8.0mm pitch but after I milled the lid I quickly realised it was a 0.3 inch pitch! 😳

Regards
John

closed 2020-10-11 by clough42