clough42/electronic-leadscrew #55: Servo/Stepper wire diagram to the KL-5080H

jscottmoss, 2020-02-26:
I know there are documented on the automation web page found here 
https://www.automationtechnologiesinc.com/products-page/hybrid-servo-system/hybrid-servo-drive-kl-5080h/

From the web page:
U to Brown
V to Blue
W to Black

My wires are not the same they are the following colors, 
Black, Blue, Yellow, and a RED wire. the Red wire was cut flush to the end of the cable.
I have tried
W to Black
V to  Blue
U to Yellow

When I try and test my rig, using the programming cable thru a USB cable no matter what configuration I config the test with, it only moves slightly, then does absolutely nothing the fault light on the driver lights up red, and an Alarms "Position Following Alarm".
Can anyone shed light on this? From other posts I noticed that I need to rewire my encoder connection to the rocket board. I'll do that once I am done at work, and send an update in a few hours.

Johnboy251, 2020-02-26:
Hi... I’m wonder as you have three wire plus the cut back one if this is a pole stepper motor. If you’ve got a multi meter you can check winding resistances. If it is a three phase motor any combination of three ‘long’ wire should be pretty close in resistance. If there is only one pair with a resitting value and the other one of three is open circuit - try measuring the resistance across to the cut wire. If that has a resistance then we could suspect to be a two pole stepper and not a 3 phase motor. I hope that makes sense😳 John 🇬🇧

jscottmoss, 2020-03-03:
I was able to finally test the 3 longer wires, and its a 3 phase motor, which is a KL23-2N-1000. 
I corrected my wiring problem with  smaller rotary encoder, and it now show RPMs when it is spun, however the other drive the KL23-2N-1000 does not sync up with its little brother. it does nothing. Any pointers on that part?

Johnboy251, 2020-03-03:
What is the wiring for the control connections to the hybrid drive module can you describe it preferably with a photo? I won’t speculate until I get an idea of hows it’s connected.
John

Scubamachi, 2020-05-12:
John,
Just as an FYI, the KL23-2N-1000 listed above is the same motor with the 1.2 degree per pulse that we were discussing on my threads (69 & 70).