clough42/electronic-leadscrew #56: Interface PCB available?

7b-w, 2020-03-04:
Just looked at your eBay page in regards to the "boost" PCB (?) or as I call it interface PCB and you are still out of stock. So are there any more on the way or better yet (for me that is) I sure could use a schematic of this PCB and where it connects to the TI board but  I have not found any. If I had that I could create my own PCB to my own tastes but I do like the way it plugs into the TI PCB. I can probably find a drawing that shows the dimensions of those 2 connectors on the TI board so I can do the same.
So are PCB's available now or where can I get a drawing/schematic of the same PCB?
Thanks for a great project!
Bruce

7b-w, 2020-03-04:
Ok so I started looking at a video posted (#16?) and found a different eBay link to that PCB and was lucky enough to purchase one. So is there a schematic available for troubleshooting if somehow I screw up assembly?
Thanks

JonWoellhaf, 2020-03-05:
The schematic is available on CircuitMaker. It's a free download.
Once you have an acount and are logged in, click Home, then click Projects.
Enter Clough in the Search box and hit Enter.
Click Electronic Leadscrew V2 Booster Pack, then click Open Design.
Finally, on the left, double click ELS_Boost.SchDoc. Wa-la!

JonWoellhaf, 2020-03-05:
Yes, there is a schematic. Did you see my post on GitHub?

From: 7b-w 
Sent: March 4, 2020 12:42
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Interface PCB available? (#56)

Ok so I started looking at a video posted (#16?) and found a different eBay link to that PCB and was lucky enough to purchase one. So is there a schematic available for troubleshooting if somehow I screw up assembly?
Thanks

7b-w, 2020-03-05:
Yes, got the schematic thru Circuit Maker, Thanks.

closed 2020-03-05 by 7b-w