clough42/electronic-leadscrew #57: defect voltage regulator?

carbon58, 2020-03-15:
Hi James, 

today I started soldering the parts on the PCB (following your video no. 12). First i soldered the power connector and motor connector and checked for 5V and 3.3V. 
I am measuring 12.2V at the 5V connection, and 11.3V at the 3.3V connection. My powersupply supplies 12.2V. The PCB starts to get warm after a few seconds.

I have some basics in electronics but i am surely not experienced as you. I cant see anything on the board, so I think the voltage regulator is defect but I also know that you check all of your boards. On the pictures you can see the PCB. Is there anything I can do or look for? 

thanks again for sharing your project :)

regards, 
Ilhan

[image]
[image]
[image]
[image]
[image]

JonWoellhaf, 2020-03-15:
Hi, Ilhan.

I believe you are supposed to be supplying 5 volts to the board, not 12.

Sorry, you have probably damaged some of the components. Hope not!

Jan (Yan)

From: carbon58 
Sent: March 15, 2020 17:17
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: [clough42/electronic-leadscrew] defect voltage regulator? (#57)

Hi James,

today I started soldering the parts on the PCB (following your video no. 12). First i soldered the power connector and motor connector and checked for 5V and 3.3V.
I am measuring 12.2V at the 5V connection, and 11.3V at the 3.3V connection. My powersupply supplies 12.2V. The PCB starts to get warm after a few seconds.

I have some basics in electronics but i am surely not experienced as you. I cant see anything on the board, so I think the voltage regulator is defect but I also know that you check all of your boards. On the pictures you can see the PCB. Is there anything I can do or look for?

thanks again for sharing your project :)

regards,
Ilhan

carbon58, 2020-03-16:
Hi Yan,

you are totally right. Don't know why I used 12 V....

JonWoellhaf, 2020-03-16:
I checked the specs for several 3.3 volt regulators and all allowed more than 12 volts in. So 12 shouldn’t have damaged the regulator. I agree. It looks like it’s defective.

From: carbon58 
Sent: March 15, 2020 18:01
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] defect voltage regulator? (#57)

Hi Yan,

you are totally right. Don't know why I used 12 V....

carbon58, 2020-03-16:
Hi Jon, 
thank you for the hint. 
I checked the datasheet for AMS1117. Here is a link:
 http://www.advanced-monolithic.com/pdf/ds1117.pdf

On page two, they write that the maximum input is 15V. 
I don't know how to interpret the information for line regulation and load regulation in the table on page 2. Can someone explain it to me?

Is it a good idea to put a new voltage regulator on the PCB? I never soldered that small parts with an iron. Does anyone have an advise? 
Should I also check the micro controller before i try to repair it? How can I check it? 

One more question about the amps: in the datasheet i can read something about 1A and 0.8A as maximum. Does this also refer to the power supply I should use? 
The advise from James is DC 5V and >1A . I am a bit confused on this.

thank you Yan and Jon for the quick response

regards,
Ilhan

JonWoellhaf, 2020-03-16:
Ilhan,

You can ignore the line and load regulation figures. They indicate how well the regulator maintains 3.3 volts when the input voltage varies or the load current varies.

I believe you will not be able to replace the regulator without proper equipment and skill.

I suggest you return the board to James Clough and ask for a replacement. He will undoubtedly be interested in determining the problem with the board.

You haven’t plugged the board into the TI LaunchXL board, have you? As I recall, you hadn’t yet soldered those connectors.

Don’t worry about the amps. Think of it this way: if you have a 12 volt bulb and power it from a source that supplies 12 volts at up to 1000 amps, all will be well. The bulb will only draw as much current as it wants, no matter how much is available.

So when Clough recommends a 5 volt source that can supply more than 1 amp, everything will be fine if you use a 5 volt source that can supply 100 amps. Only if the source cannot supply sufficient current will you have problems.

Hope this makes sense.

Jan (Yan)

PS: Jan, Jon, and Yan are the same person. I thought long ago that changing my name would avoid confusion. I was wrong. 

From: carbon58 
Sent: March 16, 2020 03:58
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] defect voltage regulator? (#57)

Hi Jon,
thank you for the hint.
I checked the datasheet for AMS1117. Here is a link:
http://www.advanced-monolithic.com/pdf/ds1117.pdf

On page two, they write that the maximum input is 15V.
I don't know how to interpret the information for line regulation and load regulation in the table on page 2. Can someone explain it to me?

Is it a good idea to put a new voltage regulator on the PCB? I never soldered that small parts with an iron. Does anyone have an advise?
Should I also check the micro controller before i try to repair it? How can I check it?

One more question about the amps: in the datasheet i can read something about 1A and 0.8A as maximum. Does this also refer to the power supply I should use?
The advise from James is DC 5V and >1A . I am a bit confused on this.

thank you Yan and Jon for the quick response

regards,
Ilhan

carbon58, 2020-03-16:
Hi Jan :)           (after trying all 3, I will stay with Jan)

thanks again for your support. I will message james and ask what to do and if he is interested in this board. If he's not interested, i will try to learn how to repair.

No, luckily i didn't put the board on to the LaunchXL. I only soldered the power supply and the motor connector and then checked the voltage (as shown in the pictures). 

Yes, your thoughts on current are helping me. I "feeled" that way but wasn't sure if it is correct.

regards,
Ilhan

carbon58, 2020-03-20:
Hi everybody, 

here's a short update:
after I messaged James, he was so nice to replace the board. 
I couldn't stop thinking about my mistake, using a 12V supply and if that was the problem. So I ordered several AMS1117 3.3 and wanted to replace the old one. 
Since it is my first time soldering those little things, I was a bit struggeling but everything went well.

then I ran the board with 5.2V for more than 20 minutes and measured if I the voltage regulator gives me 3.3V over the time. Then I increased the voltage to 12.2V for more than 5 minutes and also measured the output from the voltage regulator. Still 3.3V.
I increased again up to 14V for a bit more than 2 minutes and the output was still at 3.3V.

[image]

Now, I would like to check if the eeprom is still working without putting it on the LaunchXL. Can anyone give me an advise?

regards,
Ilhan

clough42, 2020-10-11:
I don't know of any way to test the EEPROM without the microcontroller.  The firmware for my test jig is in the source repository, but that firmware runs on a microcontroller (and it requires a special adapter PCB, so it won't be of much use.

At this time, the ELS firmware doesn't actually use the EEPROM, so even if it isn't working, it shouldn't be an issue.  (Unless it's a smoking crater in the board.  :)

closed 2020-10-11 by clough42