clough42/electronic-leadscrew #58: Gear Ratio

Vrossi050, 2020-03-18:
Mr Clough,
I have Grizzly G9972 lathe. I have been trying to get the ratio for threading right. So far no luck.The Configiration.h file reads:
Encoder uses 1:1 ratio from the spindle(45T to Idler 45T) to 60T stacked on the idler and 60T on the encoder. Resolution set to 4096
Servo motor same as yours 1000 P/R  
Servo motor 24T to 72T L/S 3:1 ratio
Stepper Resolution 1000
Microsteps 3
What am I missing? Any help would be greatly appreciated.

Thank you,
Vrossi8050

JonWoellhaf, 2020-03-19:
When you lhread 8 TPI, what TPI do you get?

Vrossi050, 2020-03-19:
I was trying 11 TPI. The leadscrew is 8 TPI. What the result is looks 32 TPI or smaller

JonWoellhaf, 2020-03-19:
Please try 8 TPI, because that should be 1:1 spindle to leadscrew, since you have an 8 TPI leadscrew. See if you get 24 TPI on the work.

From: Vrossi050 
Sent: March 19, 2020 13:04
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

I was trying 11 TPI. The leadscrew is 8 TPI. What the result is looks 32 TPI or smaller

Vrossi050, 2020-03-19:
I will give it a try over the weekend. Thank you

Johnboy251, 2020-03-19:
Hi Vrossi8050

Are you using a hybrid stepper or servo,to drive lead screw?
Regards 
John

Johnboy251, 2020-03-19:
I’ve read you initial message - you mention the you have the same motor as James. James is using a hybrid servo not a stepper.

You need to set the Mictosteps for that motor to 1 

Take a look at James’s YouTube video ELS part 10 about 19 minutes in. James explain things there.

John

Johnboy251, 2020-03-19:
Hi again...

If your try for 11 TPI with your configuration/setup you would get 33 TPI with your lead screw. ✔️

JonWoellhaf, 2020-03-19:
Hi, Johnboy.

I agree the microsteps need to be set to 1, but only if the hybrid servo is 
driving the leadscrew directly.

In his configuration.h comments, James writes, “NOTE: If you are using a 
servo drive with something other than a 1:1 drive ratio, you can use the 
STEPPER_MICROSTEPS to compensate. For example, if you have a servo 
configured for 1000 steps/revolution and it drives the leadscrew through a 
3:1 reduction, you can set STEPPER_RESOLUTION to 1000 and STEPPER_MICROSTEPS 
to 3."

In his Part 5 video starting at 28:52, James mounts a 24 tooth pulley on the 
hybrid servo and a 72 tooth pulley on the gearbox input shaft. I think I 
remember him saying he set the gearbox to 1:1. So he's using the 3:1 
reduction mentioned above, and I assume has set his microsteps to 3, even 
though in his Part 10 video, which you referenced, James sets it to 1.

I'm confused. What do you think?

It seems to me Vrossi050 is driving his leadscrew directly, but has his 
microsteps set to 3.

From: Johnboy251
Sent: March 19, 2020 15:31
To: clough42/electronic-leadscrew
Cc: JonWoellhaf ; Comment
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

I’ve read you initial message - you mention the you have the same motor as 
James. James is using a hybrid servo not a stepper.

You need to set the Mictosteps for that motor to 1

Take a look at James’s YouTube video ELS part 10 about 19 minutes in. James 
explain things there.

John

Vrossi050, 2020-03-19:
Hi all,
Thank you for your help on this.
I have configuartion.h file set to :
3 microsteps
1000 P/R.
The as mentioned above yes I have a 3:1 ratio as James shows 24T on Hybrid servo motor and the lead screw input is 72T . I wonder if the hardware version matters or?

JonWoellhaf, 2020-03-19:
You have the gearbox set to 1:1 input shaft to leadscrew, right?

From: Vrossi050 
Sent: March 19, 2020 17:22
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

Hi all,
Thank you for your help on this.
I have configuartion.h file set to :
3 microsteps
1000 P/R.
The as mentioned above yes I have a 3:1 ratio as James shows 24T on Hybrid servo motor and the lead screw input is 72T . I wonder if the hardware version matters or?

Vrossi050, 2020-03-20:
Yes the gearbox is set 1:1.

JonWoellhaf, 2020-03-20:
I have another thought.

When you changed the values in configuration.h, and downloaded the program to the LAUNCHXL board, are you sure you were in Release mode?

From: Vrossi050 
Sent: March 19, 2020 17:22
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

Hi all,
Thank you for your help on this.
I have configuartion.h file set to :
3 microsteps
1000 P/R.
The as mentioned above yes I have a 3:1 ratio as James shows 24T on Hybrid servo motor and the lead screw input is 72T . I wonder if the hardware version matters or?

Johnboy251, 2020-03-20:
Hi Jan & Vrossio50

I thought I had it - I’m not sure now.😳 I hope James can through some light on what’s happening.

Hope you’re all keeping well.
John

Johnboy251, 2020-03-20:
Jan & Vrossio50

Just looked at James YouTube video ELS Part 11  about 15 minutes in...
I’m not sure if that helps.

JonWoellhaf, 2020-03-20:
Did you see my question about building in Release mode?

Vrossi050, 2020-03-20:
Yes I flashed the Release to the board.

Vrossi050, 2020-03-21:
I reflashed the board and no changes in thread until I change the gearbox dials A=1 and B selected. 
[image]

Johnboy251, 2020-03-21:
[image]

You encoder gear train looks like this?
John

Vrossi050, 2020-03-21:
No just 60 to 60T. The 45 sits behind the 60
[image]

Johnboy251, 2020-03-21:
I see the pcd of the gears is different but I can’t see how it’s driven. 😳

Vrossi050, 2020-03-21:
Next to the black 45t is the spindle gear also 45t but larger in diameter.

Vrossi050, 2020-03-21:
[image]

Johnboy251, 2020-03-21:
Forgive me if I’m wrong -

I have configuartion.h file set to :
3 microsteps
1000 P/R.
The as mentioned above yes I have a 3:1 ratio as James shows 24T on Hybrid servo motor and the lead screw input is 72T

I still think microsteps should be set to 1 not 3
Could you try it for me?

Johnboy251, 2020-03-21:
Gears on encoder - that’s the same effectively as my diagram giving you as you say 1:1 ✔️👍

Johnboy251, 2020-03-21:
I’m turning in now for the night... let us know how you get on with this. 🤔👍

Vrossi050, 2020-03-21:
Sure.

Vrossi050, 2020-03-22:
I tried 1 microsteps and puts me back to where I started. Maybe cut resolution in half on the encoder or increase steps to 4. We shall see tomorrow. Any thoughts on this?

Johnboy251, 2020-03-22:
Hi Vrossi050
I’ll read though the messages again and some of James’s video again when I get a chance. Let’s see what we can achieve. 👍
Regards, ConfusedJohn.😳

JonWoellhaf, 2020-03-24:
Vrossi050, I have a new thought.

Can you verify the RPM display is accurate?

Vrossi050, 2020-03-24:
Hi Johnboy251,
I just tested the RPM with a tachometer on the chuck. The ELS display and tachometer match. I marked all the gears and pulleys and they are also 1:1 except for the 24t to 72T servo to Leadscrew which is 3:1. Any thoughts?

JonWoellhaf, 2020-03-25:
Have you tried cutting 8 TPI?

Vrossi050, 2020-03-25:
Yes see above photo. Set on 8 TPI but it looks like 16 TPI.

JonWoellhaf, 2020-03-25:
Thanks, I missed that photo.
So to summarize,
    11 TPI looks like 32 or 33
    8 TPI looks like 16
    tach readout is correct.
Without changing anything in the code, please try cutting 8 TPI (again), 11 TPI (again), 24 TPI, and 32 TPI.
This is a very interesting problem, but I imagine it's quite frustrating for you. Sorry. :)

Vrossi050, 2020-03-25:
It is quite the challenge. Thanks for your help.

Vrossi050, 2020-03-25:
with the gear box set at 1 to 1?

JonWoellhaf, 2020-03-25:
Yes! Always.

From: Vrossi050 
Sent: March 24, 2020 19:04
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

with the gear box set at 1 to 1?

Vrossi050, 2020-03-25:
8 TPI 
[image]
[image]

Vrossi050, 2020-03-25:
[image]

JonWoellhaf, 2020-03-25:
How many teeth on gear a and gear b?

From: Vrossi050 
Sent: March 24, 2020 19:25
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

JonWoellhaf, 2020-03-25:
Please mark a reference on the spindle and leadscrew, then rotate the spindle one turn. Does the leadscrew rotate one turn?

Vrossi050, 2020-03-25:
In the gearbox? I don't know. I haven't had it open. 11tpi looks the same as 8tpi
[image]

JonWoellhaf, 2020-03-25:
No, the end gears on the left.

From: Vrossi050 
Sent: March 24, 2020 19:37
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

In the gearbox? I don't know. I haven't had it open. 11tpi looks the same as 8tpi

Zymurgy42, 2020-03-25:
Initially I had similar issues, caused by trusting the specs in my manual for Leadscrew pitch. I physically measured (manual was wrong) input correct details;

#define LEADSCREW_HMM 250

And issue gone. A quick scan of this message doesn't show 'PITCH'.

Vrossi050, 2020-03-25:
I measured with caliper last week. Came up with 8tpi. I defined that in configuration.h file. Tomorrow reflashed board to reflect 3 microsteps and 1000 p/m for servo settings.

Vrossi050, 2020-03-25:
45t idler to 45tspindle 60t idler to 60t encoder. See photos above.

JonWoellhaf, 2020-03-25:
OK. Another new thought:

With the leadscrew marked, and the pulley attached directly to the gearbox marked, turn the pulley one rotation and confirm the leadscrew turns one rotation.

Johnboy251, 2020-03-25:
I agree with Jan - can you find out what gear setting you need to be to get 1:1 though the gearbox?

The other thing - could you confirm the pitch or TPI of the lead screw?

I’ve been working out various gear settings to try to understand where the error is hiding. In the gear box setting of ‘1 A’ there is a compound gear (120/127) that in the gear train which under mechanical configuration will give you either metric pitch or TPI with the reverent change gears.
John
P.S. stay well & away from this virus.🤞

Vrossi050, 2020-03-25:
I just marked the lead screw and pulley. It is set one to one with each other.One revolution and all the hash marks line up. Both gear box knobs are set all the way to the left. Just as James shows in his video. He says his is 12 TPI.  Maybe I should adjust encoder resolution and or Micro steps in Configuration .h file?.

Stay safe as well !!

JonWoellhaf, 2020-03-25:
I've been looking at the Grizzly G9972z manual.

It looks to me that change gear 'a' rotates at spindle speed and drives change gear 'b' through a 127 tooth idler. Thus, the gearbox input shaft, to which 'b' is attached, rotates at fraction a/b of spindle speed. The number of teeth on the idler is immaterial. But note the change gear configuration is different for cutting metric threads.

To cut 8 TPI, the lever configuration is IB, and the change gears are 'a' = 35 and 'b' = 70. Thus, the gearbox input rotates at half spindle speed, and the gearbox multiplies its input by 2, which turns the leadscrew at spindle speed, as required for an 8 TPI leadscrew to cut an 8 TPI thread. So in setting IB, the gearbox turns the leadscrew by one half its input shaft speed.

To cut a 16 TPI thread, the same change gears are used, but the levers are shifted to IA. To cut a 16 TPI thread with an 8 TPI leadscrew, the leadscrew must turn at half spindle speed. So in setting IA, the gearbox turns the leadscrew the same speed as its input shaft speed.

So I agree lever setting IA is 1 to 1 gearbox input shaft speed to leadscrew speed.

Vrossi050, 2020-03-25:
So where do I go from here?

Vrossi050, 2020-03-25:
[image]
The parts manual shows inch lead screw

JonWoellhaf, 2020-03-25:
Indeed! That is the question. We seem to have eliminated everything. This just gets interestinger and interestinger!

OK. How about this:

Set the ELS to inch and 8 TPI, then rotate the spindle by hand and watch the leadscrew. Does the leadscrew turn one turn for each turn of the spindle?

From: Vrossi050 
Sent: March 25, 2020 16:53
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

So where do I go from here?

Johnboy251, 2020-03-25:
Well I spent most of the morning checking gear ratio's to try to determine the settings on the gear box selector.
I haven't come up with anny conculsions yet... :-(

Vrossi050, 2020-03-25:
Can't turn by hand. Servo motor engaged

JonWoellhaf, 2020-03-25:
Turn the spindle, not the gearbox pulley.

From: Vrossi050 
Sent: March 25, 2020 17:03
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

Can't turn by hand. Servo motor engaged

JonWoellhaf, 2020-03-25:
James Clough, are you watching this conversation and laughing?
From: Vrossi050 
Sent: March 25, 2020 17:03
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

Can't turn by hand. Servo motor engaged

Vrossi050, 2020-03-25:
I got it. One revolution of the chuck by hand and the lead screw rotates about a quarter. It takes 12 revolutions by hand of the chuck to equal 1 revolution of the lead screw

JonWoellhaf, 2020-03-25:
OK. Now we’re on to something!

This is with the gearbox levers set to I A, correct?

This is why your threads are way too fine.

From: Vrossi050 
Sent: March 25, 2020 17:07
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Gear Ratio (#58)

I got it. One revolution of the chuck by hand and the lead screw rotates about a quarter. It takes 12 revolutions by hand of the chuck to equal 1 revolution of the lead screw

Johnboy251, 2020-03-25:
I'm in the UK - so it past my bedtime! perhaps I'll dream about this tonight (or it will keep me awake!) and I might wake up with the answer!! 
Good Night all....

Vrossi050, 2020-03-25:
Yes just as the picture shows.
Ok good night stay safe!!!

[76 more replies]

closed 2020-04-04 by Vrossi050