clough42/electronic-leadscrew #62: Stepper does not move !

pikolo335, 2020-04-10:
Hi there! I realy need some help. I connected everything together and the closed loop stepper does not move. The stepper i chose is  this [https://www.ebay.co.uk/itm/DE-12N-m-Nema34-154mm-Closed-Loop-Servomotor-Servosteuerung-HSS86-Driver-CNC-Kit/273916651738?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649](url). I connected everything in the driver as the wiring all diagram shows but then the motor controler has a green light but when i spin the encoder the motor does not move and it is stuck!
[image]
[image]

gusosborne, 2020-04-10:
When you spin the encoder, does the RPM readout show anything?

pikolo335, 2020-04-10:
the rpm moves up and down like normal but the stepper doesnt move...its like if the controller does not get the infomation i even changed the variables in the progrm and still nothing at all...is there a way to check if the board  sends signal to the controller?

> When you spin the encoder, does the RPM readout show anything?
> It is normal for a powered up stepper to feel stuck.

JonWoellhaf, 2020-04-10:
Do you have an oscilloscope?

From: pikolo335 
Sent: April 10, 2020 09:41
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

the rpm moves up and down like normal but the stepper doesnt move...its like if the controller does not get the infomation i even changed the variables in the progrm and still nothing at all...is there a way to check if the board sends signal to the controller?

  When you spin the encoder, does the RPM readout show anything?
  It is normal for a powered up stepper to feel stuck.

pikolo335, 2020-04-10:
hi there!I dont have one and i also dont know how to use one......i have checked everything alot of times and still the motor is not turning...i even tried to program it with different settings but nothing at all......i m really sad right now..

JonWoellhaf, 2020-04-10:
Don’t despair! I am certain the group here will be able to help you solve the problem. Without a scope, troubleshooting will be a bit difficult. Do you have a friend with a scope?

From: pikolo335 
Sent: April 10, 2020 13:32
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

  Do you have an oscilloscope? From: pikolo335 Sent: April 10, 2020 09:41 To: clough42/electronic-leadscrew Cc: Subscribed Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) the rpm moves up and down like normal but the stepper doesnt move...its like if the controller does not get the infomation i even changed the variables in the progrm and still nothing at all...is there a way to check if the board sends signal to the controller? When you spin the encoder, does the RPM readout show anything? It is normal for a powered up stepper to feel stuck.

hi there!I dont have one and i also dont know how to use one......i have checked everything alot of times and still the motor is not turning...i even tried to program it with different settings but nothing at all......i m really sad right now..

pikolo335, 2020-04-10:
i hope cause i am buildig this project for a long time right now and i the last two days i am stuck at it...maybe i can buy one? dont know someone that has one....but i wouldnt know how to use it ...is this something easy that i could learn?

pikolo335, 2020-04-10:
will something like this do the job? [https://www.a-store.gr/dsO138-tft-digital-oscilloscope](url)

JonWoellhaf, 2020-04-10:
It will be easy to walk you through using a scope, preferably by phone. Where are you located?

From: pikolo335 
Sent: April 10, 2020 13:43
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

  Don’t despair! I am certain the group here will be able to help you solve the problem. Without a scope, troubleshooting will be a bit difficult. Do you have a friend with a scope? From: pikolo335 Sent: April 10, 2020 13:32 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) Do you have an oscilloscope? From: pikolo335 Sent: April 10, 2020 09:41 To: clough42/electronic-leadscrew Cc: Subscribed Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) the rpm moves up and down like normal but the stepper doesnt move...its like if the controller does not get the infomation i even changed the variables in the progrm and still nothing at all...is there a way to check if the board sends signal to the controller? When you spin the encoder, does the RPM readout show anything? It is normal for a powered up stepper to feel stuck.

i hope cause i am buildig this project for a long time right now and i the last two days i am stuck at it...maybe i can buy one? dont know someone that has one....but i wouldnt know how to use it ...is this something easy that i could learn?

pikolo335, 2020-04-10:
i live in Greece...my knowledge to electronics is not very good....a friend o mine made the connections and followed the plans...JonWoellhaf  maybe you have facebook or something to talk about ?Thanks for your kindness to help me!

JonWoellhaf, 2020-04-10:
OK. Greece is a bit far from me in Utah, USA! The scope link does not work for me. Just a thought: Did you set the LAUNCHXL board switches as Clough specified? Specifically, S3 and S4 – the switches hidden under the Boost board?

From: pikolo335 
Sent: April 10, 2020 13:49
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

  It will be easy to walk you through using a scope, preferably by phone. Where are you located? From: pikolo335 Sent: April 10, 2020 13:43 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) Don’t despair! I am certain the group here will be able to help you solve the problem. Without a scope, troubleshooting will be a bit difficult. Do you have a friend with a scope? From: pikolo335 Sent: April 10, 2020 13:32 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) Do you have an oscilloscope? From: pikolo335 Sent: April 10, 2020 09:41 To: clough42/electronic-leadscrew Cc: Subscribed Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) the rpm moves up and down like normal but the stepper doesnt move...its like if the controller does not get the infomation i even changed the variables in the progrm and still nothing at all...is there a way to check if the board sends signal to the controller? When you spin the encoder, does the RPM readout show anything? It is normal for a powered up stepper to feel stuck.

i live in Greece...my knowledge to electronics is not very good....a friend o mine made the connections and followed the plans...JonWoellhaf maybe you have facebook or something to talk about ?Thanks for your kindness to help me!

pikolo335, 2020-04-10:
Yes i did the s4 to the left and both the s3 to the right as  Clough specified....i really cant find solution to the problem..

JonWoellhaf, 2020-04-10:
Can you have your friend double check the wiring?

From: pikolo335 
Sent: April 10, 2020 14:03
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

  OK. Greece is a bit far from me in Utah, USA! The scope link does not work for me. Just a thought: Did you set the LAUNCHXL board switches as Clough specified? Specifically, S3 and S4 – the switches hidden under the Boost board? From: pikolo335 Sent: April 10, 2020 13:49 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) It will be easy to walk you through using a scope, preferably by phone. Where are you located? From: pikolo335 Sent: April 10, 2020 13:43 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) Don’t despair! I am certain the group here will be able to help you solve the problem. Without a scope, troubleshooting will be a bit difficult. Do you have a friend with a scope? From: pikolo335 Sent: April 10, 2020 13:32 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) Do you have an oscilloscope? From: pikolo335 Sent: April 10, 2020 09:41 To: clough42/electronic-leadscrew Cc: Subscribed Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) the rpm moves up and down like normal but the stepper doesnt move...its like if the controller does not get the infomation i even changed the variables in the progrm and still nothing at all...is there a way to check if the board sends signal to the controller? When you spin the encoder, does the RPM readout show anything? It is normal for a powered up stepper to feel stuck.

Yes i did the s4 to the left and both the s3 to the right as Clough specified....i really cant find solution to the problem..

pikolo335, 2020-04-10:
yes we did today alot of times and all seem to be ok ....he told me that he cant understand why it is not working?Maybe this motor is not capable of working with the board?

JonWoellhaf, 2020-04-10:
Are you using the same controller and motor Clough used?

From: pikolo335 
Sent: April 10, 2020 14:15
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

yes we did today alot of times and all seem to be ok ....he told me that he cant understand why it is not working?Maybe this motor is not capable of working with the board?

pikolo335, 2020-04-10:
inmusing a different driver and motor but hte controller is the same...

JonWoellhaf, 2020-04-10:
We servo driver and servo motor are you using?

From: pikolo335 
Sent: April 10, 2020 14:27
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

  Are you using the same controller and motor Clough used? From: pikolo335 Sent: April 10, 2020 14:15 To: clough42/electronic-leadscrew Cc: JonWoellhaf ; Comment Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62) yes we did today alot of times and all seem to be ok ....he told me that he cant understand why it is not working?Maybe this motor is not capable of working with the board?

inmusing a different driver and motor but hte controller is the same...

gusosborne, 2020-04-10:
Unless you just have way too low of a voltage to power the motor..? It sounds like the wiring has to be mixed up / wrong, either at the Launchpad to the controller connection, or the controller to the motor connection..?

I’d say most likely wrong at the launchpad, since the motor is locked under power and showing a green status (that’s good). 
When you power everything off, are you able to spin the motor?

If the wiring is 100% confirmed to be correct, a scope may be the next best step. Im betting on a mixup at the launchpad to the servo-controller wiring though.

pikolo335, 2020-04-10:
I think that the  wiring is correct but i will post a diagram tommorow of how i connected it. This is the combo i used https://www.ebay.co.uk/itm/DE-12N-m-Nema34-154mm-Closed-Loop-Servomotor-Servosteuerung-HSS86-Driver-CNC-Kit/273916651738?_mwBanner=1&ul_ref=https%3A%2F%2Frover.ebay.com%2Frover%2F0%2Fe11051.m43.l1123%2F7%3Feuid%3D2e4e03a83e964ac9bfcc18c5b7916036%26bu%3D43505389723%26segname%3D11051%26crd%3D20200409192233%26osub%3D-1~1%26ch%3Dosgood%26loc%3Dhttps%253A%252F%252Fwww.ebay.co.uk%252Fulk%252Fitm%252F273916651738%26sojTags%3Dbu%3Dbu%2Cch%3Dch%2Csegname%3Dsegname%2Ccrd%3Dcrd%2Curl%3Dloc%2Cosub%3Dosub%26srcrot%3De11051.m43.l1123%26rvr_id%3D0%26rvr_ts%3D65f24e901710ad4eb597a788fff2e2b4&ul_noapp=true&pageci=2322f993-611b-46cf-a48a-a32dd55d8db3     with a 70v power supply

gusosborne, 2020-04-10:
A picture showing the wiring on both ends (launchpad to controller) would be very helpful too. 

When you power everything down, the motor should spin pretty freely (with some light cogging / clicks), have you tried that? Just to make sure your motor is not totally frozen up.

Vrossi050, 2020-04-10:
Things to check
1. Did you use the motor extension  cable that came with the kit? If you don't it won't work.
2. Does the servo alarm when turned on?
3. Does it turn freely or is locked in position?
4. Voltage correct to the motor?
5. micro steps correct?

pikolo335, 2020-04-10:
I have tried this before i connect the motor and indeed it spined with some cogging...i didnt try this now but i will try it again! The driver has some different signs on it.....it doesn't have the fLt so i connected the alm+ to alm+  and alm- to alm-  Also the new driver has a pend+ and pend - option that i didnt connect it.

pikolo335, 2020-04-10:
> Things to check
> 
> 1. Did you use the motor extension  cable that came with the kit? If you don't it won't work.
> 2. Does the servo alarm when turned on?
> 3. Does it turn freely or is locked in position?
> 4. Voltage correct to the motor?
> 5. micro steps correct?
I used the extension cable and the servo alarm doesnt light at all when turned on.I connected the motor like the connections it has on it. I put the driver microstep to 4000 and to thw firmware microstep to 4 and Resolution to 1000 as clough42 suggested to me

Vrossi050, 2020-04-10:
The wiring diagram from the controller doesn't look like it is getting positive voltage 
[image]
and the wiring looks different from James' wiring diagram

Vrossi050, 2020-04-10:
[image]

pikolo335, 2020-04-10:
It says +5v on the pul+ side  i connected as clough42 had it on thw diagram.Do you believe that it needs something connected differently?

Vrossi050, 2020-04-11:
[image]

Vrossi050, 2020-04-11:
Switches on the driver are they set right? See manual page 6
[Servo Stepper Driver UK 2 HSS86.pdf](https://github.com/clough42/electronic-leadscrew/files/4463905/Servo.Stepper.Driver.UK.2.HSS86.pdf)

pikolo335, 2020-04-11:
I have connected everything exactly as you say on the driver...tha switches are also set for 4000 resolution as the pdf shows... The power supply i use is this  Single Output AC to DC Power Supply 400W 70V 5.7A  ....i really dont know what is wrong... :(

JonWoellhaf, 2020-04-11:
Do you have an LED and a 330 or 470 ohm resistor?

From: pikolo335 
Sent: April 11, 2020 02:51
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

I have connected everything exactly as you say on the driver...tha switches are also set for 4000 resolution as the pdf shows... The power supply i use is this Single Output AC to DC Power Supply 400W 70V 5.7A ....i really dont know what is wrong... :(

pikolo335, 2020-04-11:
Sorry for the question but where should i have the led and this resistor? I found a friend that says can find an oscilloscope or buy this one?but
[image]
 how do i use it and what do i search for?

JonWoellhaf, 2020-04-11:
With the lathe running, you want to use the LED or scope to verify the pulse signal is active. I thought an LED would be a simple way to do this. If you can get one, connect the resistor to the anode. apply +5 volts to the other end of the resistor. Touch the cathode to the pulse– input. I expect the LED will light. At slowest speed it may flicker. Also probe the direction and enable inputs. Direction should cause the LED to light when direction is one way and be dark the other way. If it stays lit or dark when you change direction, then we’re on to something. Enable should light the LED whenever the servo should be turning.

That little digital scope looks great! I just bought one. But it’s a kit. You said you have little experience with electronics, so I doubt you will be able to build it. But maybe someone can do that for you. You’ll also want to buy a scope probe.

From: pikolo335 
Sent: April 11, 2020 05:33
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

Sorry for the question but where should i have the led and this resistor? I found a friend that says can find an oscilloscope or buy this one?but

how do i use it and what do i search for?

pikolo335, 2020-04-11:
I will tell my friend if he can do it...on the other hand i have a cnc and i have another stepper and driver so i could try to use it with this and see if it runs then it is something on the motor and driver.if it doesnt run thw it is the  board or the launchpad...

Johnboy251, 2020-04-11:
Hi. Are lines of the code 70 to 72 active. They look ‘Remed’ out. 
John

JonWoellhaf, 2020-04-11:
Yes. Good idea.

The little scope has low bandwidth and may not be able to allow you to see the step pulses.

From: pikolo335 
Sent: April 11, 2020 06:01
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

I will tell my friend if he can do it...on the other hand i have a cnc and i have another stepper and driver so i could try to use it with this and see if it runs then it is something on the motor and driver.if it doesnt run thw it is the board or the launchpad...

JonWoellhaf, 2020-04-11:
Yes, those lines are commented out. Line 70 will always be commented out, since it’s a comment.

Clough’s original has 71 and 72 commented out, but it seems to me they should not be, since all the controller inputs are active low.

What am I missing?

From: Johnboy251 
Sent: April 11, 2020 06:13
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

Hi. Are lines of the code 70 to 72 active. They look ‘Remed’ out.
John

Johnboy251, 2020-04-11:
Hi...
I remember I had the same problem. Two things...
(1.) is the dip switches on the hybrid stepper drive set? Can you access the drive via an RS232 serial cable to a laptop/pc? Software is available to read drive parameters i I have the link somewhere 😳

(2.) I found my default settting stop my system working before S9 on the TI LaunchPad was not set to the ‘default’
I write this at the time...
One thing that I found is that my TI LaunchPad was supplied with the S9 set as a default to CAN Route. Which I overlooked on initial installation. Otherwise it would have run first time after download the code.

Stay well 🤞
John

pikolo335, 2020-04-11:
I tried to make them all active and nothing changed...i re programmed the xllaunch board many times but nothing changed... RS232 serial is this an ethernet cable or a lan? Maybe i can find one ...?

pikolo335, 2020-04-11:
The driver i use is not the one clough42 is using...mine is this one hss86 https://www.ebay.co.uk/itm/DE-12N-m-Nema34-154mm-Closed-Loop-Servomotor-Servosteuerung-HSS86-Driver-CNC-Kit/273916651738?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3De7cdbb5f62fc4be9900fce552b22a487%26pid%3D100675%26rk%3D1%26rkt%3D15%26mehot%3Dnone%26sd%3D273916651738%26itm%3D273916651738%26pmt%3D0%26noa%3D1%26pg%3D2380057&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3A9f94185e-7bf8-11ea-9864-74dbd18044f5%7Cparentrq%3A696f4f1d1710ac3e25836aadfffdfa46%7Ciid%3A1

pikolo335, 2020-04-11:
I can see on the clough42 photo that s9 is set to the right to can route.....you say that it shouldnt be there??

pikolo335, 2020-04-11:
I am looking at a youtube video that a person is progeamming the driver and it has this picture in the manual 
[image] that looks like some oins are active low??is this the problem?

Johnboy251, 2020-04-11:
S9 shouldn’t be selected to CAN Bus - have look at th last of my messages. My TI board was not set to the same default as Jame’s board. 
My system didn’t work until I changed S9

Regards 
John

Johnboy251, 2020-04-11:
😱

Regards 
John

Johnboy251, 2020-04-11:
Hi  I’ll look later Sorry in middle of something else on my lathe. 👍

Regards 
John

pikolo335, 2020-04-11:
> S9 shouldn’t be selected to CAN Bus - have look at th last of my messages. My TI board was not set to the same default as Jame’s board. My system didn’t work until I changed S9 Regards John
> […](#)
> Sent from my iPhone6s

So it should be to the left or to the right i see in the clough42 wiring all diagram that it is to the can route side ....to the right side.  This is the documentation maybe someone can find something
[image]

pikolo335, 2020-04-11:
[image]
I think that it is set to the can route side...isnt it?

pikolo335, 2020-04-11:
So i changed the s9 and then the screen didnt work...so it is not this.....

pikolo335, 2020-04-11:
[image]
Maybe the answere is here?

Johnboy251, 2020-04-11:
Hmmm Sorry I’ve got my reasoning wrong. My S9 switch was set to the Left - I interpreted this switch as ‘on’ i.e. aligned with the dot (left hand end as ‘on’) I through I understood the if this switch was’ on’ then CAN ROUTE would be enabled which we do not want. I hope this make sense!😳🤞

Johnboy251, 2020-04-11:
[image]

pikolo335, 2020-04-11:
Nothing works....it is really frustrating....to have alla done and see this thing not moving....the motor is moving by hand when it is not powered but will not move when the power is on

Johnboy251, 2020-04-11:
Ok, can you measure the ‘pull up’ voltage coming out of the CLOUGH42 interface board with a multi meter (DVM)? It should be about 5 Volts for me?

Johnboy251, 2020-04-11:
Don’t Jan & myself sorted Vrossio50 problem out so we’ve got a ‘track record’ 😋

pikolo335, 2020-04-11:
> Don’t worry - Jan & myself sorted Vrossio50 problem out so we’ve got a ‘track record’ 😋
i really hope you ca help me out....i think that i have found something!!
The driver says that it multiplies the stepper resolution by 4 so the stepper encoder has 1000 lines so the stepper resolution is now 4000    and the microstep is set to 1000 at this moment    but in the cloughs 42 software i cant change the stepper resolution as the program says that i have to choose from 1 to 2000 ......but mine is 4000..... i think this is the problem? what do you think?

Johnboy251, 2020-04-11:
If it was - the encoder wouldn’t work....

Regards
John

Sent from my iPadPro

> On 11 Apr 2020, at 15:31, pikolo335 <notifications@github.com> wrote:
> 
> it?

pikolo335, 2020-04-11:
i dont understand exactly... cant see why this is not working.....

Johnboy251, 2020-04-11:
I’m sorry I can’t get back to you on this right now - will get some time later. It’s 17.41 GMT here in the UK at time of writing this message....

Regards
John

pikolo335, 2020-04-11:
ok thanks if you can please reply layter on :)

JonWoellhaf, 2020-04-11:
pikolo335,

Here's what we know:
1) Your spindle encoder is working properly, because you get an rpm readout.
2) Your control panel is working properly, because you get an rpm readout.
3) A large portion of the LAUNCHXL firmware is working properly, because you get an rpm readout.
4) Wiring from the Boost board to the servo driver appears to be correct and has been double checked.
I suspect the problem is with the CONNECTION of the wires from the boost board to the servo driver, on one or both ends.
Please remove one connection at a time and be certain the WIRE -- not the insulation -- is making contact. Pull on each wire after reconnecting to insure the connection is tight.
Also, I would like you to change the daisy-chained +5 volt connections at the servo driver. You now have two wires in one connector. Please put individual wires in each of the + connectors, then join those three wires to the one from the boost board.
Regards,
Jan (Yan)

pikolo335, 2020-04-11:
> pikolo335,
>
> Here's what we know:
>
> 1. Your spindle encoder is working properly, because you get an rpm readout.
> 2. Your control panel is working properly, because you get an rpm readout.
> 3. A large [...]

a friend of mine that connected the wires checked them with a voltometer so as to be sure that the cable is touching and every time the voltometer made a sound...
i will check them again though. could you please explain  a little better what i have to do ?  because the pul+ wire is going to have again two wires...or you mean that the dir + should not be connected with Dir+ and EAN+? and alla these 3 go to  the + of the boost board?

JonWoellhaf, 2020-04-11:
In your first post on Github, you show a closeup view of the servo driver.

PUL+ has two wires in it, both red. One goes to + on the Boost board and the other goes to DIR+.

A second wire, yellow, goes from DIR+ to ENA+.

The wiring is correct, but I am concerned the two wires in PUL+ and DIR+ are not making contact.

I suggest you remove all the wires in PUL+, DIR+, and ENA+, then put one wire in each. Connect those three wires to the one going to + on the Boost board.

Checking continuity with an ohm meter is a very good idea. With power off, check continuity from SCREW to SCREW on the connectors.

Do you have another 2HSS86 servo driver?

From: pikolo335 
Sent: April 11, 2020 12:33
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Stepper does not move ! (#62)

  pikolo335,

  Here's what we know:

    1.. Your spindle encoder is working properly, because you get an rpm readout. 
    2.. Your control panel is working properly, because you get an rpm readout. 
    3.. A large portion of the LAUNCHXL firmware is working properly, because you get an rpm readout. 
    4.. Wiring from the Boost board to the servo driver appears to be correct and has been double checked.
    I suspect the problem is with the CONNECTION of the wires from the boost board to the servo driver, on one or both ends.
    Please remove one connection at a time and be certain the WIRE -- not the insulation -- is making contact. Pull on each wire after reconnecting to insure the connection is tight.
    Also, I would like you to change the daisy-chained +5 volt connections at the servo driver. You now have two wires in one connector. Please put individual wires in each of the + connectors, then join those three wires to the one from the boost board.
    Regards,
    Jan (Yan) 
a friend of mine that connected the wires checked them with a voltometer so as to be sure that the cable is touching and every time the voltometer made a sound...
i will check them again though. could you please explain a little better what i have to do ? because the pul+ wire is going to have again two wires...or you mean that the dir + should not be connected with Dir+ and EAN+? and alla these 3 go to the + of the boost board?

[181 more replies]

closed 2020-04-16 by pikolo335