clough42/electronic-leadscrew #63: problem/Servo turns only on one direction

pikolo335, 2020-04-16:
So i continue the thread to this new issue. After thegreat help i received i managed to make the motor run. Now the problem is that it turns on one direction only cw and when rotating the encoder anticlockwise it does not turn. I want to  say again that when i connected the  servo to my cnc using the diagram of connecting a dm54a drivers(it connectd i think te dir + and -) it moved both ways when pressing page up and page two. Looking forward to your suggestions :)

Vrossi050, 2020-04-16:
Try pushing the fwd/Rev button on the display.

JonWoellhaf, 2020-04-16:
When you rotate the encoder anticlockwise, do you get an rpm readout on the control panel?

Vrossi050, 2020-04-16:
Third button from the left bottom row on the display module.

pikolo335, 2020-04-16:
> Try pushing the fwd/Rev button on the display.

i did but this didnt do something ....i made a video showing the problem. I dont show this on the video but the motor should move when i change the encoder direction but it does not...it only moves in one direction   https://files.fm/u/kqg98pud?fbclid=IwAR1Q6mx57fdvbsDgET4i-MhcDBaT1TnEiCQucMQ5zwSolND8belttr5hDss#/view/VID_20200416_172318.mp4;play

pikolo335, 2020-04-16:
> When you rotate the encoder anticlockwise, do you get an rpm readout on the control panel?

this is a nice question and i think it does i think alsothat it can be seen in the video that it does but cant remember for sure right now......i think that the dm54a drivers use a different connection and i when i used it on the cnc it did move both ways.also the motor when reversing right now unlocks and then spins freely by hand...

JonWoellhaf, 2020-04-16:
I watched the first 39 seconds of the video and then it stalled for ten minutes. I'll watch again and look for the rpm display.

pikolo335, 2020-04-16:
i will  try to upload it in another site this one is no good...

pikolo335, 2020-04-16:
give me some time i am reuploading the file

Vrossi050, 2020-04-16:
Sounds like a another bad connection on the screw terminal or channel A&B are reversed

pikolo335, 2020-04-16:
> Sounds like a another bad connection on the screw terminal or channel A&B are reversed

i think it is not a bad connection this time cause i checked them really well....i dont want to belive that the servo side driver has also a bad connection/..and also it did work before with the cnc and this proves that the servo can work both ways...also the the connections know on the board are checked.Also i think that the cnc uses a different connection pattern cause i remember that it uses dir + and dir - and also pul+ and pul- but doesnt move the ena inmputs.....could be that the dir+ should be set to true in the software?

pikolo335, 2020-04-16:
So this is the new video link!  https://vimeo.com/408504381

Vrossi050, 2020-04-16:
I am not sure about changing the software.

Vrossi050, 2020-04-16:
Is it wired like this? 
[image]

JonWoellhaf, 2020-04-16:
Sorry, this one freezes too, after about 40 seconds.

pikolo335, 2020-04-16:
> Is it wired like this?
> [image]

Yes this how i connected it. @JonWoellhaf  did you try to press pause for some minutes so as to load the video?In the first video link you can download it too to see it from your pc.

pikolo335, 2020-04-16:
@JonWoellhaf  please try again cause vimeo now send a link that says the video is ready...maybe it was being convert by vimeo at the time you saw it...

JonWoellhaf, 2020-04-16:
OK. I finally got the video to load and watched it. Unfortunately, I didn't see the rpm readout when you reversed the encoder direction.

Please tell me when you can observe the rpm readout for both encoder directions.

pikolo335, 2020-04-16:
> OK. I finally got the video to load and watched it. Unfortunately, I didn't see the rpm readout when you reversed the encoder direction.
> 
> Please tell me when you can observe the rpm readout for both encoder directions.

Ok i will tell you tommorow that i will go again to see it...i posed the video and i saw that it had measured the rpm but to be sure i will do it again! @JonWoellhaf  do you believe it is a software issue?

JonWoellhaf, 2020-04-16:
OK. I suspect the rpm readout will be the same no matter which way you turn the encoder.

With that in mind, I am thinking about what fault could cause your symptom.

Ideas anyone?

Vrossi050, 2020-04-16:
Just a thought. His encoder and motor have one more wire each than what James has suggested. Could it be that it is only pulling high in one direction and the wiring or settings in the driver need tweaked?

Vrossi050, 2020-04-16:
SW2 off =ccw off
SW2 on = Cw on

pikolo335, 2020-04-16:
Dont know what is wrong again....i connected the encoder just like in the issue with the encoder picture...and the lcd as  it ia shown in the diagram....i think it is one to one? Should i try to reprogram using the define active dir high option?

pikolo335, 2020-04-16:
> SW2 off =ccw off
> SW2 on = Cw on

Tried this...it changes the motor direction but this doesnt mean that it spins both ways...

Vrossi050, 2020-04-16:
I noticed in the video that the drill is spinning in the same direction on both tests.

Johnboy251, 2020-04-16:
> > SW2 off =ccw off
> > SW2 on = Cw on
> 
> Tried this...it changes the motor direction but this doesnt mean that it spins both ways...

Which is correct. The DIP switches will only set the direction you choose as to make the lead screw go in the direct you wish to refer to as forward or backwards. This really looks like the direction signal is not getting to the Servo Diver unit.
John

Johnboy251, 2020-04-16:
Alex
One thing I picked up on was that you were using the drill to rotate the encoder. Thats ok to demonstrate fast RPM but if rotating the encoder like James demonstrates early on in his video - the servo moves backward & forward. Which works for me! Ha ha! 😂🤣😂
I wouldn’t really worry about true rpm until it fitted to the lathe!

pikolo335, 2020-04-16:
> I noticed in the video that the drill is spinning in the same direction on both tests.

It might look like this but it is not....i will upload another video tommorow so as to prove it..

Johnboy251, 2020-04-16:
I think this is the area you need to concentrate on.👍

pikolo335, 2020-04-16:
> Alex
> One thing I picked up on was that you were using the drill to rotate the encoder. Thats ok to demonstrate fast RPM but if rotating the encoder like James demonstrates early on in his video - [...]

When i spin it even by hand it does not rotate like james did....it works only when i rotate the encoder clockwise...when i turn the encoder anti clockwise the servo does not move and i think that it unlocks its position too...

pikolo335, 2020-04-16:
> Which is correct. The DIP switches will only set the direction you choose as to make the lead screw go in the direct you wish to refer to as forward or backwards. This really looks like the [...]

Yes i believe it too that the signal doesnt go to the servo or that there is no signal when it ia rotated opposite to what it needs?Can someone post the connection pins for the screen and the encoder? I think in james sketch it is not very clear..i want to  check it again for any mistakes..

pikolo335, 2020-04-16:
Doew this picture tells ua if the pul dit and ena are active low or active high?
[image]

pikolo335, 2020-04-16:
Also another one...it looks like it needs 5v to move one direction and 4.5 to move the other? It also says that i has two modes...
[image]

Johnboy251, 2020-04-16:
Alex
You wrote “ Also another one...it looks like it needs 5v to move one direction and 4.5 to move the other? It also says that i has two modes...”  

Which is wrong.

Pleas note the following...
In this type of device creating the logic level for a logic ‘0’ is equivalent to 0.0 Volts to 0.8 Volts. A logic level for a logic ‘1’ is equivalent to 2.0 Volts to 5.0 Volts. The ‘dead’ zone where it consider neither a logic ‘0’ or ‘1’ is between anything greater than (represented by a ‘>’ symbol) 0.8 Volts and less than (represented by a ‘<’ symbol) 2.0Volts. This is conventional TTL logic levels.

It is 23.06 hours BST - I will speak with you tomorrow. I have to think how to direct you for tomorrow. 😳👍

John 🇬🇧

pikolo335, 2020-04-16:
Ok i will post back also more info tommorow! :)

JonWoellhaf, 2020-04-17:
OK. I have an idea.

We know (or suppose) these facts:

1) The servo motor turns when the encoder is spun CW.

2) The servo motor does not turn when the encoder is spun in the opposite direction.

3)  When the servo motor doesn't turn it is no longer being held in position and can be freely turned by hand. (We think.)

4) We know from observation that the DR- output changes state when the encoder direction is reversed.

So, what happens if EN- and DR- have been inadvertently swapped? That means EN- is connected to DIR-, and DR- is connected to ENA-.

1) When the encoder is spun CW, DR- (now connected to the servo driver's ENA-) happens to be in the state that enables the servo driver. The state of EN- (now connected to the servo driver's DIR-) is unimportant, because irrespective of its state, the motor will turn one direction or the other, and at this point we con't care which direction it turns -- only that it does turn.

2) When the encoder is spun CCW, DR- (now connected to the servo driver's ENA-) changes to the state that DISABLES the servo driver. So the driver ceases to hold motor position and the motor can easily be turned by hand.

What do you think?

Johnboy251, 2020-04-17:
06.30 BST
Alex
Do you have a copy of this download?http://leadshine.com/UploadFile/Down/ProTunerCS_sm31.pdf

You wrote../
“ When i spin it even by hand it does not rotate like james did....it works only when i rotate the encoder clockwise...when i turn the encoder anti clockwise the servo does not move and i think that it unlocks its position too...”

If you select the fastest leadscrew feed rate, the smallest TPI or the largest pitch the servo drive will move faster. 
Look at the push button designations that James has labelled the box the LED keypad is housed.

pikolo335, 2020-04-17:
> OK. I have an idea.
>
> We know (or suppose) these facts:
>
> 1. The servo motor turns when the encoder is spun CW.
> 2. The servo motor does not turn when the encoder is spun in the opposite direction.
> 3. [...]

This is it i think!But how is it solved?also when i connected it to cnc i had only the dir- and the dir+ connections  and also pul+ and pul- .So what do i have to do?

JonWoellhaf, 2020-04-17:
Check the wiring. I believe you have DR- and EN- swapped.

If it still doesn't work, disconnect both EN- and DR- and try again.

pikolo335, 2020-04-17:
Ok thabks i will do it and post back results!

pikolo335, 2020-04-17:
Oh my goddddddd!!!!you were righttttt by avcident i had them connected wrong!!!!!!!it workssssss it worksssss!!!!!!

pikolo335, 2020-04-17:
i willl upload videos with the lathe working!!1I really really really really thanks you!!!!!

gusosborne, 2020-04-17:
Wow. Great work fellas!
We need to find a way to do something cool for Jan & John - these guys are amazeballs!

pikolo335, 2020-04-17:
> Wow. Great work fellas!
> We need to find a way to do something cool for Jan & John - these guys are amazeballs!
i was thinking about this too!! They realy helped me!!!!

JonWoellhaf, 2020-04-17:
pikolo335,

It was a great treat to wake up this morning to such good news! I am delighted we were able to solve this very interesting mystery.

Over the years -- decades, really -- that I've been messing with electronics and software, I've encountered many problems whose symptoms were so bazaar it seemed they couldn't possibly be real. And in such situations, especially with software, I've often thought, "If someone asked me to write code that would behave like I'm seeing, I would have absolutely no idea how to do it!"

This problem is definitely in that category. The symptoms you described seemed impossible, but I believed you were reporting them correctly.

The satisfaction of helping someone and making new friends is great and reward enough! I won't speak for John, but I think he will agree.

Stay well, everyone!

Jan

pikolo335, 2020-04-17:
This post is to thank everyone that helped me to solve the problem that without your help i wouldnt be able to make it through!i bought the lathe from the scrapyard in an awful condition...i mean nothing moved cause of the rust. This lathe didnt had the ability to cut threads as it was from the factory so the els could make it possible...the lathe was like this 
[image]
[image]
[image]

After a lot of effort the lathe became like this 
[image]
[image]
So with what i had on hand i made a mechanism so as to make it able to feed and cut threads this is the video i took today https://vimeo.com/408959371
@JonWoellhaf  i could donate some money to you so as to buy you a coffee for your help like we call it in Greece! I really appreciate that someone who doesnt know me gave its time to help me! @Vrossi050 @Johnboy251 @gusosborne @JonWoellhaf  i really really really really thank you!!!!!!!!!!!!!!!!!!!

JonWoellhaf, 2020-04-17:
Those are great pictures. You've done a great job refurbishing that machine!

I think you should have a YouTube channel so we can follow your continuing progress. Or do you already have one?

Your appreciation, alone, is more than compensation for my time helping you. Thanks to you, I learned several important things about James' ELS, and you presented two very interesting puzzles that were fun to solve!

Best wishes with the continuation of your project.

Jan

Johnboy251, 2020-04-17:
Wow Alex. 
When I saw the first picture of your lathe - I wondered what you had taken on! But all credit to you - it’s a great job!
Is that a hydraulic pump under underneath? If it is, is it a profile copying lathe?

Like Jan said it is good to help - we ‘old ones’ can’t download our knowledge and usually we ‘take it with us’ So it’s great to pass on what know to guys like you who want to learn.... 👍

pikolo335, 2020-04-17:
> Those are great pictures. You've done a great job refurbishing that machine!
>
> I think you should have a YouTube channel so we can follow your continuing progress. Or do you already have one?
>
> Your [...]

I dont have a channel yet but i have thought about it i may do it if there is interest!I have a lot of projects to do as i am an art teacher for living and i am a luthier too...I am also building a concrete milling machine and if i am able to cut threads i will complete it!My facebook is Alexandros nikolakis if someone wants to be friends there and talk from time to time about our projects!

pikolo335, 2020-04-17:
> Wow Alex.
> When I saw the first picture of your lathe - I wondered what you had taken on! But all credit to you - it’s a great job!
> Is that a hydraulic pump under underneath? If it is, is it a [...]

I really thank you too john!!this is a heavy duty lathe for factory use yes it is a copying lathe that only had a feed screw...i have a lot more pictures of this lathe restoration! It is my passion my hobby....i did need a lathe to be able to complete my milling machine ..and a milling machine to be able to make parts for  luthiery...also the cnc is a diy project...a heavily modified momus cnc...

Johnboy251, 2020-04-17:
I found this on a website here in the UK
Information on UMA lathes.   http://www.lathes.co.uk/uma/

pikolo335, 2020-04-17:
Yes the last one is like mine...it is belt driven and inside the table of it it has a gearbox with speeds for the spindle!

gusosborne, 2020-04-17:
Amazing restoration on that lathe man. Unbelievable. 
Wonder if you could resurrect the old pump and use it as a multi-purpose, for something like a press / brake / shear / etc. 

I see lots of guitar makers on YouTube, but not a whole lot else in the way of luthiers. What type of instruments do you primarily do? What do you prefer to do?

Johnboy251, 2020-04-18:
Yes - I agree with Jan. 
It’s really good to help. As Jan mentioned being in Engineering - Electronic, Electrical & Mechanical it’s sometimes overwhelming trying to find a fault that should not exist! But with careful analysis and trying to ‘step back’ to check the most simple of things pays benefits. It’s all part of the learning curve and I believe it makes you a better Engineer.
I’ve been fortunate having an interesting & enjoyable career. I’m glad to be able to pass my knowledge I’ve acquired. 
Considering the differences in the language and global locations we’ve made friends and we’ve all cracked it! 

Keep well and stay away from this virus. This Pandemic is a Global game changer...
John 🙂

pikolo335, 2020-04-18:
> Amazing restoration on that lathe man. Unbelievable.
> Wonder if you could resurrect the old pump and use it as a multi-purpose, for something like a press / brake / shear / etc.
>
> I see lots of guitar [...]

I built greek bouzoukis baglamas  tzouras guitars everything it comes to my hand but this is also a hobby for me...I studied arts at school of arts in thessaloniki and i am a painter that teaches art in school! Two photos of the bouzouki
[image]
[image]

pikolo335, 2020-04-18:
> Yes - I agree with Jan.
> It’s really good to help. As Jan mentioned being in Engineering - Electronic, Electrical & Mechanical it’s sometimes overwhelming trying to find a fault that should not [...]

I really thank you too john you helped me alot from day one!!!!:)) I hope you all too are good and stay  away from.this virus!!!!!

gusosborne, 2020-04-19:
Amazing restoration man. 

> I built greek bouzoukis baglamas tzouras guitars everything it comes to my hand but this is also a hobby for me...I studied arts at school of arts in thessaloniki and i am a painter that teaches art [...]

That is a beautiful instrument - you seem to have some serious talent. I know this probably isn’t the best place for this discussion- but did you do those inlays on the fretboard or the top? 
And the lathe restoration is really unbelievable too. You should definitely make some videos.

pikolo335, 2020-04-19:
@gusosborne  Yes the i lays were made by me! Thanks a lot my friendd!!!I. Have tried to make some videos with the lathe restoration before but it is very difficult to find thw setup(i mean which camera and program and how to hold it while doing work) I will try again for sure!The lathe took me a lot of man hours and i dont know if it worth it but time will tell...if i am.able now to cut threads then i will be very happy and make more stuff!!!

closed 2020-10-12 by clough42