clough42/electronic-leadscrew #67: no step count or enable signals to stepper motor

Clive598, 2020-04-26:
hello there I'm just wondering if anybody can shed some light on an issue I am having with the output to the stepper driver. 

I have loaded up the software V 1.1.02 onto the launch board and I have the boosted board plug in the top when I spin the encoder the RPM  display works fine. 

the problem I'm having the is there is no change of signal on the step output or on the enable output.  I do though get a output on the direction pin depending on which way I turn the encoder.

I have tried changing the #define INVERT_STEP_PIN true but all this does is change the output value to low. 

if I tried to invert or comment out the //  #define INVERT_ENABLE_PIN true   it stays at high potential whatever I do in the configuration h. file

I have put an oscilloscope on both outputs and not seen any change signal. Also I have connected three LEDs with 4.7 K resistors to the 5v supply from the boosted board to each of the outputs.
the only LED that changes when the encoder turns is direction.

I would really appreciate if anybody can help me with this problem I'm having maybe it's something silly I have done or overlook. also many thanks to John for sharing this awesome project.

JonWoellhaf, 2020-04-26:
Hi, Clive598.

Hope I can help.

Please verify that position of S3 and S4 on the TI board. They're under the Boost board.

Both S3 switches need to be positioned away from U9 -- toward its dot. S4 also needs to be positioned away from U9 -- toward its dot.

JonWoellhaf, 2020-04-26:
Please try this:

1) Disconnect the resistors and LEDs.

2) Put one of the resistors from + to STP- on the servo driver connector.

3) Put your scope on the STP- lead, spin the encoder, and see what you get.

Clive598, 2020-04-26:
HI Jon

Thank you for responding to my request for help.

I have checked the switches and they are as you say.

I have disconnected the LEDs and placed a 4.7 K resistor between 5V+ and STP- when I rotate the encoder I get no signal On the scope.

[image]

JonWoellhaf, 2020-04-26:
OK. This is like déjà vu all over again. :)

Please put your probe on pin P00 of J8 on the back of the TI board, spin the encoder, and see what you get.

Clive598, 2020-04-27:
HI Jon 

I found the pulse at the pin 5 µs  3.3v

[image]

I also read pro at the boosted board and found the signal 5 V

[image]

JonWoellhaf, 2020-04-27:
OK!

I think there's a typo in your last message. Am I correct that you saw the pulse on the STP- terminal on the servo driver connector?

If so, its polarity is incorrect, so in Configuration.h change line 71 to be

    71 // #define INVERT_STEP_PIN true

Then check the STP- signal. It should be a 5 us wide negative going pulse at about 5 volts.

Clive598, 2020-04-27:
HI Jon

I changed the configuration file as you said copy below

// Steps and microsteps
#define STEPPER_MICROSTEPS 8
#define STEPPER_RESOLUTION 200

// Step, direction and enable pins are normally active-high
// #define INVERT_STEP_PIN true
// #define INVERT_DIRECTION_PIN true
// #define INVERT_ENABLE_PIN true
// #define INVERT_ALARM_PIN true

// Enable servo alarm feedback
// #define USE_ALARM_PIN

this is the signal I get now

[image]

this is a stepper motor driver I have tried to interface to
[image]

JonWoellhaf, 2020-04-27:
OK. Let me take a look at that driver.

Meanwhile, do you have heat shrink tubing? I am concerned the wires from A+, A-, B+ and B- to your motor connector are very close to shorting. Please cover the connector contacts with heat shrink tubing before applying servo power again.

Clive598, 2020-04-27:
HI Jon
Okay I will tidy up loose cables. I also have datasheet on stepper driver

[Stepper Motor Driver Controller B6600.pdf](https://github.com/clough42/electronic-leadscrew/files/4537019/Stepper.Motor.Driver.Controller.B6600.pdf)

JonWoellhaf, 2020-04-27:
Please try this:

1) Remove all the wires from J2 on the Boost board and the Signal connector on the servo driver.

2) Connect + on J2 to PUL+ on the servo driver.

3) Connect STP- on J2 to PUL- on the servo driver.

What servo motor are you using?

The TB6600 driver pdf is a bit sparse! :)

Clive598, 2020-04-27:
HI Jon

thank you very much for your time and patience with me
success all working the servo leads going from driver to the motor were crossed I have followed the datasheet but I should have also checked with the meter the coils of the motor as well.

after realising this mistake I made I thought I'd just quickly reverse the enable pin to see if it would work. and it did.

The stepper motor I'm using is a Nema 17  but the biggest one I could find. the lane is quite a small one and I would only be using it for light thread cutting.

if there wasn't enough power I was thinking about putting two of those stepper motors on the lead screw.  My idea is to try and build everything into the lane itself there is room in a box that the lead screw passes through where both the motor could be.

I will put the first motor on temporarily to see if it has enough power.

once again I thank you very much I really was at a loss and I couldn't see a way forward to get this  working.

when I get it all together I will post pictures of this conversion on the Waco 

Kind regards, Clive

[Nema 17 stepper motor.pdf](https://github.com/clough42/electronic-leadscrew/files/4537110/Nema.17.stepper.motor.pdf)

[image]

JonWoellhaf, 2020-04-27:
Excellent!

Thanks for another fun puzzle to solve. (You made this one too easy, though.)

Jan (Yan)