clough42/electronic-leadscrew #69: TPI not accurate

Scubamachi, 2020-04-30:
I have the project up and running on my G0752Z lathe. When I attempt to thread 13 TPI it comes out about 16TPI, and when I thread at 24 TPI it comes out around 32 TPI. I know the lead screw is set one to one. My rpm on the machine matches the readout on the LED KEY. I am using the servo package from Automation Technologies with the KL 5080H driver and the KL23-2N-1000 servo motor. I am using the 3 to 1 configuration on the pulleys. I am using the stepper resolution of 1000 and the stepper microsteps are set to 3 for the 3:1 reduction. I have tried the default DIP switch setting on the KL5080H along with trying the settings at 800 and 1000 with the DIP switches. I am a little confused that my KL 5080H has DIP switches and the pictured one on Automation's site does not. My KL5080H has a step angle of 1.2 degrees which seems to differ from most that are 1.8 degrees. Appreciate any help I can get with this an a big shout out to James for putting this project together.

Johnboy251, 2020-04-30:
Can we have a picture of the DIP switches on the KL 5080H controller? That’s a good place to start... 
regards 
John

Scubamachi, 2020-04-30:
John,
I replied to the notification email I received with a picture of the KL5080H. I am having trouble uploading the picture here, but it might be my work email creating the issue. I will try tonight to post a picture, from my home email. Thank you very much for helping with this. .

Scubamachi, 2020-04-30:
[image]
[image]

Johnboy251, 2020-04-30:
For 1000 pulse per rev of the servo DIp,Switch 4 should be ‘off’ i.e.  It should be in the other position to what it is as depicted. The next thing to look at is the settings for the servo - off the top of my head I can’t remember in what video that is... it’s now 00.40 am here in uk - I’ll look tomorrow morning.
John

JonWoellhaf, 2020-05-01:
John,

Have you followed this thread on GitHub?

I hope I didn’t lead him astray!

Jan

From: John 
Sent: April 30, 2020 17:41
To: clough42/electronic-leadscrew 
Cc: Subscribed 
Subject: Re: [clough42/electronic-leadscrew] TPI not accurate (#69)

For 1000 pulse per rev of the servo DIp,Switch 4 should be ‘off’ i.e. It should be in the other position to what it is as depicted. The next thing to look at is the settings for the servo - off the top of my head I can’t remember in what video that is... it’s now 00.40 am here in uk - I’ll look tomorrow morning.
John

Scubamachi, 2020-05-01:
Hi guys,
No one has led me astray. The new package I ordered is from Amazon, and I can easily return it if I decide not to use it. I starting watching James's  videos over again today as I thought he made mention of a change regarding a 1.2 step angle. I have not found it yet. I believe the actual motor I have,(KL23-2N-1000 from Automation technolog  ) might of been in James's videos at some point as I tried to match everything he bought. Like I said on the other thread, I have projects that I would like to do, now that James has shared his pearls of wisdom, and I have spares of everything for the future.  I had purchased a spare encoder to play with, and I am thinking of mounting that on the lead screw coupled to a readout just till this plays out.

Johnboy251, 2020-05-02:
Hi Jan...

How are you? I sincerely hope you’re all keeping well and away from this virus. I been following developments in the States and the other parts of the globe. It’s not looking good... just need to ’weather’ this out the best we can and stay away from this invisible invader.

I’ve only pick up on a couple of the messages via Github. I have to read more of the thread - I’ve been doing some maintenance on the lathes headstock - a long story as they usual are, I notice an error message flashing up on the VFD  inverter that i have connected to the lathe motor. It was flashing up the ‘stall’ error. After investigation I determined it was the headstock bearings. I came up  with the conclusion there was swarf (chips) which had got into the roller bearings which proved to be correct. After washing the bearing out then ultrasonically cleaning, repacking with good quality grease then reassemble - all is good once more. These things always take longer than you imagine. 

Regards
John

JonWoellhaf, 2020-05-06:
Scubamachi,

I analyzed your data several different ways, and I can't get my computations to match your observations. I'm very interested in understanding what's happening.

You state in the other thread (can it be combined with this one some way?) that with the spindle set to 360 rpm, the controller set to 12 TPI, and the driver set to 1000 microsteps per revolution, you measured the leadscrew turning at 118 rpm. Correct?

Please confirm you had these settings in Configuration.h:
#define STEPPER_MICROSTEPS 3
#define STEPPER_RESOLUTION 1000

Thanks!

Jan (Yan)

Scubamachi, 2020-05-06:
Hi Jan,
Those are my  settings in Configuration,h: . Just to refresh from thread #70, my lathe rpm and the rpm on the LedKey match. I ran the lathe at 360 rpm and chose 12 TPI, (which is a match for my lead screw thread of 12 TPI), and measured 297 rpm at the lead screw with the default setting (which I now believe is 400 pulses per revolution). I then tried 800 pulses and got a lead screw rpm of 148, and then 1000 pulses and got a lead screw rpm of 118. I called Automation Technologies and he could not answer why my KL5080 has DIP switches, and it is different from what is pictured. I ordered the communication cable from him just to have it for the future. I started installing a new Nema 24 motor with 1.8 step angle and a 2HSS60 hybrid servo driver last night and will have it done tonight and tested. Hopefully this will cure my issue. I am saving the Nema 23 1.2 step angle motor for a power feed project on my vertical mill. I have a spare encoder and TI board and want to set it up on the bench,  just to see if I can get to the bottom of the current issue. I am still thinking James had used that exact servo motor (KL23-2N-1000) with the 1.2 step angle in his videos. I tried to copy everything he used exactly to make my life easier. I could be wrong though and maybe he was using a 1.8 step angle all along. The link he posted in his video for the motor/driver was no longer available on E Bay when I went to purchase it,  so I went with Automation Technologies. I am still watching the videos again to see if I can find where I saw(KL23-2N-1000). I very much appreciate you helping me out with this, and I would like to find the issue if possible. I will keep you posted of how the new motor works out.

Scubamachi, 2020-05-06:
Also I verified my 3:1 ratio with having a 20 tooth pulley on the motor and a 60 tooth pulley mounted on the leadscrew. Just trying to cover all the angles.

JonWoellhaf, 2020-05-07:
Here's my analysis. Please tell me what I'm missing!

Facts:
	1) Leadscrew is 12 TPI.
	2) Spindle is rotating at 360 rpm as measured by independent tachometer.
	3) Control display reads 360 rpm.
	4) Control shows 12 TPI selected.
	5) Configuration.h has #define STEPPER_MICROSTEPS 3 and #define STEPPER_RESOLUTION 1000.
	6) Your hybrid stepper motor has 1.2 degrees per step. That equals 300 steps per revolution.
	7) Driver is set to 1000 microsteps per revolution.
	8) The servo motor drives the leadscrew through a 3 to 1 reduction.
	9) The leadscrew turns at 118 rpm.

Speculation:
	The driver is designed to run a motor with 1.8 degrees per step. That equals 200 steps per revolution.

Analysis:
	1) With a 12 TPI leadscrew, the leadscrew must turn at spindle rpm to cut a 12 TPI thread. Since the spindle is turning at 360 rpm, the leadscrew must turn at 360 rpm.
	2) The ELS software expects 3000 pulses will turn the leadscrew one revolution.
	3) The driver sends 3000 microsteps to the motor expecting that will cause it to turn three full revolutions.
	4) If the motor had a 1.8 degree step angle, 3000 microsteps would cause it to turn three full revolutions, and because of the 3:1 reduction, the leadscrew would turn at spindle rpm, as required to cut a 12 TPI thread.
	5) Irrespective of microsteps, a 1.2° motor will rotate 2/3 as far as a 1.8° motor, if connected to the same driver.
	6) 3000 microsteps sent to a 1.2° motor will cause it to turn two full revolutions, not three. That's because 1.8° is 200 steps per revolution and 1.2° is 300 steps per revolution.
	7) Since the motor turns 2 revolutions, not three, the leadscrew will turn 1/3 times 2 or 2/3 of a revolution, not one revolution.
	8) Two-thirds of 360 rpm is 240 rpm.

Conundrum:
	Under the prescribed conditions, the leadscrew turns at 118 rpm, not 240 rpm, as predicted. Note 118 is very close to one half the predicted rpm. What causes the further 2 to 1 reduction?
	Is it possible the driver thinks the motor encoder has more or less than 1000 steps per revolution? How can we know, since we have no manual for the KL5080H. Or do you have its manual?

Scubamachi, 2020-05-07:
I downloaded the KL5080H manual off Automation Tech's website but because of the issue with my stepper having DIP switches and no - in lettering, I'm not sure what driver it truly is. I completed the install of the new Nema 24 motor - stepper package tonight. I set the pulses to 1000, and the spindle rpm and leadscrew rpm match perfectly when I select 12 tpi. I cut a sample thread also to verify. 
[image]. I still plan on using the 1.2 step angle servo for a power feed project for my mill. I do want to thank you for all the time you have spent with me on this. I very much appreciate your expertise.

JonWoellhaf, 2020-05-07:
Excellent! Good news.

I guess the cause of your initial problem will have to remain a mystery, unless someone sees what I'm missing.

Thanks for the kind words, but I'm afraid my "expertise" is in its infancy. I learn something new from every problem that's presented here, so thanks for advancing my education!

Jan

Scubamachi, 2020-05-07:
I do plan on setting up old motor and driver on bench with spare parts I bought and seeing if I can find the problem. Maybe once I get communication cable and can see drive parameters, the issue might be there. I plan on spinning encoder with a variable speed drill or equivalent to simulate lathe running and will try to achieve the right lead screw rpm. Will let you know my findings.

Scubamachi, 2020-05-07:
This project really tested my abilities and I have gained alot from seeing it through. The trouble shooting part of it was very interesting. I read all the other issues both open and closed and learned from them as well. On my lathe I have a digital rpm readout by factory and I used a 20 dollar tachometer from Amazon with proximity sensor ( super glued magnet) to lead screw pulley face. I held the probe close and got good readings. I verified rpm's with a laser unit as well. If others have similar issues I think its valuable to set to threading and tpi of the machine's leadscrew, and get the spindle and leadscrew numbers to compare to 1 to 1. Makes things way faster than scratch passes each time. With an extra magnet someone could get both rpm's accurately.

JonWoellhaf, 2020-05-07:
Scubamachi,

I have a new thought.

If the leadscrew rotates at 120 rpm instead of 360, then it needs to turn three times faster. That means the servo has to turn three times faster. So change #define STEPPER_MICROSTEPS from 3 to 9 and leave #define STEPPER_RESOLUTION at 1000.

If the leadscrew rotates at 118 rpm, then it needs to turn 3.051+ times faster. We can't simply change microsteps, because that value has to be an integer.

However, note that in every case in the ELS software, the product of STEPPER_MICROSTEPS and STEPPER_RESOLUTION is used. So STEPPER_MICROSTEPS 3 and STEPPER_RESOLUTION 1000 has exactly the same effect as STEPPER_MICROSTEPS 1 and STEPPER_RESOLUTION 3000.

Therefore, we can set STEPPER_MICROSTEPS to 1 and STEPPER_RESOLUTION to 9153 and get to 99.995% of the speed we want -- close enough, I think. (360 / 118) * 3000 = 9152.542+

To do this, the test in SanityCheck.h for STEPPER_RESOLUTION will have to be changed. I don't know why James limited it to 2000, but he probably had a good reason.

It will be interesting to see if this works!

In my code, I replaced STEPPER_MICROSTEPS * STEPPER_RESOLUTION with PULSES_PER_LEADSCREW_REVOLUTION, because that's more clear to me.

Johnboy251, 2020-05-09:
Hi Jan & Scubamachi.
Any update on this problem Scubamachi?

Scubamachi, 2020-05-11:
Morning,
I had to order another power supply and when it arrives, I will do some bench testing to see if I can pinpoint the issue. The new motor and driver works very well. 
[image]

JonWoellhaf, 2020-05-11:
Nice! You won’t find that at Home Depot! 

From: Scubamachi 
Sent: May 11, 2020 07:41
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] TPI not accurate (#69)

Morning,
I had to order another power supply and when it arrives, I will do some bench testing to see if I can pinpoint the issue. The new motor and driver works very well.

clough42, 2020-10-11:
I just have to comment...that bolt looks awesome!

It reminds me of a video that Adam Savage uploaded a while back:  https://www.youtube.com/watch?v=98MCz9gQaiE

I cringed when he tried to explain the threading gearbox, though.  It looked complicated.  :)

ex-caliper, 2021-07-26:
Hi Everyone,

I have an issue and I hope it is related to this one. 

Boxford BUD lathe with 3mm metric leadscrew. 
ELS Version 1.4.00

I have used James’s ELS since it’s inception but always, when cutting metric threads, used the TDI to engage the half-nut rather than keeping it constantly engaged.  

Anyone who is familiar with this lathe will know that the Metric Thread Indicator chart is used in conjunction with the TDI in position 1(20T) or 2(21T). 

e.g for a 1.50 pitch, engage on either A or E with the TDI in position 1.

[image]

[image]

I was then presented with the challenge of an 11/16” x 16TPI thread. Having watched James’s video https://www.youtube.com/watch?v=10Ketpqq-eM&list=PLDlWKv7KIIr90ZZ7Zqt-ge5nVVdS3WVgg&index=7
whereby the reciprocal is true, I set the ELS to 16TPI and kept the half-nut engaged all of the time to cut the thread.

Hereby lies the problem, I cannot get any consistency with tracking the thread. I hope that my understanding of the concept is correct, in that keeping the half-nut engaged it is ultimately the relationship between the spindle speed and the feed rate of the saddle driven by the leadscrew all controlled by the ELS.

If anyone has had a similar problem, I would be grateful to know what is going wrong please.

Apologies if I have missed the obvious.