clough42/electronic-leadscrew #70: Sorry, did not mean to close #69 after posting pictures, signed Newbie.

Scubamachi, 2020-04-30:


Scubamachi, 2020-04-30:
[image]
[image]

Scubamachi, 2020-04-30:
I have the project up and running on my G0752Z lathe. When I attempt to thread 13 TPI it comes out about 16TPI, and when I thread at 24 TPI it comes out around 32 TPI. I know the lead screw is set one to one. My rpm on the machine matches the readout on the LED KEY. I am using the servo package from Automation Technologies with the KL 5080H driver and the KL23-2N-1000 servo motor. I am using the 3 to 1 configuration on the pulleys. I am using the stepper resolution of 1000 and the stepper microsteps are set to 3 for the 3:1 reduction. I have tried the default DIP switch setting on the KL5080H along with trying the settings at 800 and 1000 with the DIP switches. I am a little confused that my KL 5080H has DIP switches and the pictured one on Automation's site does not. My KL5080H has a step angle of 1.2 degrees which seems to differ from most that are 1.8 degrees. Appreciate any help I can get with this an a big shout out to James for putting this project together.

JonWoellhaf, 2020-04-30:
Hi, Scubamachi.

I am quite perplexed!

I see your servo driver is a KL5080H. Please note the missing hyphen. The driver James uses is a KL-5080H.

The KL<dash>5080H does not have  switches to set the pulses/rev. Rather, it is programmed via its RS-232 port.

I have been unable to find a manual for the KL<no dash>5080H.

Where did you get it?

Jan(Yan)

JonWoellhaf, 2020-04-30:
Another thought:

You wrote, "When I attempt to thread 13 TPI it comes out about 16TPI, and when I thread at 24 TPI it comes out around 32 TPI."

My first thought was 16 is about 25% more than 13 and 32 is about 25% more than 24.

If both were 25% more, then 13 would be 16.25 and 24 would be 30. If you didn't measure very precisely, it's conceivable you could mistake 16.25 for 16 and 30 for 32. 

Note that 1000 is exactly 25% more than 800. So with the switches set to 800, the leadscrew will turn a bit too slow and the threads will be a bit finer.

Your photo shows switches 1 through 4 are ON.

The chart on the driver shows that is the default configuration, but gives no pulses/rev that correspond to that setting.

You said you tried the settings for 800 and 1000. Did you get the same threading result for both?

If so, I speculate the driver reads the switches only when it's powered up. So please try changing the switch setting to 1000 and cycling power.

Scubamachi, 2020-05-01:
I got home late tonight and did not try threading. Instead I set lathe to 360 rpm at chuck. I chose 12 TPI, and on the default setting (all switches on) I recorded 297 rpm at the lead screw. Changing to 800 pulses per revolution (#1 DIP off) I got 148 rpm. Switching to 1000 pulses per revolution (DIP #4) off I got 118 rpm. I powered everything down prior to moving switches, and then booted up and chose the 12 TPI setting.  I figured measuring lead screw rpm at this point is a more accurate approach. I was just using a thread gauge on prior threads cut, and best guessing fit. 
 I am still curious if the step angle of my motor being 1.2 degrees instead of 1.8 degrees mentioned by James in videos is a factor as well, and can adjusted in programming.   Great catch on the missing - in part number. I have not had great luck with AutoTech Inc. I received the wrong encoder cable from them, and already asked about issues with my KL 5080H having DIP switches and all I got was attitude of why I waited so long to contact them, being I purchased the package back in Jan. I made my own encoder extension cable, and would just purchase the proper KL-5080H from another source if I have to. I very much appreciate the help with this, as it as been an awesome project so far, testing and challenging my abilities. James's in depth videos have opened my eyes to other projects like building  my own power feeds for my vertical mill next.

JonWoellhaf, 2020-05-01:
I like your idea of measuring spindle and leadscrew rpm.

Just thinking out loud here:

The spec sheet for your G0752Z lathe says the leadscrew is 12 TPI.
So to cut a 12 TPI thread, the spindle and leadscrew must turn at the same rpm.
You set the spindle to 360 rpm, and observed the following leadscrew rpms:
    800 ppi setting, 148 rpm,
    1000 ppi setting. 118 rpm.

I think the step angle of 1.2 degrees, rather than 1.8 degrees is definitely a factor.
    360 / 1.2 = 300 steps per revolution.
    360 / 1.8 = 200 steps per revolution.

Note that in Configuration.h, James has #define STEPPER_RESOLUTION 200. That corresponds to 1.8 degrees per step. He also has #define STEPPER_MICROSTEPS 8. So the software will send 1600 steps per revolution.

I think STEPPER_RESOLUTION should be set to 300, for a motor with 1.2 degrees per step. Unfortunately, your driver does not have a setting that's an integral multiple of 300.

I'm sorry to conclude that your driver and motor combination are not compatible with James' ELS software.

I hope I'm missing something.

Scubamachi, 2020-05-01:
Again thanks for all your help. I ordered a NEMA 24 1.8 step servo motor and a driver this morning to replace it from Amazon. I can use the 1.2 step motor on a upcoming power feed project for my mill. It should be an easy swap out and I only have to bore the drive pulley up to 8mm from 1/4". I'll post the results of how I make out after it arrives.

JonWoellhaf, 2020-05-01:
Scubamachi,

If you ordered the same driver James uses, you will want to also order the RS-232 programming cable.

Then, unless you have a computer with an RS-232 port, you will want to order an RS-232 to USB converter.

The programming software is free.

The hardware of the KL-5080H must be a very old design. I can't imagine a new design that wouldn't have a USB interface.

I would love it if someone smarter then me -- James Clough, are you listening -- would design a hybrid stepper driver using the same TI board the ELS uses.

I hope I haven't led you astray! 

Jan

forrestlw, 2020-11-26:
1.2 degree is a 3 phase motor. 1.8 degree is a 2 phase motor. I thought That James used a 3 phase 1.2 degree motor and driver? The 3 phase motors and driver set/kits are more expensive and hard to find. They are more precise and accurate than a 2 phase motor and driver. 
Forrest