clough42/electronic-leadscrew #71: Hybrid motor wires don't match diagrams or videos

loeschr, 2020-05-19:
I purchased the hybrid motor / driver from Automation Technology, as most did.  However, when I got around to wiring up for bench testing, I noticed that the wiring diagrams here along with the wiring diagrams and videos at Auto Tech Inc, don't match my motor.  All documentation show wiring of the 3 phases to be Brown (U), Blue (V), and Black (W).  However, my motor has a 4-wire cable with one of them (Red) cut off short with the outer wrap.  The 3 remaining wires are Yellow, Blue, and Black.  I figured the Yellow was really the Brown and wired the motor up accordingly.  The controller has the fault light lit.  I tried to contact Auto Tech Inc, but they were no help.  Has anyone used a motor with wires similar to mine and gotten it to work?  If so, how did you wire the 3 phases?
[image]

loeschr, 2020-05-19:
This is the same issue discussed in #55, but no answer was given.

Vrossi050, 2020-05-19:
What are the model numbers of both driver and and stepper motor?

JonWoellhaf, 2020-05-21:
My notes show Clough's current implementation uses the Automation Direct NEMA 23 Hybrid Servo kit (closed-loop stepper system), which comprises "one Hybrid Servo Driver KL-5080H, one NEMA23 KL-23-2N-1000 Hybrid Servo Motor with one Encoder 1000 line"

Based on your experience, loeschr, and Scubamachi's (Issue #69) experience with Automation Direct, I will NEVER BUY ANYTHING FROM THEM.

Scubamachi, 2020-05-21:
Morning,
 I have the same package and mine works with Black on "W", Blue on "V", and Yellow on "U". After dealing with Automation Technology, and receiving poor customer service with wrong cabling and other issues, I will not purchase anything from them again, even if it is thru Amazon.
 Jon, just for the record Automation Direct is a different vendor, which I have purchased from. On a side note, I did have trouble getting the threading to work with this motor as it is a 1.2 step degree angle, and the norm is a 1.8 step degree. I did wind up going with a different motor/driver combo on the leadscrew project in the end, and I am still going to see on the bench, if I can recreate my issue to define where the actual problem is with using the 1.2 step motor.
loeschr, on your driver is it labeled KL-5080H with no DIP switches, or is it KL5080H with DIP switches? This is one of the issues that Automation Tech could not explain to me, why I did not receive what was pictured on their website. I had a similar issue with them over a different package last year as well.
 I'd be happy to share any info I have to help others, and just wanted to say thanks again for all the help to date, from folks on here.

loeschr, 2020-05-21:
I finally got mine to work with substituting the Yellow wire for the Brown wire to the U phase.  I didn't (initially) want to wire it up wrong not knowing if I could damage the driver by mis-wiring the phases.  I'm not knowledgeable about 3-phase stepper motors to know what would happen if the phases were wired incorrectly.

My driver is the KL-5080H with DIP switches.  To set the DIP switches for 1000 steps (on-on-on-off).  Not sure what folks would do if theirs didn't have dip switches.  My dealings with Automation Technology has not been good.  I sent them pictures of the wires, label on motor and cover of driver. All they said is they would look into it.

loeschr, 2020-05-21:
Just received this from Automation Technology - they did respond.  Their information is being provided for future folks that may have the same issue and not sure what to do.
[KL23-2N-1000.pdf](https://github.com/clough42/electronic-leadscrew/files/4662708/KL23-2N-1000.pdf)

closed 2020-05-21 by loeschr

Scubamachi, 2020-05-21:
loesschr,
Let us know if you have issues with threading not being accurate once you get up and operating, as I have already been down this road. You can also check my threads #69 and #70.

JonWoellhaf, 2020-05-21:
Scubamachi,

Thanks for the heads up on Automation Technology and Automation Direct being different companies. I had conflated them.

Jan (Yan)

gusosborne, 2020-05-21:
Yes- I had as well!