clough42/electronic-leadscrew #73: Feature request / question: Depth stop

jchurc16, 2020-05-27:
Hey,

I've been following this project and have a question regarding a depth stop feature. So I'm thinking, the controller knows how many steps it takes to move the leadscrew through 1 revolution, therefore knows how far the carriage moves based on that. Is it possible to have a mode which is decoupled from the spindle rpm and purely deals with depth calculations? I'm thinking this would be handy for facing to a shoulder, internal boring, drilling and possibly threading if this can be tied to spindle rpm too.

I've not seen any discussion on this and would be interested to know if this could be a feature, though with the hardware as it is I'm not sure if this could be added.

gusosborne, 2020-05-27:
Check out this thread:

https://github.com/clough42/electronic-leadscrew/issues/10

jchurc16, 2020-05-27:
Thanks, I hadn't considered the disengagement of the leadscrew, I will follow that thread in case of any updates.

closed 2020-05-27 by jchurc16