clough42/electronic-leadscrew #74: Launchpad flashed, code runs, nothing happens, display (almost) dead

luxige, 2020-05-28:
First, I'll say my computer experience is of the point n click type.
I'll skip the ordeal I went through to get to this point, but after many hours of struggle I flashed the Launchpad and the code seems to run. At least, it gets to the breakpoint. When I hit Resume, nothing happens further. The display board and LEDs stay dark. When I click on the control buttons, a tiny LED to the left of the digital array lights briefly, but only some of the time. Mostly the board is dark.
When I turn the rotary encoder shaft, there is no response from the servo.

Advice on troubleshooting will be appreciated. Here's the plan so far:
First I'll disconnect everything and check the cables to make sure no pins are misconnected. Also will check the onboard switches.
I'll remove the booster board and peripherals and power the Launchpad from USB with the jumpers back in place. *** Is there a simple test program I could DL and run? ***
Then I'll put the Clough42 booster back on, with the jumpers off. Then add the display board with a short test cable.

I have ordered a new display board to try in case I have damaged the current one.

JonWoellhaf, 2020-05-29:
I think the first step of your plan is good: check wiring and LAUNCHXL switch positions. Also check the jumpers.

There is no LAUNCHXL test program I know of, so after checking the switch positions, I suggest you skip the second step and replace the Boost board.

With power connected to the Boost board, do the two LEDs labeled 3V3 and 5V light?

Johnboy251, 2020-05-29:
Hi luxige & Jan

Looks like S9 on the TI LauncePad is set incorrectly..
Hope you're all well.
John

JonWoellhaf, 2020-05-29:
John, I concur. Good thinking!

luxige, the TI Launch Pad Development Kit PDF, on page 12, shows the default position for S9 is 1, which routes GPIO32 and GPIO33 to the Boost board. If it's in the other position it puts a CAN (Controller Area Network -- automotive, etc.) bus on J14.

We have seen at least one board with this switch in the wrong position. If it is, the display will not work.

S9 is under the Boost board.

Please let us know what you find.

Jan

luxige, 2020-06-02:
Thanks for your help, Jan and John!
I hope you guys weren't thinking I was ignoring your input, I just had to walk away from this for a while. I was thiiiis close to dragging the project out to a remote field and giving the whole mess an Office Space-style geek beatdown. Also, waiting on a new display board, which arrived today.

I was aware of the S9 default position. Knowing the meaning of the word "default," I simply left it alone, never even touched the tape. Maybe TI doesn't know what a default is; you were right, the switch was in CAN bus position. So, now the encoder is driving the servo!

Unfortunately, the display is still out other than the small LED to the left of the digits. That light had been intermittent, which I traced to a short between the CLK and GND pins in one of my connectors. That's fixed now so the LED is on steady, but that's it. I checked continuity from the display board header pins all the way to the first components on the booster board (C5, C4, C3, C6, and +5v). The first display and the replacement respond the same, or rather they don't.

Next? New booster board?

If either of you are in the Springfield MA area, let me know so I can get you that beer I owe you.

JonWoellhaf, 2020-06-02:
Luxige,

Progress! Good. Do you have an oscilloscope so we can do some remote troubleshooting?

Jan

luxige, 2020-06-03:
Hi Jan,
I will see if I can borrow one.
They're for monitoring the status of the local oscillo population, right?

JonWoellhaf, 2020-06-03:
Yup. 

From: luxige 
Sent: June 2, 2020 21:27
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad flashed, code runs, nothing happens, display (almost) dead (#74)

Hi Jan,
I will see if I can borrow one.
They're for monitoring the status of the local oscillo population, right?

luxige, 2020-06-03:
While I'm asking around after an oscilloscope, I went ahead and ordered a new Clough booster board.

Re: 'scope, anyone in the Springfield, MA area able to help out?

Johnboy251, 2020-06-03:
lukige... if you were in the UK I could give you scope! - I recently bought a ‘digital’ Tektronix scope - now I have another ‘spare’ kicking about my electronics workspace! 😳

luxige, 2020-06-03:
Thank you, John.
And many thanks to all for the help. I have stumbled across the rest of the problem/answer. 
When I reset S9 from the incorrect (non-default) position, I didn't flip it all the way. It was about in the middle.
It's working like a champ now.

closed 2020-06-03 by luxige

JonWoellhaf, 2020-06-04:
Good news, luxige! Must have been great fun to see it finally working. Take care.

Jan