clough42/electronic-leadscrew #75: Arduino Mega

WildPoYo, 2020-05-29:
Hello James, 

I know in earlier videos you mentioned that Arduino might not be suitable for this project but do you think if there was enough interested would you adapt the code to Arduino platform? For me it is easier to work on Arduino platform. There might be other people sharing the same interest.

DavidThijs, 2020-06-19:
I believe that the reason why this project is developed on a non-arduino platform is that the regular arduino's aren't capable to support the pulse train that is generated by the position encoder of the spindle. With a 1000 or 1024 resolution/rev , you might need a separate quadrature decoder to handle the signals as I presume that even when using interrupts it will still be not good enough and eat away all cpu resources. Maybe with a Due, but then you also need 5-3.3V logic converters.

clough42, 2020-06-23:
I ruled out the mega due to its performance limitations and lack of a
hardware quadrature decoder.

Building an electronic leadscrew with an ATMega device would require an
entirely different approach, likely based on trying to estimate and match
speed instead of directly feeding back position.

James

WildPoYo, 2020-06-24:
I did not really asses the technical requirements of it, but both of you are right regarding to capacities of Mega. I think TI board is best suiting hardware with given circumstances.  Thank you for taking time and giving a proper answer!

closed 2020-06-24 by WildPoYo