clough42/electronic-leadscrew #76: EK-TM4C1294XL Launchpad

dboe, 2020-06-18:
Already have a EK-TM4C1294XL Launchpad, which is same dual BoosterPack XL connectors and runs at 120MHz and has 1MB of flash. Can the software be recompiled for this Launchpad?

dboe, 2020-06-18:
FYI it also has a Quadrature Encoder interface on either the Booster Pack 1 interface or the 100 pin interface on the side

Johnboy251, 2020-06-18:
Unless anybody can advise otherwise - I can’t see any reason that jumps out at me, to run the ELS software compiled for this development board. Presumably the C++ script will compile once the TI Code Composer Studio has been configured to your board. There may be some I/O re-configuration in the original code. I don’t have any experience of your TI board but it must be worth a try. 🤞

John

dboe, 2020-06-18:
Looking at it more, the TM4C1294 has fpu, quadrature encoder, same booster pack pinouts, but may missing level shifting on the encoder port from 5 to 3.3v, but that is no biggy. Could be a simple board on the second boosterpack

Johnboy251, 2020-06-18:
Sound good - let us know how you get on....👍

closed 2020-10-11 by clough42