clough42/electronic-leadscrew #78: servo does not spin and no readout on cotrol panel for rpms.

tombmatt, 2020-08-08:


JonWoellhaf, 2020-08-10:
I suggest you start by disconnecting the motor controller.

Just so I understand:

When you apply 5 volts to Clough's boost board the display lights and goes through its greeting, "CLOUGH42" then "ELS-1.1.0". Correct?

Does the display then show blank, blank, blank, 0 (in the left-hand four seven-segment displays) and blank with decimal, 005 in the right-hand four displays?

tombmatt, 2020-08-10:
Yes and yes, that is what the display reads.

From: JonWoellhaf
Sent: Sunday, August 9, 2020 8:13 PM
To: clough42/electronic-leadscrew
Cc: tombmatt; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

I suggest you start by disconnecting the motor controller.
Just so I understand:
When you apply 5 volts to Clough's boost board the display lights and goes through its greeting, "CLOUGH42" then "ELS-1.1.0". Correct?
Does the display then show blank, blank, blank, 0 (in the left-hand four seven-segment displays) and blank with decimal, 005 in the right-hand four displays?

JonWoellhaf, 2020-08-10:
Do you have an oscilloscope?

tombmatt, 2020-08-10:
No. if I bought one of the small inexpensive ones would that work?

From: JonWoellhaf
Sent: Monday, August 10, 2020 11:58 AM
To: clough42/electronic-leadscrew
Cc: tombmatt; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Do you have an oscilloscope?

JonWoellhaf, 2020-08-10:
I think a small inexpensive scope will work. Please let me know which one you're looking at.
I'd like you to check for pulses on 1A and 1B, (on J12 of the TI board) when you rotate the encoder by hand.
I can see easily see the pulses on 1A and 1B with my scope bandwidth limited to 20 MHz.
The pulses on 1I (digit one, letter India) are harder to see and require triggering on the signal. But I believe 1I is unused for the ELS. Someone please correct me if I'm wrong about this.

tombmatt, 2020-08-10:
I ordered the scope and probe below
https://www.amazon.com/gp/product/B01ABWNUZS/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
https://www.amazon.com/gp/product/B0852Y2TY1/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

From: JonWoellhaf
Sent: Monday, August 10, 2020 3:29 PM
To: clough42/electronic-leadscrew
Cc: tombmatt; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

I think a small inexpensive scope will work. Please let me know which one you're looking at.
I'd like you to check for pulses on 1A and 1B, (on J12 of the TI board) when you rotate the encoder by hand.
I can see easily see the pulses on 1A and 1B with my scope bandwidth limited to 20 MHz.
The pulses on 1I (digit one, letter India) are harder to see and require triggering on the signal. But I believe 1I is unused for the ELS. Someone please correct me if I'm wrong about this.

tombmatt, 2020-08-10:
I have been reading the other issues and have discoverd that i received from Auto tech the same servo and driver that Scubamachi  received. and i bought it in january like he did. they must have been looking to clear old inventory the picture from Amazon showed the KL-5080H but that isnt what i received. would still like to see why it isnt turning but will have to purchase the correct one eventually.

Scubamachi, 2020-08-11:
tombmatt,
The motor and controller I purchased did in fact work, but not accurately based on the 1.2 step angle and not the 1.8 angle. I have since started installing them on a power feed for a vertical mill.  You are in good hands with Jon helping out. I'm more than willing to share any info I have. Good luck

tombmatt, 2020-08-11:
Scubamachi
Thank you for your reply. I knew yours was running just that the 1.2 degrees of rotation made it unworkable for Clough42’s project. I just want to find out why mine wont spin at all. Will eventually buy the proper controller motor.

From: Scubamachi
Sent: Tuesday, August 11, 2020 9:04 AM
To: clough42/electronic-leadscrew
Cc: tombmatt; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

tombmatt,
The motor and controller I purchased did in fact work, but not accurately based on the 1.2 step angle and not the 1.8 angle. I have since started installing them on a power feed for a vertical mill. You are in good hands with Jon helping out. I'm more than willing to share any info I have. Good luck

Scubamachi, 2020-08-11:
[image]

Scubamachi, 2020-08-11:
I purchased this one off Amazon for the ELS project, and I am happy with it.

tombmatt, 2020-08-12:
I received the scope but only have alligator clip lead. the other one would not connect didn't have bnc

JonWoellhaf, 2020-08-12:
Are you able to see the test signal on the scope?

The alligator clip should work fine. Connect black to one (or both) of the terminals labeled GND on JP7 on the TI board.

With the encoder connected, put a short piece of thin wire in the red clip and probe 1A and 1B.

May need a helper to spin the encoder.

tombmatt, 2020-08-13:
Yes got test signal
Here are pics of 1A and 1B;

Don’t know why 1B wave is higher than 1A same settings on scope.

Cat5 used as wire from daughter board to driver orange is + orange white -blue + blue white – green + green white -  on driver orange is pul+
Orange white is pul_  pul + is jumpered to the other +connectios blue is dir- blue white is en- and green white is al+ green is al-

From: JonWoellhaf
Sent: Wednesday, August 12, 2020 6:19 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Are you able to see the test signal on the scope?
The alligator clip should work fine. Connect black to one (or both) of the terminals labeled GND on JP7 on the TI board.
With the encoder connected, put a short piece of thin wire in the red clip and probe 1A and 1B.
May need a helper to spin the encoder.

JonWoellhaf, 2020-08-13:
Thomas,
For some reason the waveform images are missing from the email and from GitHub. Please repost.
The wiring is for the servo controller, right?
Please show the wiring for the encoder.
I suggest you unplug the servo controller from the boost board's J2 while we're troubleshooting.
Jan (Yan)

tombmatt, 2020-08-13:
1B

1A

From: JonWoellhaf
Sent: Thursday, August 13, 2020 2:53 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Thomas,
For some reason the waveform images are missing from the email and from GitHub. Please repost.
The wiring is for the servo controller, right?
Please show the wiring for the encoder.
I suggest you unplug the servo controller from the boost board's J2 while we're troubleshooting.
Jan (Yan)

JonWoellhaf, 2020-08-13:
Bummer! Still no images here (GitHub) nor in the email. Is the problem on my end? Do you see images?

tombmatt, 2020-08-13:
No my end images were too large trying again
1A

1B

Control panel

Power for motor is not plugged in

From: JonWoellhaf
Sent: Thursday, August 13, 2020 4:45 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Bummer! Still no images here (GitHub) nor in the email. Is the problem on my end? Do you see images?

JonWoellhaf, 2020-08-13:
Still no joy.

tombmatt, 2020-08-13:
i will try here
1A
[image]
1B
[image]
control panel
[image]

tombmatt, 2020-08-13:
Put them on GitHub page

From: JonWoellhaf
Sent: Thursday, August 13, 2020 5:04 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Still no joy.

JonWoellhaf, 2020-08-13:
Success! I can finally see the images. I like that little scope!

The signal on 1B is very likely the problem. 1A and 1B should be the same. (Actually their phase is different, but that's unimportant now.)

Please unplug the servo driver connector at J2 on Clough's boost board.

Please show me how you've wired the encoder to J12 on the TI board.

tombmatt, 2020-08-13:
here are new pics
1B
[image]
1A
[image]
j12
[image]

blue grnd brown 5v+ orange white black

tombmatt, 2020-08-13:
you think the encoder is bad?

JonWoellhaf, 2020-08-13:
I doubt the encoder is bad.
Please try moving the connector to J13 and see what 1A and 1B look like.

JonWoellhaf, 2020-08-13:
Correction: See what 2A and 2B look like.

tombmatt, 2020-08-13:
there is no square wave on the line on either A or B just flat jittery line no effect when encoder is turned.

JonWoellhaf, 2020-08-13:
Interesting! My waveforms are identical. I can't explain that.
With the encoder unplugged, do you see 5 volts on 1A and 1B?
Do you have an ohm meter? If so, with the encoder connector removed, please measure between blue and black, and blue and white. You should get the same reading.
I'll be back in a couple hours.

tombmatt, 2020-08-13:
1A .4 v  and 1B 5.1

From: JonWoellhaf
Sent: Thursday, August 13, 2020 6:26 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Interesting! My waveforms are identical. I can't explain that.
With the encoder unplugged, do you see 5 volts on 1A and 1B?
Do you have an ohm meter? If so, with the encoder connector removed, please measure between blue and black, and blue and white. You should get the same reading.
I'll be back in a couple hours.

tombmatt, 2020-08-13:
Measured 1A and 1Busing ground on jp8 , got .41V on A and 5.1V on B with an Ohm meter. Used meter to read blue and black and blue and white and got the same readings.

From: JonWoellhaf
Sent: Thursday, August 13, 2020 6:26 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Interesting! My waveforms are identical. I can't explain that.
With the encoder unplugged, do you see 5 volts on 1A and 1B?
Do you have an ohm meter? If so, with the encoder connector removed, please measure between blue and black, and blue and white. You should get the same reading.
I'll be back in a couple hours.

JonWoellhaf, 2020-08-14:
OK. I think we are zeroing in on the problem.
Just to confirm: With the encoder DISCONNECTED, you measure 0.41 volts on 1A and 5.1 volts on 1B?
It looks like the signal on 1B is bad in the first set of photos, but 1A is bad in the second set. Could you have misidentified the photos?
Based on the voltage measurements, 1A should be bad.

tombmatt, 2020-08-14:
yes i could have crossed them

JonWoellhaf, 2020-08-14:
OK. Like I've been fond of saying for decades, "the time spent double-checking is never wasted." :)
Please confirm that with the encoder disconnected, you read 0.41 volts on 1A and 5.1 volts on 1B.

tombmatt, 2020-08-14:
okay retook pictures
1A
[image]
1B
[image]

tombmatt, 2020-08-14:
yes the encoder dis connected 1a is .4 and 1B is 5.1

JonWoellhaf, 2020-08-14:
OK.
We have positively confirmed that the 1A input on the TI board is bad.
This likely has nothing to do with the encoder.
Is it possible that you inadvertently connected 24 volts to that input?

Anyway, the TI board has a second encoder input, so let's see if we can get that one working.
To confirm once again, with nothing connected to J13, what voltage do you read on 2A and 2B?

JonWoellhaf, 2020-08-14:
Please look at the jumpers in the lower left of the TI board, the ones below the red LED.
In your photo, it appears that the bottom jumper is connected.
That's JP2, and it must be disconnected.
Please confirm.
Note the jumpers are (confusingly) labeled JP1, JP3, and JP2, top to bottom.
I have all three of mine disconnected.

tombmatt, 2020-08-14:
2a 1.36  2b .48

From: JonWoellhaf
Sent: Thursday, August 13, 2020 11:49 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

OK.
We have positively confirmed that the 1A input on the TI board is bad.
This likely has nothing to do with the encoder.
Is it possible that you inadvertently connected 24 volts to that input?
Anyway, the TI board has a second encoder input, so let's see if we can get that one working.
To confirm once again, with nothing connected to J13, what voltage do you read on 2A and 2B?

tombmatt, 2020-08-14:
All 3 jumpers are off 

From: JonWoellhaf
Sent: Thursday, August 13, 2020 11:58 PM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Please look at the jumpers in the lower left of the TI board, the ones below the red LED.
In your photo, it appears that the bottom jumper is connected.
That's JP2, and it must be disconnected.
Please confirm.
Note the jumpers are (confusingly) labeled JP1, JP3, and JP2, top to bottom.
I have all three of mine disconnected.

tombmatt, 2020-08-14:
[image]
all disconnedted

JonWoellhaf, 2020-08-14:
Acknowledged, all three jumpers are disconnected.

I just checked the TI board schematic to be sure my memory was correct, and confirmed 1A, 1B, 2A, and 2B all have 1k ohm pullups to +5 volts and 1000pF bypass caps to ground.

To get the voltages you're seeing, something on the TI board has failed. All should read +5 with nothing connected.

Is it possible 24 volts was accidentally applied to them?

I am afraid you need a new TI board.

tombmatt, 2020-08-14:
i dont think 24v were ever close but i could have touched 3 & 5v pads at same time reading power im kinda shakey

tombmatt, 2020-08-14:
guess i get a new ti board

JonWoellhaf, 2020-08-14:
You're very welcome.
Please let us know if a new TI board fixes the problem.
Sorry it wasn't a simple fix.
Anyone else have any thoughts about this?

Vrossi050, 2020-08-14:
Are both switches set on S3 and S4 on the TI board per the wiki page picture?

tombmatt, 2020-08-14:
yes

From: Vincent Rossi
Sent: Friday, August 14, 2020 10:55 AM
To: clough42/electronic-leadscrew
Cc: Thomas Matthews; Author
Subject: Re: [clough42/electronic-leadscrew] servo does not spin and noreadout on cotrol panel for rpms. (#78)

Are both switches set on S3 and S4 on the TI board per the wiki page picture?

tombmatt, 2020-08-18:
ordered and received new TI panel. programed and installed other bits and It Works. Thank You JonWoellhaf and all others that helped.

JonWoellhaf, 2020-08-18:
Great news, Thomas!
I hope the rest of the project goes smoothly.

clough42, 2020-10-12:
Thanks, @JonWoellhaf .  You're a real gem.

closed 2020-10-12 by clough42