clough42/electronic-leadscrew #80: Another question regarding stepper gearing.

Rjaxe, 2020-08-12:
My old lathe has a 4TPI leadscrew which seems low by todays norm, I would like to run the NEMA24 stepper with a 3:1 ratio to keep its speed up a bit. What is the correct way to configure the software to accommodate this arrangement? Do I change the steps or micro steps or what. Needless to say I have no previous experience in programming so I need help.

JonWoellhaf, 2020-08-16:
Since no one who knows the answer has chimed in, I'll take a stab at it.
In Code Composer Studio, open the file Configuration.h.
On line 43 you will see #define LEADSCREW_TPI 12
Change that to 4 and complie and download.

Rjaxe, 2020-08-16:
Thanks JonWoellhaf, all help gratefully received.

davidgleach, 2020-09-21:
Hi Rjaxe,
I had similar challenges.  My little Seig mini lathe has a metric 1.5mm pitch lead screw.  I had printed belts and pulleys to mount a stepper motor to drive the lead screw. As Jan noted all you need to do is when you fire Code Composer look at the Configuration file and make changes.  If you view James Clough video No 10 in the series on making an electronic lead screw, he takes you through the whole process step by step.  In the finish I kept tweaking both the lead screw pitch and micro steps until the lead screw produced an accurate thread.  It was worth the effort as I now can cut very accurate threads - not bad for a cheapy chinese lathe!! 

Cheers

Rjaxe, 2020-09-22:
> <electronic-leadscrew@noreply.github.com>
> <author@noreply.github.com>
> stepper gearing. (#80)

Thanks for your input David, I made up a small test rig to explore this 
issue and I think I have resolved it. As you say tweaking these settings 
does the trick. My major issue with this project is that it is nowhere 
near being fitted to my lathe as I have been commanded to carry out a 
major house redecoration. No prizes for guessing what comes first, but I 
will get there in time.
Cheers

davidgleach, 2020-09-23:
I understand the priority.  We all have these demands on our time
:-)
.

David Leach

Home: 03 5310 6313

Mob:    0416 119 381

Cc: "davidgleach" <davidgleach@iinet.net.au>; "Comment" 
<comment@noreply.github.com>
Sent: 23/09/2020 2:19:16 AM
stepper gearing. (#80)

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus