clough42/electronic-leadscrew #83: Launchpad programming

bacca5419, 2020-08-31:
Hi all. I am havind problems programming the launchpad. I get as far as where it pauses at break 69. I hit the reume button and it appears to do nothing except bringing up the launchpad model and tells me its running. The control panel shows a led power light but nothing else. I need help as I am a dumd arse with electronics and computers alike. Regards Noel from australia.

JonWoellhaf, 2020-09-01:
Hi, Noel.

I'd like to help, but I don't yet understand what you're seeing.

You have Code Composer Studio running and were able to load James' code. Correct?

bacca5419, 2020-09-01:
Hi. Yes I can get right up to to loading the launchpad to the first break point , when i hit resume nothing happens. It wont continue to load. So the control panel wont light up. This  is where it stops. Pics show where it stops.I feel giving up.Its a new computer. Regards  noel

JonWoellhaf, 2020-09-01:
I don't understand what you mean by up to the first break point.
When I load the code there is no break point.
Please tell me step by step what you are doing.

bacca5419, 2020-09-01:
Ok. I have loaded the studio code IDE version 9 from TI.com for the launchpad then I have loaded the code for the clough42 firmware. I get right up to loading the program onto the launchpad with the control panel. When loadind onto the launchpad it stops at the breakpoint in the program ,same as it does when james is loading the lanchpad , however when I click the resume button in the program it does not resume as it does when James loads it. I have followed the same proceedures as james. I dont know if you can see the pics I sent. That shows as far as I can get. I dont know if this helps for your understanding. Regards. Noel. Thanks for your assistance , it is apreciated.

JonWoellhaf, 2020-09-01:
OK. I think I now understand. I reviewed James' ELS video Part 10, and at 24:00 he explains the program always stops in Main when in Debug.

This is the break point you're talking about, right? Where does the 69 come from?

JonWoellhaf, 2020-09-01:
When you say "brings up the launchpad model and tells me it's running, do you mean that you see in the Debug window the text "Texas Instruments XDS110 USB Debug Probe/C28xx_CPU1 (Running)?"

JonWoellhaf, 2020-09-01:
Don't give up, Noel,. I'm certain we -- all of us here -- can help you get it running.

bacca5419, 2020-09-01:
That was the number at the break point. I have just tried deleting the firmware from my com and reloaded it , I loaded the updated version of the firmware code. Still get the same problem but the break point is 77. I guess there is more information on the latest update. I am starting to think there is a problem with the microcontroller launchpad. But I have no idea about electronics. Thats why I try to follow James setup. Regards.

JonWoellhaf, 2020-09-01:
OK. Have you double checked the wiring from the boost board to the LED board?

JonWoellhaf, 2020-09-01:
Mine stops a line number 91.
Can someone else please check and see at what line number theirs stops?

bacca5419, 2020-09-01:
I have checked but I am going to make up a new wire harnes tomorrow. I am sure that isnt it as I have continuity through each wire

bacca5419, 2020-09-01:
Yes to the texas instrument debug probe running

JonWoellhaf, 2020-09-01:
I think it's unlikely that your TI board is defective.

I'd like you to set a breakpoint in Main. This will let us prove the code is running.

Do you want to do that?

JonWoellhaf, 2020-09-01:
I do not see any pictures you sent.

bacca5419, 2020-09-01:
I will do anything to get this working. But how do i set a new break point ? I tell you I am as dumb as when it comes to this sort of thing. I am surprised I have gotten this far. Ok it musnt let me send pics.

JonWoellhaf, 2020-09-01:
It's easy to set a breakpoint.
When you hit the breakpoint at the top of Main, scroll down until you see the line that reads
    for(;;) {

Below that, you'll see
    userInterface.loop();

Mine is on line 143, but yours is probably different -- we can figure out why later, I don't think it's important now.

To set the breakpoint double click the line number. A dot will appear in the column to the left of the line number.

Now, with this breakpoint set, when you run in Debug, the program will stop at the start of Main -- like before -- and when you click Resume it will immediately stop again at the breakpoint you just set.

Cliocking Resume again will let the program run through a pile of code until it returns to that point again. It'll look like it didn't do anything.

So the second time it hits the breakpoint, click the Step Into icon on the tool bar or press F5. This will let you step through the code a statement at a time.

If you are able to step through the code -- which I'm quite sure you will be able to do -- we will know a LOT is working!

To remove the breakpoint click Resume, then again double click the line number.

Do you have a volt meter?

davidgleach, 2020-09-20:
Hi Jon, What was the outcome of your suggestion.  I have exactly the same problem so will try your suggestion. 
  
Thinking the issue was the LED indicating panel I purchased several new panels and fitted one of them with the same result. 

 In despair I buttoned up the whole system and behold it started to work correctly.  It continued to do so for some weeks until I decided to make some improvements to my layout by making a new enclosure for the LED panel.  Guess what - the indicating panel no longer functioned, only the LED on the on the PCB was alight.  All connections checked out with continuity testing.

Once again I replaced the LED indicating panel with a new one  - to no avail.  

I've downloaded and replaced all the the Clough 42 files as well as a new version of Code Composer Studio.  

Recompiled and reflashed the TI board - and still the program will not run.

davidgleach, 2020-09-20:
I have carried out the steps you suggested and the program steps through the loops a few times until it stops again at line 71 WAIT_FOR_SERIAL
Does this indicate a bad connection somewhere?

davidgleach, 2020-09-20:
I shoudl have noted that it stopped at line 71  WAIT_FOR_SERIAL in the SPIBus.cpp file

davidgleach, 2020-09-20:
Problem solved.  It is dodgy connections.  Too quick to blame the electronics!

JonWoellhaf, 2020-09-20:
Hi, David.

I have not heard back from bacca5419.

You wrote, "I have carried out the steps you suggested and the program steps through the loops a few times until it stops again at line 71 WAIT_FOR_SERIAL. Does this indicate a bad connection somewhere?"

No. This is perfectly normal.

Please tell us what the problem with your connections was. That could help someone later.

Great that you solved the problem!

Regards,

Jan (Yan)

bacca5419, 2020-09-20:
Hi guys. Still no go on the leadscrew. Checked all connections , no probs there that I can see. I have ordered a couple of new conntrol panels from China. Shipping is extremly slow. One thing I found on the control board that doesnt seem right to me is that with no wiring connected I am getting a continueity between the gnd and the DIO terminals. Dont know if that makes any sense. Will maybe know more when replacement control panel arrives. Will keep you informed as I learn more. Thanks guys.

JonWoellhaf, 2020-09-21:
Hi, Noel.

How are you measuring continuity between GND and DIO on the control board?

I get 10 meg ohms.

Jan (Yan)

JonWoellhaf, 2020-09-21:
What's your location, Noel?

I have several control boards I bought by mistake. I didn't realize I was ordering a set of two each when I ordered two the second time. So I now have five!

davidgleach, 2020-09-21:
> <electronic-leadscrew@noreply.github.com>
> <comment@noreply.github.com>

Hi Jan
I noticed when I was stepping through the program that Code Composer was 
providing a number of messages (coloured red) that i took to be 
warnings.  The messages were "Cannot load from non-primitive location".  
The messages were all associated with statements from the SPIBus.cpp 
file relating to Input/Output data from the Clough42 board to the LED 
indicating panel.  By trial and error I established that the plug 
positioned on the output pin DIO was not always making a good 
connection.  Replacing the plug fixed the problem.

I appreciate your advice on how to step through the program as that was 
the key information that set me on the right path.

Cheers
David

David Leach

Home: 03 5310 6313

Mob:    0416 119 381

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

bacca5419, 2020-09-21:
Hi Jan. On the control panel ( led board ) with no wiring connected I start red lead + on vcc I move black lead - down the terminals gnd , stb , clk and dio. I use the buzzer set on my meter. Then I go down the line , red + on gnd , black - down the line of terminals. When I have the + on the gnd and - on the dio I am getting a continuity ( buzzer sounds ). Nothing on any other terminals. I do the same test on the  clough42 board. I get no continuity on that board.  If that makes any sense. I have 2 replacement boards coming from china to see it that is the problem. ( slow boat from china I think. )I am not in USA I live in Australia. Regards from Aust

JonWoellhaf, 2020-09-21:
Noel,

Based on your testing, I agree the control board is likely defective. That would explain everything. Hope the "slow boat from China" arrives soon. I, too, had to wait for one of those. :)

Are you using a digital VOM (volt/ohm meter) to make the measurements? If so, please try the same procedure again with the meter set to ohms. With the wires disconnected, short STB, CLK, and DIO to ground. This discharges the input filter capacitors. Then measure resistance from each of those inputs to ground. You should see a few kohms initially and then greatly increasing resistance, settling around 20 meg ohms or so. That's the effect of the capacitor charging.

If STB and CLK behave differently from DIO, it'll be another indication your board is defective.

Jan (Yan)

bacca5419, 2020-10-08:
Hi Jan. Well the new control boards arrived today , plugged one into the microcontroller and low and behold , still nothing. Time to give up I think. Go back to the gears ( at least they work ) I pretty dissapointed. Regards. Noel from Australia.

JonWoellhaf, 2020-10-08:
Yeah, that’s disappointing, Noel.

Do you have an oscilloscope, or access to one?

If not, if shipping from Australia to Utah, USA is not outrageously expensive, I will be happy to look at your setup and get it working. 

Jan

From: bacca5419 
Sent: October 7, 2020 22:47
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad programming (#83)
Hi Jan. Well the new control boards arrived today , plugged one into the microcontroller and low and behold , still nothing. Time to give up I think. Go back to the gears ( at least they work ) I pretty dissapointed. Regards. Noel from Australia.

bacca5419, 2020-10-08:
Hi Jan. No I dont have a oscilloscope or access to one. Even if I did I wouldnt have a clue what to do with it. I am hopeless with electronics and thats on a good day. Thanks for the offer to get it working for me , however I think shipping costs and times would be outrageous. Regards. Noel.

davidgleach, 2020-10-08:
Hi Noel
You might have seen my issues with the control board.  In spite of being confident that every connection was secure, it turned out after much trial and tribulation there was one board connection that had poor contact.  All the boards that I had classed as being defective were in fact okay.  Based on that that experience it might be an idea to be very skeptical about the quality of every connection.  It was a pain and took many hours of frustrating trials before the eureka moment!! 
Cheers

bacca5419, 2020-10-08:
Thanks for the advise. I will check over everything tomorrow and see what happens. Cheers

bacca5419, 2020-10-09:
Hi all. Still no go here. I have decided to desolder the connectors on the clough42 board ( the connectors that connect the board to the microcontroller ) I really have no answers to this problem. I send my thanks to you all for the advise to get me up and running. Cheers Noel ( bacca5419 )

Zymurgy42, 2020-10-09:
Where in Oz are you Noel?

bacca5419, 2020-10-10:
Hi. I am on the north side of Brisbane.

Zymurgy42, 2020-10-10:
A tad far from Melbourne.

JonWoellhaf, 2020-10-10:
Yes, but not as far as from Utah, USA! 

From: Zymurgy42 
Sent: October 9, 2020 20:27
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad programming (#83)
A tad far from Melbourne.

On Sat, 10 Oct 2020 at 12:41, bacca5419 <notifications@github.com> wrote:

> Hi. I am on the north side of Brisbane.

bacca5419, 2020-10-10:
Yeah just a tad. I guess I will work it out. Frustrating , but achievable. Thanks Gordon. Stay safe down there.

bacca5419, 2020-10-10:
That is for sure true. Im going to try some resoldering today. I guess it will either work or it wont. Fingers crossed. Cheers

bacca5419, 2020-10-10:
Right. I think I found the problem. New board that connects to the microcontroller. I found the gnd terminal on the board has nothing for the solder to stick to the board. Solder only sticks to the pin on the connmector block. I will order a new board from James ans start again. I will try attaching a photo here , but I think it wont allow me to attach. At the end of the day I figure I just stuffed the board when trying to solder. All other connections I did are great. Another seniors moment in the life of bacca5419.

davidgleach, 2020-10-11:
Hi Noel  That sounds like the problem I had!  For what its it is worth here is a picture of my final working set up.  
[image]

bacca5419, 2020-10-11:
Nice looking setup. Well done.

clough42, 2020-10-12:
FYI, I just added a continuity testing section to the troubleshooting documentation.  It has a map of most of the pin connections and test points on the board that can be used to verify continuity.  After many messages back and forth, when it turned out it was a lifted pad, I figured it was worth documenting for others who might run into similar trouble.

https://github.com/clough42/electronic-leadscrew/wiki/Continuity-Checks

bacca5419, 2020-10-13:
Thank you for this information. This is very useful for someone like me ( hopeless with electronics ). I have ordered new boards from james. When the boards arrive I am confident all will be good. Regards to all and I thank you all for the help you have given. I will keep you informed.

JonWoellhaf, 2020-10-13:
Excellent! Thanks, James.

Jan (Yan)

From: clough42 
Sent: October 12, 2020 17:03
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad programming (#83)
FYI, I just added a continuity testing section to the troubleshooting documentation. It has a map of most of the pin connections and test points on the board that can be used to verify continuity. After many messages back and forth, when it turned out it was a lifted pad, I figured it was worth documenting for others who might run into similar trouble.
https://github.com/clough42/electronic-leadscrew/wiki/Continuity-Checks

carneeki, 2020-10-18:
Hi Noel,

I'm in Sydney if you need some hands on help, and I have a spare ELS board I can lend / swap until your replacements arrive if it's time critical.

bacca5419, 2020-10-18:
Hi Adam. Thanks for the offer. I have a couple of boards on the way. Time isnt a problem. Im retired and the lathe is a hobby to help keep my motorcycles on the track. I probably will need help with loading James new update to studio suite. Im not that clever with computers. Regards. Noel.

bacca5419, 2020-10-23:
Hi all once again. Well the latest news , the new boards arrived from James (clough42), which he kindly assembled for me. New control panels from China. I reloaded the code composser from texas instruments , loaded the latest firmware , tested the continuity as per instructions from James ( all tested perfect ) plugged the launchpad into my computer , configured everything and whala , still nothing. I am at a complete lose of what to try next.

JonWoellhaf, 2020-10-23:
Hi, Noel.

What a bummer!

Please take advantage of Adam’s offer to help.

I am dying to know what the problem is! Have you tried stepping through the code?

Jan 

From: bacca5419 
Sent: October 23, 2020 00:08
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad programming (#83)

Hi all once again. Well the latest news , the new boards arrived from James (clough42), which he kindly assembled for me. New control panels from China. I reloaded the code composser from texas instruments , loaded the latest firmware , tested the continuity as per instructions from James ( all tested perfect ) plugged the launchpad into my computer , configured everything and whala , still nothing. I am at a complete lose of what to try next.

carneeki, 2020-10-23:
Let's try doing the equivalent of the Arduino "Blink" program?

Basically, wanting to start with a clean slate and just the launchpad board and work up from there. Sometimes debugging is easier to start from the bottom up.

I'm out for the evening but when I get back, I'll try and find some instructions on uploading a sample program to make one of the launchpad LEDs blink once a second like hazard lights on a car. We can build up from there once we rule out all other things. It may be something simple like a missing checkbox in TI CCS to write the firmware to flash memory so that the board 'remembers' the program without needing a computer plugged in. Maybe someone else here can beat me to finding those instructions.

davidgleach, 2020-10-23:
I wonder if the TI Launch Pad is dodgy.  Maybe you have bought a lemon, being the 1vin 10000 that got through the quality control system??

bacca5419, 2020-10-24:
Hi Adam. Thanks for your assistance , it is much appreciated. I am really lost with what to try next. All the lights seam to be alight on the microcontroller when connected and loading program from the computer. Everything matches what I see when James programs the launchpad. Im lost for any answers. Thanks again , Noel.

JonWoellhaf, 2020-10-26:
Noel,

I just had a thought ... (probably a stupid one)

I assume you are running the code in Debug mode. Correct?

When you click the Debug icon on the Toolbar – the one that looks like a bug – do you see messages that indicate the program is being transferred to the TI board?

In Debug mode, after the program downloads, the system will halt at the first statement in Main. This is InitSysCtrl().

When it halts, you have to click the green arrow icon on the Toolbar to tell the program to continue running.

You’re doing this, correct?

Jan

From: bacca5419 
Sent: October 23, 2020 00:08
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad programming (#83)

Hi all once again. Well the latest news , the new boards arrived from James (clough42), which he kindly assembled for me. New control panels from China. I reloaded the code composser from texas instruments , loaded the latest firmware , tested the continuity as per instructions from James ( all tested perfect ) plugged the launchpad into my computer , configured everything and whala , still nothing. I am at a complete lose of what to try next.

carneeki, 2020-10-26:
Ahh - if starting in debug mode, that won't write the ELS code to the flash memory, and will pause at all the breakpoints until you click the continue button as Jon points out.

If you have success with the continue button, try switching from 'Debug' to 'Release'. This profile will write the code to the Launchpad's built in flash memory (and drop some debugging information like the breakpoints), so the ELS will automatically start up. The Debug/Release choice can be found under the arrow next to the hammer icon in the top bar:
[image]

bacca5419, 2020-10-26:
Tried debug and then release. No go. Today I even tried deleting the project from my v9 workspace and then reloaded the current release code and set up a new project. Then tried loading to microcontroller , still no go. So now I have new control panel , new interface panel from James , ( which pass the continuity proceedures that James uploaded to github ) , fresh download of the project to my v9 workspace. Still no go , so today I said buggar it and ordered a new microcontroller. I really cant believe the problems I am having setting this up , especially considering I have followed James proceedure to the letter. Anyway thanks all. Noel. Oh yes and tried setting new break points from advise given , no go again and again. With all this drama at least I am still smiling.

bacca5419, 2020-10-29:
My new f280049c board arrived today , all excited to plug everything in and start cutting some threads I want to do. Well no real surprise , nothing , zip , exactly the same results as before. Maybe its time to throw this in the bin. Im over it. The gears might be a pain to change for different threads , but atleast they work. Because I sure as hell cant get this els system to do anything. Regards all Noel.

JonWoellhaf, 2020-10-29:
Noel,

Very sorry to hear the bad news!

Maybe tomorrow, are you willing to try a simple test? If so, please do this:

1) Unplug the TI board’s USB connector and remove power from James’ Boost board.
2) Remove the Boost board and the encoder wires.
3) Replace jumpers JP1, JP2, and JP3 on the TI board.
4) Plug the USB connector into the TI board (other end connected to your computer, of course). Do you see two red LEDs and one blue LED?

5) In Code Composer Studio, set Debug mode.
6) Click the Debug icon (the green bug) on the toolbar. Does the “Loading Program” popup appear briefly?
7) Do you see two red and two blue LEDs, and one green LED? 
8) Click the green arrowhead on the toolbar. Does the green LED go out?
9) In the Project Explorer panel, double click main.cpp. Does the main.cpp code appear?
10) In main.cpp, scroll way down to the statement that reads debug.begin2();
11) Double click the line number to the left of this statement. You have just set a breakpoint at that statement. The program should halt at that breakpoint. Is the line highlighted with a green background?
12) Is the arrowhead, previously gray, now green?
13) Click the green arrowhead. Does it momentary turn gray then return to green?
14) Is the breakpoint again highlighted in green? This concludes the test.
15) BE SURE TO REMOVE jumpers JP1, JP2, and JP3 on the TI board.

I’m VERY interested in learning why your system is misbehaving!

Please let us know how this goes, if you decide to try it.

Best wishes.

Jan (Yan)

From: bacca5419 
Sent: October 29, 2020 00:42
To: clough42/electronic-leadscrew 
Cc: JonWoellhaf ; Comment 
Subject: Re: [clough42/electronic-leadscrew] Launchpad programming (#83)

My new f280049c board arrived today , all excited to plug everything in and start cutting some threads I want to do. Well no real surprise , nothing , zip , exactly the same results as before. Maybe its time to throw this in the bin. Im over it. The gears might be a pain to change for different threads , but atleast they work. Because I sure as hell cant get this els system to do anything. Regards all Noel.

carneeki, 2020-10-29:
Hi Noel,

If you want to post the board(s) to me, I'll see if I can get it working on my lathe and post it back; email me (my email address is carneeki@carneeki.net ), and I'll send you my postal address.

But do as Jan suggests, and take the rest of the evening off :)

bacca5419, 2020-10-29:
Thanks Jan. I will try that tomorrow. I am getting a new internet provider tomorrow. First thing I want to try is to remove the code composer and release codes and reload them. The reason being , with my old internet it was extremely slow. To the point where it would stop downloading. I would have to close wifi and open then try to continue downloading. Frustrating I tell you. I was even thinking of downloading a newer version of code composer. Not sure about that as James system works with the version he loaded and I guess others have downloaded same as him and had success. I figure it has to be in the programming in code composer. I have now replaced all the boards. I will keep you informed of my progress , good or bad. I thank you for your help. Most apreciated. Regards. Noel.

bacca5419, 2020-10-31:
Hi Jan. Tried your test. Ok.1).done .2). Done 3). Done. 4). Yes have 2 reds 1 blue. 5). Done. 6). Yes. 7). I get 2 red . 2 blue . 2 green. 8). No . Greens stay on. 9). Yes. 10). Done. 11). More blue than green , but yes it is highlighted. 12). Yes green/ blue. 13). No . Stays grey and nothing appears to be happening. 14). Yes it is highlighted. Still have zip happening. Regards. Noel.

[23 more replies]