clough42/electronic-leadscrew #85: Motor Noise

chaszz, 2020-09-07:
The hybrid servo that James used looks like a reasonable candidate for my South Bend 1001 (8 x 18). However, my lathe is really quiet and he made a comment in the video about the motor making more noise than expected. Has this been an issue for anyone?   I think listening continually to stepper whine would drive me nuts.

petmakris, 2020-09-10:
Stepper do make noise. 

This is completely outside of the scope of ELS, but my opinion would be to checkout some stepper motor drivers that feature quiet operation like those based on Trinamic ICs. 

You could checkout various video reviews on stepper motor drivers before getting into.

Independent from the noise, I would choose a closed loop motor and driver.

chaszz, 2020-09-10:
Thanks. I hadn’t thought of looking for video reviews.

Charles Walling

closed 2020-10-11 by clough42