clough42/electronic-leadscrew #86: Connect the encoder in reverse.

ofer1973, 2020-09-12:
Hello :-)

Is it possible to install the encoder in the opposite way to the spindle?
Because only then can I install in my lathe.

tmetty, 2020-09-12:
The encoder only counts pulses, it doesn’t care what direction. My encoder turns the opposite direction as my spindle.

ofer1973, 2020-09-12:
> The encoder only counts pulses, it doesn’t care what direction. My encoder turns the opposite direction as my spindle.

So what happens when you turn on the spindle in reverse?
The stepper motor keeps turning in the same direction, or also turns in reverse?

JonWoellhaf, 2020-09-13:
My understand is the TI board's encoder interface comprises a bi-directional counter. Count increases for one direction of rotation and decreases for the other.

Clough's software compensates for up/down count and overflow/underflow, so spindle rpm, which is what drives the leadscrew rpm, is always a positive value, irrespective of the direction the encoder is turning.

The FWD/REV button on the control panel is what changes leadscrew rotation direction. It's independent of spindle rotation direction.

ofer1973, 2020-09-13:
Thanks friends .

clough42, 2020-10-07:
The RPM is independent of direction, and the leadscrew reverses when the spindle reverses, just like with real gears.

If the forward/reverse directions are not to your liking, you can uncomment

// #define INVERT_DIRECTION_PIN true

In Configuration.h to reverse it.

Typically, you want the forward direction to cut a right-hand thread.

ofer1973, 2020-10-07:
Thank you very much for the explanation 🙂

בתאריך יום ד׳, 7 באוק׳ 2020, 17:48, מאת clough42 ‏<notifications@github.com

clough42, 2020-10-08:
Marking this one closed.

closed 2020-10-08 by clough42