clough42/electronic-leadscrew #87: no stepper motor output

Selole, 2020-09-13:
Hi,

I have an issue with the stepper motor which is not rotating. My configuration is not similar to James. I try to use a PoStep60 motor driver and JMC-motion stepper motor. My encoder is as suggested. (ELS V2.1 and latest software) I haven't done any changes to software. (I enabled metric leadscrew, but changed it back for debugging purpose)

When I rotate the encoder, the control panel shows far more rev than I am turning and I have no output to my PoStep60 board. I tested my configuration of PoStep board with an arduino, and it works, so I assume this is ok. I suspect no signal on STEP pin. Since i bought three set of both TI boards and ELS boards, I have tested another set, but sadly with same results. I am sure I am doing something wrong, but can't figure out what. 

My wiring is as follows:
ELS->PoStep60
5V->5V (9)
Step- -> Step (3)
Dir- -> DIR (3)

Ground TI-> pin 1, 2, 4, 6, 8, 10
Pin 1 on PoStep60 is enable. When I connect it to ground, I can hear the motor is humming, but it is not rotating.

If someone can suggest which signal I should have where, I would be grateful. I am stuck for the moment :-)

Selole, 2020-09-14:
Hi again,

I can see pulses on GPIO0 and pin 1 on Q4, but not on pin 3. Please see attached screen dump. First picture measured at STP- and second on GPIO0. Could I have destroyed Q4? Shouldn't STP- have a higher voltage?

I soldered my last ELS board and tested it without connecting PoStep60 in case this caused some problem, but same results. 

I assume I am missing something here, but I don't see it. 
[image]
[image]

JonWoellhaf, 2020-09-14:
Selole,

I'll try to help.

What did you have connected to STEP- when you took the measurement?

STEP- is an open drain, so unless you have a pullup resistor to +5, you'll see little on STEP-.

Regards,

Jan (Yan)

JonWoellhaf, 2020-09-14:
I suspect your Arduino is programmed for an internal pullup on STEP, and that's one reason it works.

Does the PoStep60 configuration software allow you to configure its STEP input as active high or low, and does it have a configurable internal pullup?

Selole, 2020-09-15:
Hi Jan and thank you for the help. 

I added a pullup resistor (5k6) to 5V and that helped a lot. Now I can see pulses on Step-, with a maximum of 4.8V. If I connect PoStep60 together with the pullup resistor, maximum voltage peak is 1,72V, which is not enough to trigger PoStep60. I have read the manual, and searched the driver software, but can't find anything about configurable pullup or pulldown resistors, or active high or low.

I assume I need a pullup resistor also for dir-?

Regards, 
Ole

JonWoellhaf, 2020-09-15:
When Q4 is off, it is essentially an open circuit. So all the current flowing out of STEP- comes through the pullup resistor. A 5.6k ohm resistor attached to 5 volts will supply a maximum of about 0.9 mA. This is not nearly enough to pull the PoStep60's step input high.

Since you see 1.72 volts, the input impedance of the PoStep60 is about 3 kohms -- if my computations are correct. So the pullup resistor needs to be about 470 to 1000 ohms to pull the input high enough.

I think an Arduino output can source up to 40 mA, so it would have easily been able to drive the PoStep60 input high.

When Q4 turns on, it will have to handle the 5 to 10 mA flowing through the pullup resistor. It can easily handle 10 times that much, so no worries there.

Yes, I think you will also need a similar pullup resistor on the DIR- output.

Selole, 2020-09-16:
Hi Jan,

It works!!! I added a 470 ohms resistor as you suggested, and it works. One step closer to throw out my geared leadscrew. Really appreciate your help, thank you.

Regards, 
Ole

closed 2020-09-16 by Selole

JonWoellhaf, 2020-09-16:
Ole,

Great news! You're welcome. It's always fun to give my rusty old troubleshooting gears a little workout. :)

Don't you still have the issue of the inaccurate rpm readout?

Jan

Selole, 2020-09-17:
Hi Jan,

Yes, I have, but I assume this will be software configuration. I need to study what to expect as output in the display. I think you should see rev of your spindle, and if you have 1:1 ratio between your spindle and your encoder, you will see rev of your encoder. As it is now, I see a lot more rev in the display than I am turning the encoder. For the moment, I do not have constant speed on my encoder, so I don't know if I am off by a hole number, or a fraction. I think this will be the first to solve, before I post any new challenges for you :-) Just to tell you that it is not correct, is too little information.

I haven't studied the program yet, but I assume it's default ratio is 1:1

I can