clough42/electronic-leadscrew #88: Spindle Speed Readout Problem

davidgleach, 2020-09-22:
Having built the ELS and tested out its accuracy w.r.t. screw cutting I am very pleased with the thread accuracy.  However the readout for the lathe spindle rotation rpm is exactly half the actual rotation speed.  I hadn't concerned myself with this error previously as I was anxious to get the thread cutting facility up and running to suit a current project.
 
I don't have programming skills but I assume somewhere in the code is a statement that converts encoder pulses to rotational speed.  Could that be changed?

Has anyone any clues as to how this error could come about and what might be the solution?  As I said the thread cutting is spot on.

JonWoellhaf, 2020-09-22:
Hi, David.

Two questions:

Are you using the exact same rotary encoder James used?

How are you measuring the actual spindle rpm?

Jan (Yan)

davidgleach, 2020-09-23:
> <electronic-leadscrew@noreply.github.com>
> <author@noreply.github.com>
> Problem (#88)

Hi Yan
I am an idiot.  The encoder I bought was an OMRON E6B2 - CWZ6C .  The 
package and data sheet said 1024 p/rev.  However, I have just looked 
closely at the actual unit as mounted and guess what - it is a 600 
p/rev.  No wonder I had to tweak the pitch and microstep config data!!  
At my age (84) one would think I'd be more thorough. Anyway I can now 
put things right  for which I am most grateful to you.

David Leach

Home: 03 5310 6313

Mob:    0416 119 381

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2020-09-23 by davidgleach

JonWoellhaf, 2020-09-23:
Good news, David.

I think the idiot is in the Omron shipping department! :)

The 600 p/r encoder will probably work fine.

If you change #define ENCODER_RESOLUTION in Configuration.h to 2400, is the rpm readout correct?

Doing this, of course, will mean you'll have to tweak the other parameters to get the threading to work properly.

Good luck!

Please let us know how this all works out.

Regards,

Jan (Yan)

davidgleach, 2020-09-23:
Yes - that is exactly what I've done and the system works just fine.  I 
think the whole concept is brilliant.

Cheers

David Leach

Home: 03 5310 6313

Mob:    0416 119 381

Cc: "davidgleach" <davidgleach@iinet.net.au>; "State change" 
<state_change@noreply.github.com>
Sent: 23/09/2020 11:54:06 AM
Problem (#88)

https://www.avast.com/antivirus