clough42/electronic-leadscrew #89: Intermittent "Break Tool" mode and fix

ngreenou, 2020-09-26:
Just completed my build and installation on an Atlas/Craftsman 12" lathe. Fantastic project!! However, my build has a third mode called "Break Tool". This randomly happens every few minutes. It will act like all buttons have been pressed simultaneously. The leadscrew reverses, the display shows "No Settings", the mode flips to metric threading and the feed speed increases dramatically. Oops. I was able to suppress this significantly by adding 33 ohm series resistors to the head cable STB, CLK and DIO lines on both ends, but it still occasionally happens. My head cable is a 4-foot piece of standard 8-wire #22 shielded industrial control cable. I suspect that the "feature" is caused by either a corrupted "read switches" code being read by the display, or a false readback of all-switches-pressed for unknown reasons. 
I've added a crude sanity check to the switch readback code in ControlPanel.cpp. The simplest way to do this is to check that the value of newkeys.all is less than or equal to 128 for a reasonably valid keyswitch readback. I've made the following 3 line addition to 1.1.02 ControlPanel.cpp in the ControlPanel::getKeys() routine around line 239:

newkeys - readKeys();   // Original Clough42 code
// Detect invalid key responses    NLG
if(newKeys.all > 128)  {  //NLG
      return noKeys;  // Bail out if bogus key readback    NLG
}    /NLG
if( newkeys.all != this->keys.all ) { // Original Clough42 code

A slightly better but slower solution:
    newKeys = readKeys();  //Clough42 code
    // Detect invalid key responses  NLG 9/23/20
    switch(newKeys.all) {
        case 128:
        case 64:
        case 32:
        case 16:
        case 8:
        case 4:
        case 1:
        case 0:
            break; // Pass on if OK
        default:
            return noKeys;  // Bail out on bad key data   NLG
        break;
    }   // End of NLG mod
    if( newKeys.all != this->keys.all ) { // Clough42 code

The only valid values for newKeys.all are 128, 64, 32, 16, 8, 4, 1 and 0. All others are either fat fingers or bogus reads.
I suspect others can do a much better code job than this. This code addition has completely solved my issue. 
Thanks again for the fantastic project.

clough42, 2020-09-27:
It sounds like you're experiencing some kind of electrical noise issue
causing a problem with the button readback.  It's a bidirectional
open-collector bus, so excess capacitance or other issues can be a big
problem.

It's best to keep the control panel cable as short as possible and keep it
away from the motor or power wiring.  I've seen a few people also connect
the grounds for the logic power and the motor power together, and this
causes no end of problems with noise and false triggering.

You might also try using a different 5V power supply.  I've seen weird
issues with some cheap cell phone chargers, in particular, with voltages
fluctuating between 4.7 and 5.25V as they try to communicate with the
device to figure out the proper charging voltage and current.

James

ngreenou, 2020-09-27:
Thanks for your quick reply! I've already tried all of your suggestions so
reverted to the button sanity check code. Sanity checks are always good
when cables or humans are involved. I think it should be included in the
release code.
Once again, thanks for the incredible project!

clough42, 2020-10-05:
I am starting work on the next version of the firmware in the new "dev" branch.  I pushed a change today that slows down the SPI bus from 250Kbps to around 97Kbps.  My hope is that this will make the control panel a little less noise-sensitive, or at least less sensitive to cable capacitance.

The relevant changes are here:  https://github.com/clough42/electronic-leadscrew/commit/5e3fac01ea128eefeb8c957153cba54e006c96a4

If you try it, let me know if it helps.  I have been so far unable to reproduce the issue.

tmetty, 2020-10-06:
I am also having this issue, it happens virtually every time I use the lathe. Fortunately I haven't broken any tools but it's unnerving to see it switch to a metric left-hand thread in the middle of a finishing pass. Invariably it will reverse direction, change from inch to metric and change from either thread to feed, or feed to thread. Occasionally it will also invoke the set button message.

The motor is a Leeson 108092 1.5hp PM DC motor with a Minarik MM23001C controller. The motor controller is mounted externally with only the speed potentiometer wires coming to the lathe.
I'm using all the recommended electronics except that Automation Technology substituted a KL23-31N-1000 servo for the KL-23-2N-1000. It's a 2 phase hybrid with more torque, and it uses the KL-5080D4 instead of the KL-5080H. All the components are installed within the old gear-change housing, so lead lengths are minimal.

clough42, 2020-10-06:
I checked in a change over the weekend (on the dev branch only) that slows
down the communication with the control panel to try to help with noise
issues.  You might try this to see if it helps.

https://github.com/clough42/electronic-leadscrew/commit/5e3fac01ea128eefeb8c957153cba54e006c96a4

I am planning to integrate some of the other changes submitted by users
(locking the control panel and validating inputs) but eliminating the noise
in the first place is the best solution.

Keeping the logic wiring away from the power wiring and making sure you
don't have the logic ground connected to the servo/stepper ground are very
important.

James

clough42, 2020-10-06:
The more I think about this, the more I suspect the real problem here is that the TM1638 chip in the display module isn't properly interpreting the command to read back the button states.  The chip works like this:

- MCU sends 0x40 (set increment mode)
- MCU sends 0x42 (read keys)
- MCU toggles the clock for four bytes and the TM1638 transmits the values

My hypothesis is that the TM1638 chip is occasionally missing the 0x42 command.  Since the bus is open-collector, if the chip misses the command, it won't activate, and the default state of the bus is high, so it will read all 1's.

Brainstorming possible causes of missing the read command:

- Electrical noise
- Loose or intermittent connections
- Borderline timing

Here are some ideas for dealing with this possibility:

- Slow down the serial communication to reduce the chance of a missed command (already on the dev branch)
- Increase the rise time and input delays slightly
- Validate the inputs and ignore any values that don't represent exactly one button press
- Require any input pattern to be stable for multiple read cycles before responding
- Ignore all input while the spindle is turning

It will probably be a few days before I can make the changes.

Thoughts?

tmetty, 2020-10-06:
Today I turned on just the ELS and walked away, within an hour it had changed settings.

Thank you, James. I am thoroughly enjoying your project.

clough42, 2020-10-06:
I assume more than one thing changed?  I.e. mode, direction, metric?  Can
you tell?

clough42, 2020-10-06:
I just put a few more changes into the dev branch that I hope will address most of the reliability issues people have been seeing.  Here are the changes:

- Fix overflow that caused problems with large metric threads (thanks @hugosman95)
- Slow down the SPI bus from 250Kbps to 98Kbps
- Increase SPI timing delays slightly
- Aggressively filter bad key reads:
  - Require multiple, identical, consecutive reads before acting
  - Validate that only one key is pressed at a time (thanks @ngreenou)
  - Require all keys to be released between key presses
- Do not allow changes to mode or direction while the machine is in motion (thanks @C4fishfood)
- Add configuration option to ignore all keypresses when the machine is in motion (thanks @C4fishfood)

You can either pull the dev branch or you can download a zip of the code with these fixes here:

  https://github.com/clough42/electronic-leadscrew/archive/dev.zip

If you try this, please report back.  I would like some positive feedback before I mark it as a global release.

tmetty, 2020-10-06:
Yes, more than one thing changed. I put it into left-hand 3.0 mm thread mode, an hour later it was in default .005” turning mode.

clough42, 2020-10-07:
Awesome!  Not because your problem is worse, but because it fits with my current theory.

I have a software fix that I will release later tonight (version 1.2.00) that should prevent this from happening, though I'm a little uncomfortable not fully understanding the cause.

One thing you might try before upgrading, just as a test: try wiggling the control panel wiring to see if it causes the issue.  There's a possibility that a loose or intermittent connection could possibly cause this as well, though the updated software should prevent it from causing problems.  It might also prevent you from diagnosing loose wiring because it won't let it manifest.

clough42, 2020-10-07:
I just made a new release (v1.2.00) that likely fixes this issue.  Please try it and let me know if it helps.

https://github.com/clough42/electronic-leadscrew/releases/tag/v1.2.00

tmetty, 2020-10-07:
I installed v1.2.00  using the default key press settings in configuration.h (line 139). I configured it in left-hand 3 mm threading mode and left it for 12 hours with no spindle rotation. The mode didn't change, I'll do some more testing but it looks like success! Thank you, James.

clough42, 2020-10-07:
Awesome!  Thanks for the feedback.

clough42, 2020-10-08:
I'm marking this one closed.

closed 2020-10-08 by clough42